coolsnowwolf/lede #9142: 无法编译armvirt efi版本

jiangcuo, 2022-03-28:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：无法编译armvirt efi版本

### (2) 路由器型号和固件版本

A： rock pi 4A,kvm

### (3) 详细日志

A：在target/linux/armvirt/config-5.4中加入
CONFIG_EFI_STUB=y
CONFIG_EFI=y
CONFIG_EFI_VARS=y
CONFIG_ARCH_SUPPORTS_ACPI=y
CONFIG_ACPI=y
执行，make kernel_menuconfig 找不到UEFI相关选项，直接编译也无法启动。
报错“error: plain image kernel not supported - rebuild with CONFIG_(U)EFI_STUB
enabled.”
ps，官方版本没问题。

WYC-2020, 2022-03-28:
没能力不要去手动修改config，efi也要支持才行，不是加了就可以，最好去menuconfig里面看有没这个功能

jiangcuo, 2022-03-28:
感谢回复，使用官方版本确实可以正常编译，我已经成功了。可以参考https://forum.openwrt.org/t/arm64-armvirt64-uefi-efi-openwrt-target/82740/29
只是在lede中没有成功。我会去再次尝试。

closed 2022-04-28 by jiangcuo