coolsnowwolf/lede #9145: macOS 12.3下编译crpytodev-linux编译失败

ricksuzade-maker, 2022-03-29:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：macOS 12.3下编译crpytodev-linux编译失败

### (2) 路由器型号和固件版本

A：x86 macOS 12.3

### (3) 详细日志

A：
OpenWrt-libtool: link: x86_64-openwrt-linux-musl-gcc -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/Volumes/Compile/scripts/lede/build_dir/target-x86_64_musl/udns-0.4=udns-0.4 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z -Wl,now -Wl,-z -Wl,relro -znow -zrelro -o .libs/rblcheck rblcheck.o  -L/Volumes/Compile/scripts/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/Volumes/Compile/scripts/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib ./.libs/libudns.so
x86_64-openwrt-linux-musl-gcc -DHAVE_CONFIG_H -I.   -I/Volumes/Compile/scripts/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/Volumes/Compile/scripts/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/Volumes/Compile/scripts/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include   -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/Volumes/Compile/scripts/lede/build_dir/target-x86_64_musl/udns-0.4=udns-0.4 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -MT ex-rdns.o -MD -MP -MF .deps/ex-rdns.Tpo -c -o ex-rdns.o ex-rdns.c
In file included from ex-rdns.c:28:
/Volumes/Compile/scripts/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/sys/poll.h:1:2: warning: #warning redirecting incorrect #include <sys/poll.h> to <poll.h> [-Wcpp]
 #warning redirecting incorrect #include <sys/poll.h> to <poll.h>
  ^~~~~~~
mv -f .deps/ex-rdns.Tpo .deps/ex-rdns.Po
/bin/sh ./libtool  --tag=CC   --mode=link x86_64-openwrt-linux-musl-gcc  -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/Volumes/Compile/scripts/lede/build_dir/target-x86_64_musl/udns-0.4=udns-0.4 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro   -L/Volumes/Compile/scripts/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/Volumes/Compile/scripts/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -znow -zrelro  -o ex-rdns ex-rdns.o ./libudns.la 
OpenWrt-libtool: link: x86_64-openwrt-linux-musl-gcc -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/Volumes/Compile/scripts/lede/build_dir/target-x86_64_musl/udns-0.4=udns-0.4 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z -Wl,now -Wl,-z -Wl,relro -znow -zrelro -o .libs/ex-rdns ex-rdns.o  -L/Volumes/Compile/scripts/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/Volumes/Compile/scripts/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib ./.libs/libudns.so
make[5]: Leaving directory '/Volumes/Compile/scripts/lede/build_dir/target-x86_64_musl/udns-0.4'
make[4]: Leaving directory '/Volumes/Compile/scripts/lede/build_dir/target-x86_64_musl/udns-0.4'
CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/Volumes/Compile/scripts/lede/build_dir/target-x86_64_musl/udns-0.4=udns-0.4 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/Volumes/Compile/scripts/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/Volumes/Compile/scripts/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/Volumes/Compile/scripts/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/Volumes/Compile/scripts/lede/build_dir/target-x86_64_musl/udns-0.4=udns-0.4 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/Volumes/Compile/scripts/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/Volumes/Compile/scripts/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/Volumes/Compile/scripts/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include " LDFLAGS="-L/Volumes/Compile/scripts/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/Volumes/Compile/scripts/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -znow -zrelro " make -C /Volumes/Compile/scripts/lede/build_dir/target-x86_64_musl/udns-0.4/. AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/Volumes/Compile/scripts/lede/build_dir/target-x86_64_musl/udns-0.4=udns-0.4 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CROSS="x86_64-openwrt-linux-musl-" ARCH="x86_64" DESTDIR="/Volumes/Compile/scripts/lede/build_dir/target-x86_64_musl/udns-0.4/ipkg-install"  install;
make[4]: Entering directory '/Volumes/Compile/scripts/lede/build_dir/target-x86_64_musl/udns-0.4'
make[5]: Entering directory '/Volumes/Compile/scripts/lede/build_dir/target-x86_64_musl/udns-0.4'
 /usr/local/opt/coreutils/libexec/gnubin/mkdir -p '/Volumes/Compile/scripts/lede/build_dir/target-x86_64_musl/udns-0.4/ipkg-install/usr/lib'
 /bin/sh ./libtool   --mode=install /Volumes/Compile/scripts/lede/staging_dir/host/bin/install -c   libudns.la '/Volumes/Compile/scripts/lede/build_dir/target-x86_64_musl/udns-0.4/ipkg-install/usr/lib'
OpenWrt-libtool: install: /Volumes/Compile/scripts/lede/staging_dir/host/bin/install -c .libs/libudns.so.0.0.0 /Volumes/Compile/scripts/lede/build_dir/target-x86_64_musl/udns-0.4/ipkg-install/usr/lib/libudns.so.0.0.0
OpenWrt-libtool: install: (cd /Volumes/Compile/scripts/lede/build_dir/target-x86_64_musl/udns-0.4/ipkg-install/usr/lib && { ln -s -f libudns.so.0.0.0 libudns.so.0 || { rm -f libudns.so.0 && ln -s libudns.so.0.0.0 libudns.so.0; }; })
OpenWrt-libtool: install: (cd /Volumes/Compile/scripts/lede/build_dir/target-x86_64_musl/udns-0.4/ipkg-install/usr/lib && { ln -s -f libudns.so.0.0.0 libudns.so || { rm -f libudns.so && ln -s libudns.so.0.0.0 libudns.so; }; })
OpenWrt-libtool: install: /Volumes/Compile/scripts/lede/staging_dir/host/bin/install -c .libs/libudns.lai /Volumes/Compile/scripts/lede/build_dir/target-x86_64_musl/udns-0.4/ipkg-install/usr/lib/libudns.la
OpenWrt-libtool: install: /Volumes/Compile/scripts/lede/staging_dir/host/bin/install -c .libs/libudns.a /Volumes/Compile/scripts/lede/build_dir/target-x86_64_musl/udns-0.4/ipkg-install/usr/lib/libudns.a
OpenWrt-libtool: install: chmod 644 /Volumes/Compile/scripts/lede/build_dir/target-x86_64_musl/udns-0.4/ipkg-install/usr/lib/libudns.a
OpenWrt-libtool: install: x86_64-openwrt-linux-musl-gcc-ranlib /Volumes/Compile/scripts/lede/build_dir/target-x86_64_musl/udns-0.4/ipkg-install/usr/lib/libudns.a
OpenWrt-libtool: install: warning: remember to run `libtool --finish /usr/lib'
 /usr/local/opt/coreutils/libexec/gnubin/mkdir -p '/Volumes/Compile/scripts/lede/build_dir/target-x86_64_musl/udns-0.4/ipkg-install/usr/bin'
  /bin/sh ./libtool   --mode=install /Volumes/Compile/scripts/lede/staging_dir/host/bin/install -c dnsget rblcheck ex-rdns '/Volumes/Compile/scripts/lede/build_dir/target-x86_64_musl/udns-0.4/ipkg-install/usr/bin'
OpenWrt-libtool: install: warning: `./libudns.la' has not been installed in `/usr/lib'
OpenWrt-libtool: install: /Volumes/Compile/scripts/lede/staging_dir/host/bin/install -c .libs/dnsget /Volumes/Compile/scripts/lede/build_dir/target-x86_64_musl/udns-0.4/ipkg-install/usr/bin/dnsget
OpenWrt-libtool: install: warning: `./libudns.la' has not been installed in `/usr/lib'
OpenWrt-libtool: install: /Volumes/Compile/scripts/lede/staging_dir/host/bin/install -c .libs/rblcheck /Volumes/Compile/scripts/lede/build_dir/target-x86_64_musl/udns-0.4/ipkg-install/usr/bin/rblcheck
[... cut ...]
time: package/kernel/cryptodev-linux/compile#0.73#0.76#1.70
    ERROR: package/kernel/cryptodev-linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/cryptodev-linux/compile] Error 1
make[2]: Leaving directory '/Volumes/Compile/scripts/lede'
make[1]: *** [package/Makefile:110: /Volumes/Compile/scripts/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/Volumes/Compile/scripts/lede'
make: *** [/Volumes/Compile/scripts/lede/include/toplevel.mk:230: world] Error 2

WYC-2020, 2022-03-29:
建议放弃吧，苹果的编译环境都没得，修的人少之又少，一个bug说不定等几万年都没人搞哈哈

iKenzo, 2022-05-04:
同样问题+1

hiby-arch, 2022-05-05:
同样问题+1，还以为我的操作有误呢

coolsnowwolf, 2026-09-05:
macOS 下 cryptodev-linux 编译失败缺少完整首个错误和主机工具链信息，属于主机环境兼容性问题，关闭为“不计划处理”。

closed 2026-09-05 by coolsnowwolf