coolsnowwolf/lede #9146: 最新X86模式编译后ipsec服务器问题（ipsec能连不上上网，L2TP连不上）

qstzs, 2022-03-29:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 已搜索相关issue ，未解决

## 2. 详细叙述

### (1) 具体问题

A：使用最新的LEDE源码编译的，插件就passwall frps frpc kms ipsec-server,编译选的是server，不是那个vpnc
，其他均默认。编译了几次 都是PPTP可以正常连接 也可以上网  IPSEC VPN服务器 有问题 
主要表现为 L2TP模式无法连接  IPSEC模式可以连接。

### (2) 路由器型号和固件版本

A：软路由安装在PVE7上 最新源码编译

### (3) 详细日志

A：IPSEC模式可以连接上 上不了网 下面是断开和连接的日志
`Tue Mar 29 12:13:05 2022 daemon.info : 10[ENC] parsed INFORMATIONAL_V1 request 281109273 [ HASH D ]
Tue Mar 29 12:13:05 2022 daemon.info : 10[IKE] received DELETE for ESP CHILD_SA with SPI 0ddb4c17
Tue Mar 29 12:13:05 2022 daemon.info : 10[IKE] CHILD_SA not found, ignored
Tue Mar 29 12:13:05 2022 daemon.info : 07[ENC] parsed INFORMATIONAL_V1 request 3275822627 [ HASH D ]
Tue Mar 29 12:13:05 2022 daemon.info : 07[IKE] received DELETE for IKE_SA IKEv1-PSK-XAuth[5]
Tue Mar 29 12:13:05 2022 daemon.info : 07[IKE] deleting IKE_SA IKEv1-PSK-XAuth[5] between 202.202.100.4[202.202.100.4]...202.202.100.105[192.168.123.42]
Tue Mar 29 12:13:05 2022 authpriv.info : 07[IKE] deleting IKE_SA IKEv1-PSK-XAuth[5] between 202.202.100.4[202.202.100.4]...202.202.100.105[192.168.123.42]
Tue Mar 29 12:13:05 2022 daemon.info : 07[IKE] IKE_SA IKEv1-PSK-XAuth[5] state change: ESTABLISHED => DELETING
Tue Mar 29 12:13:05 2022 daemon.info : 07[IKE] IKE_SA IKEv1-PSK-XAuth[5] state change: DELETING => DELETING
Tue Mar 29 12:13:05 2022 daemon.info : 07[IKE] IKE_SA IKEv1-PSK-XAuth[5] state change: DELETING => DESTROYING
Tue Mar 29 12:13:05 2022 daemon.info : 07[CFG] lease 192.168.4.10 by 'qwe' went offline
Tue Mar 29 12:13:30 2022 daemon.info : 06[ENC] parsed ID_PROT request 0 [ SA V V V V V V V V V V V V V V ]
Tue Mar 29 12:13:30 2022 daemon.info : 06[CFG] looking for an IKEv1 config for 202.202.100.4...202.202.100.105
Tue Mar 29 12:13:30 2022 daemon.info : 06[CFG]   candidate: %any...%any, prio 24
Tue Mar 29 12:13:30 2022 daemon.info : 06[CFG]   candidate: %any...%any, prio 28
Tue Mar 29 12:13:30 2022 daemon.info : 06[CFG]   candidate: %any...%any, prio 28
Tue Mar 29 12:13:30 2022 daemon.info : 06[CFG] found matching ike config: %any...%any with prio 28
Tue Mar 29 12:13:30 2022 daemon.info : 06[IKE] local endpoint changed from 0.0.0.0[500] to 202.202.100.4[500]
Tue Mar 29 12:13:30 2022 daemon.info : 06[IKE] remote endpoint changed from 0.0.0.0 to 202.202.100.105[500]
Tue Mar 29 12:13:30 2022 daemon.info : 06[IKE] received NAT-T (RFC 3947) vendor ID
Tue Mar 29 12:13:30 2022 daemon.info : 06[IKE] received draft-ietf-ipsec-nat-t-ike vendor ID
Tue Mar 29 12:13:30 2022 daemon.info : 06[IKE] received draft-ietf-ipsec-nat-t-ike-08 vendor ID
Tue Mar 29 12:13:30 2022 daemon.info : 06[IKE] received draft-ietf-ipsec-nat-t-ike-07 vendor ID
Tue Mar 29 12:13:30 2022 daemon.info : 06[IKE] received draft-ietf-ipsec-nat-t-ike-06 vendor ID
Tue Mar 29 12:13:30 2022 daemon.info : 06[IKE] received draft-ietf-ipsec-nat-t-ike-05 vendor ID
Tue Mar 29 12:13:30 2022 daemon.info : 06[IKE] received draft-ietf-ipsec-nat-t-ike-04 vendor ID
Tue Mar 29 12:13:30 2022 daemon.info : 06[IKE] received draft-ietf-ipsec-nat-t-ike-03 vendor ID
Tue Mar 29 12:13:30 2022 daemon.info : 06[IKE] received draft-ietf-ipsec-nat-t-ike-02 vendor ID
Tue Mar 29 12:13:30 2022 daemon.info : 06[IKE] received draft-ietf-ipsec-nat-t-ike-02\n vendor ID
Tue Mar 29 12:13:30 2022 daemon.info : 06[IKE] received XAuth vendor ID
Tue Mar 29 12:13:30 2022 daemon.info : 06[IKE] received Cisco Unity vendor ID
Tue Mar 29 12:13:30 2022 daemon.info : 06[IKE] received FRAGMENTATION vendor ID
Tue Mar 29 12:13:30 2022 daemon.info : 06[IKE] received DPD vendor ID
Tue Mar 29 12:13:30 2022 daemon.info : 06[IKE] 202.202.100.105 is initiating a Main Mode IKE_SA
Tue Mar 29 12:13:30 2022 authpriv.info : 06[IKE] 202.202.100.105 is initiating a Main Mode IKE_SA
Tue Mar 29 12:13:30 2022 daemon.info : 06[IKE] IKE_SA (unnamed)[6] state change: CREATED => CONNECTING
Tue Mar 29 12:13:30 2022 daemon.info : 06[CFG] selecting proposal:
Tue Mar 29 12:13:30 2022 daemon.info : 06[CFG]   no acceptable ENCRYPTION_ALGORITHM found
Tue Mar 29 12:13:30 2022 daemon.info : 06[CFG] selecting proposal:
Tue Mar 29 12:13:30 2022 daemon.info : 06[CFG]   no acceptable ENCRYPTION_ALGORITHM found
Tue Mar 29 12:13:30 2022 daemon.info : 06[CFG] selecting proposal:
Tue Mar 29 12:13:30 2022 daemon.info : 06[CFG]   no acceptable ENCRYPTION_ALGORITHM found
Tue Mar 29 12:13:30 2022 daemon.info : 06[CFG] selecting proposal:
Tue Mar 29 12:13:30 2022 daemon.info : 06[CFG]   no acceptable ENCRYPTION_ALGORITHM found
Tue Mar 29 12:13:30 2022 daemon.info : 06[CFG] selecting proposal:
Tue Mar 29 12:13:30 2022 daemon.info : 06[CFG]   no acceptable ENCRYPTION_ALGORITHM found
Tue Mar 29 12:13:30 2022 daemon.info : 06[CFG] selecting proposal:
Tue Mar 29 12:13:30 2022 daemon.info : 06[CFG]   no acceptable ENCRYPTION_ALGORITHM found
[... 268 lines cut ...]
Tue Mar 29 12:17:29 2022 daemon.info : 09[ENC] generating QUICK_MODE response 3844487398 [ HASH SA No ID ID NAT-OA NAT-OA ]
Tue Mar 29 12:17:29 2022 daemon.info : 13[ENC] parsed QUICK_MODE request 3844487398 [ HASH ]
Tue Mar 29 12:17:29 2022 daemon.info : 13[KNL] received netlink error: No such file or directory (2)
Tue Mar 29 12:17:29 2022 daemon.info : 13[KNL] unable to add SAD entry with SPI c1fb80ee (FAILED)
Tue Mar 29 12:17:29 2022 daemon.info : 13[KNL] received netlink error: No such file or directory (2)
Tue Mar 29 12:17:29 2022 daemon.info : 13[KNL] unable to add SAD entry with SPI 0af639bc (FAILED)
Tue Mar 29 12:17:29 2022 daemon.info : 13[IKE] unable to install inbound and outbound IPsec SA (SAD) in kernel
Tue Mar 29 12:17:29 2022 daemon.info : 13[KNL] deleting policy 202.202.100.105/32[udp/58445] === 202.202.100.4/32[udp/l2f] in failed, not found
Tue Mar 29 12:17:29 2022 daemon.info : 13[IKE] queueing QUICK_DELETE task
Tue Mar 29 12:17:29 2022 daemon.info : 13[IKE] activating new tasks
Tue Mar 29 12:17:29 2022 daemon.info : 13[IKE]   activating QUICK_DELETE task
Tue Mar 29 12:17:29 2022 daemon.info : 13[IKE] sending DELETE for ESP CHILD_SA with SPI 0af639bc
Tue Mar 29 12:17:29 2022 daemon.info : 13[ENC] generating INFORMATIONAL_V1 request 1345036091 [ HASH D ]
Tue Mar 29 12:17:29 2022 daemon.info : 13[IKE] activating new tasks
Tue Mar 29 12:17:29 2022 daemon.info : 13[IKE] nothing to initiate
Tue Mar 29 12:17:49 2022 daemon.info : 15[ENC] parsed INFORMATIONAL_V1 request 504564700 [ HASH D ]
Tue Mar 29 12:17:49 2022 daemon.info : 15[IKE] received DELETE for ESP CHILD_SA with SPI 0af639bc
Tue Mar 29 12:17:49 2022 daemon.info : 15[IKE] CHILD_SA not found, ignored
Tue Mar 29 12:17:49 2022 daemon.info : 15[ENC] parsed INFORMATIONAL_V1 request 1067882381 [ HASH D ]
Tue Mar 29 12:17:49 2022 daemon.info : 15[IKE] received DELETE for IKE_SA L2TP-IKEv1-PSK[7]
Tue Mar 29 12:17:49 2022 daemon.info : 15[IKE] deleting IKE_SA L2TP-IKEv1-PSK[7] between 202.202.100.4[202.202.100.4]...202.202.100.105[192.168.123.42]
Tue Mar 29 12:17:49 2022 authpriv.info : 15[IKE] deleting IKE_SA L2TP-IKEv1-PSK[7] between 202.202.100.4[202.202.100.4]...202.202.100.105[192.168.123.42]
Tue Mar 29 12:17:49 2022 daemon.info : 15[IKE] IKE_SA L2TP-IKEv1-PSK[7] state change: ESTABLISHED => DELETING
Tue Mar 29 12:17:49 2022 daemon.info : 15[IKE] IKE_SA L2TP-IKEv1-PSK[7] state change: DELETING => DELETING
Tue Mar 29 12:17:49 2022 daemon.info : 15[IKE] IKE_SA L2TP-IKEv1-PSK[7] state change: DELETING => DESTROYING`

mitsukileung, 2022-03-29:
IPSEC可以正常工作呀，L2TP没试，不清楚

WYC-2020, 2022-03-29:
哈哈，我可以说我的都正常吗

qstzs, 2022-03-29:
> 哈哈，我可以说我的都正常吗

特别奇葩 我自己编译的就不行，我用的别人编译的那个全家桶版本也可以！！

sbwcwfhv001, 2022-03-29:
别人不是用的最新的源码，我还在用21.10.1的在编译

qstzs, 2022-03-30:
> 别人不是用的最新的源码，我还在用21.10.1的在编译

解决了 我用另一个commit的版本源码 编译就正常了  不知道为啥  那个版本的luci-ipsec-server 这个文件在package/lede里面
不知道是哪里的问题

qstzs, 2022-03-30:
> 别人不是用的最新的源码，我还在用21.10.1的在编译

老哥你编译的L2TP可以用吗？

liangchaofu, 2022-04-07:
我用原本21.02.1，全部功能正常。

coolsnowwolf, 2026-09-05:
IPsec/L2TP 能否上网涉及策略、地址池、转发和 NAT，报告缺少 strongSwan、xfrm、路由和防火墙完整日志，无法定位 LEDE 包缺陷，关闭为“不计划处理”。

closed 2026-09-05 by coolsnowwolf