coolsnowwolf/lede #9151: 编译lean大固件时无法选择smartdns插件。

eanucc, 2022-03-30:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述
在使用make menuconfig时，终端显示：ERROR: please fix package/lean/luci-app-smartdns/Makefile - see logs/package/lean/luci-app-smartdns/dump.txt

### (1) 具体问题

A：查看dump.txt文件内容，出现以下提示：Makefile:24: ../../luci.mk: No such file or directory
make[2]: *** No rule to make target '../../luci.mk'.  Stop.

### (2) 路由器型号和固件版本

A：OpenWrt R22.3.13 / LuCI Master (git-22.086.14749-160352f)
B:  内核版本 5.15.31

WYC-2020, 2022-03-30:
谁知道你从哪里拉的，包含路径问题

WYC-2020, 2022-03-30:
> 用的就是lean大的源码。
> […](#)

l大的何时lean文件夹放luci app了，你不是最新的吧

WYC-2020, 2022-03-30:
> 用的就是lean大的源码。
> […](#)

闲麻烦用我仓库的smartdns，但是这个是我的修改版，把两个服务器分别控制，或者参考我的makefile自己修复一下

eanucc, 2022-03-30:
> > 用的就是lean大的源码。
> > […](#)
> 
> 闲麻烦用我仓库的smartdns，但是这个是我的修改版，把两个服务器分别控制，或者参考我的makefile自己修复一下

关注一下，谢谢你！

eanucc, 2022-03-30:
> > > 用的就是lean大的源码。
> > > […](#)
> > 
> > 
> > 闲麻烦用我仓库的smartdns，但是这个是我的修改版，把两个服务器分别控制，或者参考我的makefile自己修复一下
> 
> 关注一下，谢谢你！

我找了一下L大的源码，smartdns有，但luci-app-smartdns确实没有。

naoki66, 2022-03-30:
添加luci，使用  git clone -b lede https://github.com/pymumu/luci-app-smartdns.git package/applications/luci-app-smartdns
然后可以更新lean的内置的smartdns版本
sed -i 's/1.2021.35/2022.03.02/g' feeds/packages/net/smartdns/Makefile
sed -i 's/f50e4dd0813da9300580f7188e44ed72a27ae79c/1fd18601e7d8ac88e8557682be7de3dc56e69105/g' feeds/packages/net/smartdns/Makefile
sed -i 's/^PKG_MIRROR_HASH/#&/' feeds/packages/net/smartdns/Makefile

eanucc, 2022-03-30:
谢谢，你这个是对的。我先fork，然后git clone luci，完美解决。

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 日期 | 2022年03月30日 21:55 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.******@***.***> |
| 主题 | Re: [coolsnowwolf/lede] 编译lean大固件时无法选择smartdns插件。 (Issue #9151) |

添加luci，使用 git clone -b lede https://github.com/pymumu/luci-app-smartdns.git package/applications/luci-app-smartdns
然后可以更新lean的内置的smartdns版本
sed -i 's/1.2021.35/2022.03.02/g' feeds/packages/net/smartdns/Makefile
sed -i 's/f50e4dd0813da9300580f7188e44ed72a27ae79c/1fd18601e7d8ac88e8557682be7de3dc56e69105/g' feeds/packages/net/smartdns/Makefile
sed -i 's/^PKG_MIRROR_HASH/#&/' feeds/packages/net/smartdns/Makefile

closed 2022-04-28 by eanucc