coolsnowwolf/lede #9153: mac的编译问题 error: too few arguments to function 'of_get_mac_address'

DoveKi, 2022-03-30:
drivers/net/ethernet/ralink/mtk_eth_soc.c: In function 'fe_init':
drivers/net/ethernet/ralink/mtk_eth_soc.c:1405:13: error: too few arguments to function 'of_get_mac_address'
  mac_addr = of_get_mac_address(priv->dev->of_node);
             ^~~~~~~~~~~~~~~~~~
In file included from drivers/net/ethernet/ralink/mtk_eth_soc.c:26:
./include/linux/of_net.h:14:12: note: declared here
 extern int of_get_mac_address(struct device_node *np, u8 *mac);
            ^~~~~~~~~~~~~~~~~~
make[9]: *** [scripts/Makefile.build:262: drivers/net/ethernet/ralink/mtk_eth_soc.o] Error 1
make[8]: *** [scripts/Makefile.build:497: drivers/net/ethernet/ralink] Error 2
make[7]: *** [scripts/Makefile.build:497: drivers/net/ethernet] Error 2
make[6]: *** [scripts/Makefile.build:497: drivers/net] Error 2
make[5]: *** [Makefile:1734: drivers] Error 2
make[5]: Leaving directory '/Volumes/make/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.187'
make[4]: *** [Makefile:25: /Volumes/make/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.187/.modules] Error 2
make[4]: Leaving directory '/Volumes/make/lede/target/linux/ramips'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/Volumes/make/lede/target/linux'
time: target/linux/compile#8.29#6.54#16.99
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:25: target/linux/compile] Error 1
make[2]: Leaving directory '/Volumes/make/lede'
make[1]: *** [target/Makefile:18: /Volumes/make/lede/staging_dir/target-mipsel_24kc_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/Volumes/make/lede'
make: *** [/Volumes/make/lede/include/toplevel.mk:230: world] Error 2

WYC-2020, 2022-03-30:
最新提交的问题，我说过这个改了有很多驱动不匹配的自己看下吧

liaowubinf, 2022-03-30:
Same with me， 求修复

closed 2022-03-30 by AmadeusGhost