coolsnowwolf/lede #9154: 编译失败，请大佬帮忙看看！

cyzzc, 2022-03-30:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：更新内核，没改配置。

### (2) 路由器型号和固件版本

A：X86软路由5.10

### (3) 详细日志

A：
[image]

cyzzc, 2022-03-30:
[image]
这是单线程的报错。

lxwq, 2022-03-30:
touch /home/zz/lede/staging_dir/target-x86_64_musl/stamp/.gnutls_installed
mkdir -p /home/zz/lede/bin/targets/x86/64/packages /home/zz/lede/build_dir/target-x86_64_musl/gnutls-3.7.4/ipkg-x86_64/libgnutls/CONTROL /home/zz/lede/staging_dir/target-x86_64_musl/pkginfo
install -d -m0755 /home/zz/lede/build_dir/target-x86_64_musl/gnutls-3.7.4/ipkg-x86_64/libgnutls/usr/lib
cp -fpR /home/zz/lede/build_dir/target-x86_64_musl/gnutls-3.7.4/ipkg-install/usr/lib/libgnutls.so.* /home/zz/lede/build_dir/target-x86_64_musl/gnutls-3.7.4/ipkg-x86_64/libgnutls/usr/lib/
find /home/zz/lede/build_dir/target-x86_64_musl/gnutls-3.7.4/ipkg-x86_64/libgnutls -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
Package libgnutls is missing dependencies for the following libraries:
libz.so.1
make[3]: *** [Makefile:275: /home/zz/lede/bin/packages/x86_64/packages/libgnutls_3.7.4-2_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/zz/lede/feeds/packages/libs/gnutls'
time: package/feeds/packages/gnutls/compile#253.81#30.86#278.93
    ERROR: package/feeds/packages/gnutls failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/gnutls/compile] Error 1
make[2]: Leaving directory '/home/zz/lede'
make[1]: *** [package/Makefile:110: /home/zz/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/zz/lede'
make: *** [/home/zz/lede/include/toplevel.mk:230：world] 错误 2

WYC-2020, 2022-03-30:
> touch /home/zz/lede/staging_dir/target-x86_64_musl/stamp/.gnutls_installed mkdir -p /home/zz/lede/bin/targets/x86/64/packages [...]

这玩意他们不是修复了嘛，更新package仓库看下

breakings, 2022-03-30:
> 这玩意他们不是修复了嘛，更新package仓库看下

得加一个` --without-zlib`

WYC-2020, 2022-03-30:
> [image] 这是单线程的报错。

要修正patch了，最近提交改动太多，可能后面的patch位置不对了

WYC-2020, 2022-03-30:
> 得加一个` --without-zlib`

不清楚懒得折腾，等我遇到了在说

lxwq, 2022-03-30:
> 得加一个` --without-zlib`

已经都更新到最新了,make clean后还是报这个错,
这个` --without-zlib`加到哪?

breakings, 2022-03-30:
> 已经都更新到最新了,make clean后还是报这个错, 这个` --without-zlib`加到哪?

https://github.com/openwrt/packages/commit/549fe0cd7be98ad56a448f6a788ff937080a5253

lxwq, 2022-03-30:
> > > > > touch /home/zz/lede/staging_dir/target-x86_64_musl/stamp/.gnutls_installed mkdir -p /home/zz/lede/bin/targets/x86/64/packages /home/zz/lede/build_dir/target-x86_64_musl/gnutls-3.7.4/ipkg-x86_64/libgnutls/CONTROL /home/zz/lede/staging_dir/target-x86_64_musl/pkginfo install -d -m0755 /home/zz/lede/build_dir/target-x86_64_musl/gnutls-3.7.4/ipkg-x86_64/libgnutls/usr/lib cp -fpR /home/zz/lede/build_dir/target-x86_64_musl/gnutls-3.7.4/ipkg-install/usr/lib/libgnutls.so.* /home/zz/lede/build_dir/target-x86_64_musl/gnutls-3.7.4/ipkg-x86_64/libgnutls/usr/lib/ find /home/zz/lede/build_dir/target-x86_64_musl/gnutls-3.7.4/ipkg-x86_64/libgnutls -name 'CVS' -o -name '.svn' -o -name '.#_' -o -name '_~'| xargs -r rm -rf Package libgnutls is missing dependencies for the following libraries: libz.so.1 make[3]: *** [Makefile:275: /home/zz/lede/bin/packages/x86_64/packages/libgnutls_3.7.4-2_x86_64.ipk] Error 1 make[3]: Leaving directory '/home/zz/lede/feeds/packages/libs/gnutls' time: package/feeds/packages/gnutls/compile#253.81#30.86#278.93 ERROR: package/feeds/packages/gnutls failed to build. make[2]: *** [package/Makefile:116: package/feeds/packages/gnutls/compile] Error 1 make[2]: Leaving directory '/home/zz/lede' make[1]: *** [package/Makefile:110: /home/zz/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2 make[1]: Leaving directory '/home/zz/lede' make: *** [/home/zz/lede/include/toplevel.mk:230：world] 错误 2
> > > > 
> > > > 
> > > > 这玩意他们不是修复了嘛，更新package仓库看下
> > > 
> > > 
> > > 得加一个` --without-zlib`
> > 
> > 
> > 已经都更新到最新了,make clean后还是报这个错, 这个` --without-zlib`加到哪?
> 
> [openwrt/packages@549fe0c](https://github.com/openwrt/packages/commit/549fe0cd7be98ad56a448f6a788ff937080a5253)

三口,

cyzzc, 2022-03-30:
> > [image]这是单线程的报错。
> 
> 要修改补丁了，最近提交了一个位置，可能后面的补丁不对了

大佬，能指点一下怎么修改吗，万分感谢！

breakings, 2022-03-30:
> > > [image]这是单线程的报错。
> > 
> > 
> > 要修改补丁了，最近提交了一个位置，可能后面的补丁不对了
> 
> 大佬，能指点一下怎么修改吗，万分感谢！

`--- a/net/dsa/dsa2.c
+++ b/net/dsa/dsa2.c
@@ -14,6 +14,7 @@
 #include <linux/rtnetlink.h>
 #include <linux/of.h>
 #include <linux/of_net.h>
+#include <linux/of_platform.h>
 #include <net/devlink.h>
 
 #include "dsa_priv.h"
@@ -288,6 +289,7 @@
 
 		break;
 	case DSA_PORT_TYPE_USER:
+		of_platform_device_create(dp->dn, NULL, NULL);
 		of_get_mac_address(dp->dn, dp->mac);
 		err = dsa_slave_create(dp);
 		if (err)`

cyzzc, 2022-03-30:
> 

虽然看不懂，但还是感谢大佬的回复！

cyzzc, 2022-03-30:
> > > > [image]这是单线程的报错。
> > > 
> > > 
> > > 要修改了，最近提交了一个位置，可能后面的修改不正确了
> > 
> > 
> > 大佬，能指点一下怎么修改吗，万分感谢！
> 
> `--- a/net/dsa/dsa2.c +++ b/net/dsa/dsa2.c @@ -14,6 +14,7 @@ #include <linux/rtnetlink.h> #include <linux /of.h> #include <linux/of_net.h> +#include <linux/of_platform.h> #include <net/devlink.h>
> 
> #include "dsa_priv.h" @@ -288,6 +289,7 @@
> 
> ```
> 	break;
> case DSA_PORT_TYPE_USER:
> ```
> 
> * ```
>     of_platform_device_create(dp->dn, NULL, NULL);
>     of_get_mac_address(dp->dn, dp->mac);
>     err = dsa_slave_create(dp);
>     if (err)`
>   ```

虽然看不懂，但还是感谢大佬的回复！

avrileader, 2022-03-30:
解决了没？同样的问题

jvesky, 2022-03-30:
同样的问题，云编译和本地编译都出错。

hgh188, 2022-03-30:
一样的问题

aiastia, 2022-03-30:
云编译失败……

cyzzc, 2022-03-30:
编译失败，放弃了。最近运气一般。都是遇到失败。上次就遇到701补丁的事。这次又失败。等过4月份再来看看能不能成功。

kissyouhunter, 2022-03-30:
> `--- a/net/dsa/dsa2.c +++ b/net/dsa/dsa2.c @@ -14,6 +14,7 @@ #include <linux/rtnetlink.h> #include <linux/of.h> #include <linux/of_net.h> +#include <linux/of_platform.h> #include <net/devlink.h> [...]

大佬NB 
完美解决

breakings, 2022-03-30:
> 大佬NB 完美解决
[781-dsa-register-every-port-with-of_platform.zip](https://github.com/coolsnowwolf/lede/files/8380173/781-dsa-register-every-port-with-of_platform.zip)
我也就简单粗爆修改了一下 反正是能过了

closed 2022-03-30 by AmadeusGhost