coolsnowwolf/lede #9156: ``` go118.go:5:13: cannot use "quic-go doesn't build on Go 1.18 yet." ```，``` helloworld/v2ray-plugin ``` 编译失败。

signigelchan, 2022-03-30:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
编译 ``` v2ray-plugin ``` 过程，提示 ``` quic-go doesn't build on Go 1.18 yet ```，导致编译失败。
A：

### (2) 路由器型号和固件版本
新路由3 - Newifi-D2
A：

### (3) 详细日志
```
make[3]: Entering directory '/home/chanworks/Developer/lede/feeds/helloworld/v2ray-plugin'
mkdir -p /home/chanworks/Developer/lede/dl
SHELL= flock /home/chanworks/Developer/lede/tmp/.v2ray-plugin-4.44.0.tar.gz.flock -c '          /home/chanworks/Developer/lede/scripts/download.pl "/home/chanworks/Developer/lede/dl" "v2ray-plugin-4.44.0.tar.gz" "f94e8efe879a5300871dd01faec7d6635baaef4dbd39d83f4362802f716d386e" "" "https://codeload.github.com/teddysun/v2ray-plugin/tar.gz/v4.44.0?"    '
rm -f /home/chanworks/Developer/lede/build_dir/target-mipsel_24kc_musl/v2ray-plugin-4.44.0/.built
touch /home/chanworks/Developer/lede/build_dir/target-mipsel_24kc_musl/v2ray-plugin-4.44.0/.built_check
CONFIG_GOLANG_MOD_CACHE_WORLD_READABLE="" GO_BUILD_CACHE_DIR="/home/chanworks/Developer/lede/tmp/go-build" GO_MOD_CACHE_DIR="/home/chanworks/Developer/lede/dl/go-mod-cache" GO_MOD_ARGS="-modcacherw" GO_PKG="github.com/shadowsocks/v2ray-plugin" GO_INSTALL_EXTRA="" GO_INSTALL_ALL="" GO_SOURCE_ONLY="" GO_BUILD_PKG="github.com/shadowsocks/v2ray-plugin/..." GO_EXCLUDES="" GO_GO_GENERATE="" GO_INSTALL_BIN_PATH="/usr/bin" BUILD_DIR="/home/chanworks/Developer/lede/build_dir/target-mipsel_24kc_musl/v2ray-plugin-4.44.0" GO_BUILD_DIR="/home/chanworks/Developer/lede/build_dir/target-mipsel_24kc_musl/v2ray-plugin-4.44.0/.go_work/build" GO_BUILD_BIN_DIR="/home/chanworks/Developer/lede/build_dir/target-mipsel_24kc_musl/v2ray-plugin-4.44.0/.go_work/build/bin/linux_mipsle" GO_BUILD_DEPENDS_PATH="/usr/share/gocode" GO_BUILD_DEPENDS_SRC="/home/chanworks/Developer/lede/staging_dir/target-mipsel_24kc_musl/usr/share/gocode/src" GOOS="linux" GOARCH="mipsle" GO386="" GOAMD64="" GOARM="" GOMIPS="softfloat" GOMIPS64="" GOPPC64="" CGO_ENABLED=1 CC="mipsel-openwrt-linux-musl-gcc" CXX="mipsel-openwrt-linux-musl-g++" CGO_CFLAGS="-Os -pipe -mno-branch-likely -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -fmacro-prefix-map=/home/chanworks/Developer/lede/build_dir/target-mipsel_24kc_musl/v2ray-plugin-4.44.0=v2ray-plugin-4.44.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_CPPFLAGS="-I/home/chanworks/Developer/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/usr/include -I/home/chanworks/Developer/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/include/fortify -I/home/chanworks/Developer/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/include" CGO_CXXFLAGS="-Os -pipe -mno-branch-likely -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -fmacro-prefix-map=/home/chanworks/Developer/lede/build_dir/target-mipsel_24kc_musl/v2ray-plugin-4.44.0=v2ray-plugin-4.44.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_LDFLAGS="-L/home/chanworks/Developer/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/usr/lib -L/home/chanworks/Developer/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/lib -znow -zrelro" GOPATH="/home/chanworks/Developer/lede/build_dir/target-mipsel_24kc_musl/v2ray-plugin-4.44.0/.go_work/build" GOCACHE="/home/chanworks/Developer/lede/tmp/go-build" GOMODCACHE="/home/chanworks/Developer/lede/dl/go-mod-cache" GOENV=off /usr/bin/env bash /home/chanworks/Developer/lede/feeds/packages/lang/golang//golang-build.sh build -v -trimpath -ldflags "all=-buildid '1648650686' -linkmode external -extldflags '-L/home/chanworks/Developer/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/usr/lib -L/home/chanworks/Developer/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/lib -Wl,-z,now -Wl,-z,relro'"     -installsuffix "softfloat"   -ldflags "-s -w -X main.VERSION=v4.44.0 -buildid '1648650686' -linkmode external -extldflags '-L/home/chanworks/Developer/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/usr/lib -L/home/chanworks/Developer/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/lib -Wl,-z,now -Wl,-z,relro'"
Finding targets

Building targets
github.com/lucas-clemente/quic-go/internal/qtls
# github.com/lucas-clemente/quic-go/internal/qtls
../../../../../dl/go-mod-cache/github.com/lucas-clemente/quic-go@v0.23.0/internal/qtls/go118.go:5:13: cannot use "quic-go doesn't build on Go 1.18 yet." (untyped string constant) as int value in variable declaration

make[3]: *** [Makefile:71: /home/chanworks/Developer/lede/build_dir/target-mipsel_24kc_musl/v2ray-plugin-4.44.0/.built] Error 2
make[3]: Leaving directory '/home/chanworks/Developer/lede/feeds/helloworld/v2ray-plugin'
time: package/feeds/helloworld/v2ray-plugin/compile#0.67#0.25#0.43
    ERROR: package/feeds/helloworld/v2ray-plugin failed to build.
make[2]: *** [package/Makefile:116: package/feeds/helloworld/v2ray-plugin/compile] Error 1
make[2]: Leaving directory '/home/chanworks/Developer/lede'
make[1]: *** [package/Makefile:110: /home/chanworks/Developer/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/chanworks/Developer/lede'
make: *** [/home/chanworks/Developer/lede/include/toplevel.mk:230: world] Error 2
```
A：

Beginner-Go, 2022-03-30:
应该是 `v2ray-plugin` 还未适配go1.18 我去上游反馈下

signigelchan, 2022-03-31:
Same as #https://github.com/fw876/helloworld/issues/836 , so close this issue.

closed 2022-03-31 by signigelchan