coolsnowwolf/lede #9160: Netgear R7800 failed to compile since 29th of Mar.

loydcheung, 2022-03-31:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：Netgear R7800

### (3) 详细日志

A：touch /home/loyd/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/stamp/.kmod-qca-mcs_installed
removed '/home/loyd/lede/bin/targets/ipq806x/generic/packages/kmod-qca-mcs_5.4.187+31f5cd4b-1_arm_cortex-a15_neon-vfpv4.ipk'
mkdir -p /home/loyd/lede/bin/targets/ipq806x/generic/packages /home/loyd/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/qca-mcs-31f5cd4b/ipkg-arm_cortex-a15_neon-vfpv4/kmod-qca-mcs/CONTROL /home/loyd/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo
mkdir -p /home/loyd/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/qca-mcs-31f5cd4b/ipkg-arm_cortex-a15_neon-vfpv4/kmod-qca-mcs/etc/modules.d; ( echo "qca-mcs"; ) > /home/loyd/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/qca-mcs-31f5cd4b/ipkg-arm_cortex-a15_neon-vfpv4/kmod-qca-mcs/etc/modules.d/41-qca-mcs; 
find /home/loyd/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/qca-mcs-31f5cd4b/ipkg-arm_cortex-a15_neon-vfpv4/kmod-qca-mcs -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
export CROSS="arm-openwrt-linux-muslgnueabi-"  NO_RENAME=1 ; NM="arm-openwrt-linux-muslgnueabi-nm" STRIP="/home/loyd/lede/staging_dir/host/bin/sstrip -z" STRIP_KMOD="/home/loyd/lede/scripts/strip-kmod.sh" PATCHELF="/home/loyd/lede/staging_dir/host/bin/patchelf" /home/loyd/lede/scripts/rstrip.sh /home/loyd/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/qca-mcs-31f5cd4b/ipkg-arm_cortex-a15_neon-vfpv4/kmod-qca-mcs
rstrip.sh: /home/loyd/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/qca-mcs-31f5cd4b/ipkg-arm_cortex-a15_neon-vfpv4/kmod-qca-mcs/lib/modules/5.4.187/qca-mcs.ko: relocatable
(cd /home/loyd/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/qca-mcs-31f5cd4b/ipkg-arm_cortex-a15_neon-vfpv4/kmod-qca-mcs/CONTROL; ( echo "$CONTROL"; printf "Description: "; echo "$DESCRIPTION" | sed -e 's,^[[:space:]]*, ,g'; ) > control; chmod 644 control; ( echo "#!/bin/sh"; echo "[ \"\${IPKG_NO_SCRIPT}\" = \"1\" ] && exit 0"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_postinst \$0 \$@"; ) > postinst; ( echo "#!/bin/sh"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_prerm \$0 \$@"; ) > prerm; chmod 0755 postinst prerm;  )
install -d -m0755 /home/loyd/lede/bin/targets/ipq806x/generic/packages
/home/loyd/lede/staging_dir/host/bin/fakeroot /home/loyd/lede/scripts/ipkg-build -m "" /home/loyd/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/qca-mcs-31f5cd4b/ipkg-arm_cortex-a15_neon-vfpv4/kmod-qca-mcs /home/loyd/lede/bin/targets/ipq806x/generic/packages
make[2]: Leaving directory '/home/loyd/lede'
Packaged contents of /home/loyd/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/qca-mcs-31f5cd4b/ipkg-arm_cortex-a15_neon-vfpv4/kmod-qca-mcs into /home/loyd/lede/bin/targets/ipq806x/generic/packages/kmod-qca-mcs_5.4.187+31f5cd4b-1_arm_cortex-a15_neon-vfpv4.ipk
echo "kmod-qca-mcs" >> /home/loyd/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/qca-mcs.default.install
rm -rf /home/loyd/lede/tmp/stage-qca-mcs
mkdir -p /home/loyd/lede/tmp/stage-qca-mcs/host /home/loyd/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/packages
install -d -m0755 /home/loyd/lede/tmp/stage-qca-mcs/usr/include/qca-mcs
cp -fpR /home/loyd/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/qca-mcs-31f5cd4b/mc_api.h /home/loyd/lede/tmp/stage-qca-mcs/usr/include/qca-mcs;  cp -fpR /home/loyd/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/qca-mcs-31f5cd4b/mc_ecm.h /home/loyd/lede/tmp/stage-qca-mcs/usr/include/qca-mcs;
cp -fpR /home/loyd/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/qca-mcs-31f5cd4b/mc_api.h /home/loyd/lede/tmp/stage-qca-mcs/usr/include/;  cp -fpR /home/loyd/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/qca-mcs-31f5cd4b/mc_ecm.h /home/loyd/lede/tmp/stage-qca-mcs/usr/include/;
find /home/loyd/lede/tmp/stage-qca-mcs -name '*.la' | xargs -r rm -f; 
if [ -f /home/loyd/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/packages/qca-mcs.list ]; then /home/loyd/lede/scripts/clean-package.sh "/home/loyd/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/packages/qca-mcs.list" "/home/loyd/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi"; fi
if [ -d /home/loyd/lede/tmp/stage-qca-mcs ]; then (cd /home/loyd/lede/tmp/stage-qca-mcs; find ./ > /home/loyd/lede/tmp/stage-qca-mcs.files); 	SHELL= flock /home/loyd/lede/tmp/.staging-dir.flock -c ' mv /home/loyd/lede/tmp/stage-qca-mcs.files /home/loyd/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/packages/qca-mcs.list && cp -fpR /home/loyd/lede/tmp/stage-qca-mcs/* /home/loyd/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/; '; fi
rm -rf /home/loyd/lede/tmp/stage-qca-mcs
touch /home/loyd/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.qca-mcs_installed
make[3]: Leaving directory '/home/loyd/lede/package/qca/nss/qca-mcs'
time: package/qca/nss/qca-mcs/compile#4.36#0.73#5.34
make[2]: Leaving directory '/home/loyd/lede'
make[1]: *** [package/Makefile:110: /home/loyd/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/loyd/lede'
make: *** [/home/loyd/lede/include/toplevel.mk:230: world] Error 2

loydcheung, 2022-03-31:
Tried below, but it still doesn't work:
1. make clean
2. make dirclean
3. delete the folder /lede/dl
4. compile with default menuconfig settings

coolsnowwolf, 2022-03-31:
fixed

closed 2022-03-31 by coolsnowwolf

loydcheung, 2022-03-31:
> fixed

awesome, finger crossed!

loydcheung, 2022-03-31:
> fixed

Sorry boss, while the good news is I compiled successfully thanking to your fix, together there came a bad news: my router keep endless rebooting after i flashed that ROM. no TTL codes, sorry. 
I also tried flashing via TFTP, it still didn't work, how shall we fix that?

loydcheung, 2022-04-01:
> fixed

update: compiled, updated and booted, but both 2.4G and 5G Wi-Fi failed to start. And when I looked into the wifi settings, the 'operating freq' fields are all blank and nothing in the drop-down menu. I assume that should be wifi driver related. @coolsnowwolf  
[image]