coolsnowwolf/lede #9165: X86 64位tinc 无法编译，而32位的可以编译成功 望哪位老大指点迷津！

zhg-em, 2022-03-31:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：X86 64位tinc 无法编译，而32位的可以编译成功

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

WYC-2020, 2022-03-31:
日志都没得鬼给你解答

zhg-em, 2022-03-31:
> 日志都没得鬼给你解答

不好意思，老大，现在给您补上：

time: package/feeds/packages/strongswan/compile#0.85#0.38#1.09
make[3]: Entering directory '/home/xyz/lede/feeds/packages/net/tinc'
mkdir -p /home/xyz/lede/dl
SHELL= flock /home/xyz/lede/tmp/.tinc-1.1-git-3ee0d5dd.tar.gz.flock -c '        /home/xyz/lede/scripts/download.pl "/home/xyz/lede/dl" "tinc-1.1-git-3ee0d5dd.tar.gz" "fe57dcce5c17ec2cea96fefc69c5af977df7d457f1ae7c4775e00a21878a2c19" "" || (     echo "Checking out files from the git repository..."; mkdir -p /home/xyz/lede/tmp/dl && cd /home/xyz/lede/tmp/dl && rm -rf tinc-1.1-git && [ \! -d tinc-1.1-git ] && git clone  http://tinc-vpn.org/git/tinc tinc-1.1-git && (cd tinc-1.1-git && git checkout 3ee0d5dd && git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=`cd tinc-1.1-git && git log -1 --format='\''@%ct'\''` && rm -rf tinc-1.1-git/.git &&        tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c tinc-1.1-git |  gzip -nc > /home/xyz/lede/tmp/dl/tinc-1.1-git-3ee0d5dd.tar.gz && mv /home/xyz/lede/tmp/dl/tinc-1.1-git-3ee0d5dd.tar.gz /home/xyz/lede/dl/ && rm -rf tinc-1.1-git;  )    '
rm -f /home/xyz/lede/build_dir/target-x86_64_musl/tinc-1.1-git/.built
touch /home/xyz/lede/build_dir/target-x86_64_musl/tinc-1.1-git/.built_check
CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/xyz/lede/build_dir/target-x86_64_musl/tinc-1.1-git=tinc-1.1-git -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -std=gnu99  -I/home/xyz/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/home/xyz/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/home/xyz/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/xyz/lede/build_dir/target-x86_64_musl/tinc-1.1-git=tinc-1.1-git -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -std=gnu99  -I/home/xyz/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/home/xyz/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/home/xyz/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include " LDFLAGS="-L/home/xyz/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/xyz/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -znow -zrelro " make  -C /home/xyz/lede/build_dir/target-x86_64_musl/tinc-1.1-git/. AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/xyz/lede/build_dir/target-x86_64_musl/tinc-1.1-git=tinc-1.1-git -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -std=gnu99" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CROSS="x86_64-openwrt-linux-musl-" ARCH="x86_64" ;
make[4]: Entering directory '/home/xyz/lede/build_dir/target-x86_64_musl/tinc-1.1-git'
Makefile:801: *** missing separator.  Stop.
make[4]: Leaving directory '/home/xyz/lede/build_dir/target-x86_64_musl/tinc-1.1-git'
make[3]: *** [Makefile:78: /home/xyz/lede/build_dir/target-x86_64_musl/tinc-1.1-git/.built] Error 2
make[3]: Leaving directory '/home/xyz/lede/feeds/packages/net/tinc'
time: package/feeds/packages/tinc/compile#0.26#0.06#0.30
    ERROR: package/feeds/packages/tinc failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/tinc/compile] Error 1
make[2]: Leaving directory '/home/xyz/lede'
make[1]: *** [package/Makefile:110: /home/xyz/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/xyz/lede'
make: *** [/home/xyz/lede/include/toplevel.mk:230：world] 错误 2

zhg-em, 2022-03-31:
> 更新feeds 再试试

谢谢！今天晚上更新后可以了！

closed 2022-03-31 by coolsnowwolf