coolsnowwolf/lede #9169: X86架构，open-vm-tools 依赖的glib2编译失败

HoldOnBro, 2022-04-01:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：
open-vm-tools 依赖的glib2编译失败

### (2) 路由器型号和固件版本

A：
X86

### (3) 详细日志

A：
`FAILED: glib/gtester
gcc  -o glib/gtester glib/gtester.p/gtester.c.o -L/home/xyz/lede/staging_dir/host/lib -L/home/xyz/lede/staging_dir/hostpkg/lib -L/home/xyz/lede/staging_dir/target-x86_64_musl/host/lib -Wl,--as-needed -Wl,--no-undefined -Wl,-rpath,/home/xyz/lede/staging_dir/hostpkg/lib -Wl,-rpath-link,/home/xyz/lede/staging_dir/hostpkg/lib -Wl,--start-group glib/libglib-2.0.a -lintl /home/xyz/lede/staging_dir/hostpkg/lib/libpcre.so -pthread -lintl -liconv -lm -Wl,--end-group
/usr/bin/ld: glib/libglib-2.0.a(ggettext.c.o): in function `_g_dgettext_should_translate':
/home/xyz/lede/build_dir/hostpkg/glib-2.70.4/openwrt-build/../glib/ggettext.c:315: undefined reference to `libintl_textdomain'
/usr/bin/ld: /home/xyz/lede/build_dir/hostpkg/glib-2.70.4/openwrt-build/../glib/ggettext.c:316: undefined reference to `libintl_gettext'
/usr/bin/ld: glib/libglib-2.0.a(ggettext.c.o): in function `ensure_gettext_initialized':
/home/xyz/lede/build_dir/hostpkg/glib-2.70.4/openwrt-build/../glib/ggettext.c:107: undefined reference to `libintl_bindtextdomain'
/usr/bin/ld: /home/xyz/lede/build_dir/hostpkg/glib-2.70.4/openwrt-build/../glib/ggettext.c:110: undefined reference to `libintl_bind_textdomain_codeset'
/usr/bin/ld: glib/libglib-2.0.a(ggettext.c.o): in function `g_dgettext':
/home/xyz/lede/build_dir/hostpkg/glib-2.70.4/openwrt-build/../glib/ggettext.c:402: undefined reference to `libintl_dgettext'
/usr/bin/ld: glib/libglib-2.0.a(ggettext.c.o): in function `g_dcgettext':
/home/xyz/lede/build_dir/hostpkg/glib-2.70.4/openwrt-build/../glib/ggettext.c:429: undefined reference to `libintl_dcgettext'
/usr/bin/ld: glib/libglib-2.0.a(ggettext.c.o): in function `g_dngettext':
/home/xyz/lede/build_dir/hostpkg/glib-2.70.4/openwrt-build/../glib/ggettext.c:460: undefined reference to `libintl_dngettext'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
make[3]: *** [Makefile:129: /home/xyz/lede/build_dir/hostpkg/glib-2.70.4/.built] Error 1
make[3]: Leaving directory '/home/xyz/lede/feeds/packages/libs/glib2'
time: package/feeds/packages/glib2/host-compile#0.25#0.12#0.40
    ERROR: package/feeds/packages/glib2 [host] failed to build.
make[2]: *** [package/Makefile:114: package/feeds/packages/glib2/host/compile] Error 1
make[2]: Leaving directory '/home/xyz/lede'
make[1]: *** [package/Makefile:110: /home/xyz/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/xyz/lede'
make: *** [/home/xyz/lede/include/toplevel.mk:230: world] Error 2`

Beginner-Go, 2022-04-02:
更新feeds 再试试

HoldOnBro, 2022-04-02:
> 更新feeds 再试试

还是一样……

```
[100/455] Linking target glib/gtester
FAILED: glib/gtester
gcc  -o glib/gtester glib/gtester.p/gtester.c.o -L/home/xyz/lede/staging_dir/host/lib -L/home/xyz/lede/staging_dir/hostpkg/lib -L/home/xyz/lede/staging_dir/target-x86_64_musl/host/lib -Wl,--as-needed -Wl,--no-undefined -Wl,-rpath,/home/xyz/lede/staging_dir/hostpkg/lib -Wl,--start-group glib/libglib-2.0.a /home/xyz/lede/staging_dir/hostpkg/lib/libpcre.so -pthread -liconv -lm -Wl,--end-group
/usr/bin/ld: glib/libglib-2.0.a(ggettext.c.o): in function `_g_dgettext_should_translate':
/home/xyz/lede/build_dir/hostpkg/glib-2.70.4/openwrt-build/../glib/ggettext.c:315: undefined reference to `libintl_textdomain'
/usr/bin/ld: /home/xyz/lede/build_dir/hostpkg/glib-2.70.4/openwrt-build/../glib/ggettext.c:316: undefined reference to `libintl_gettext'
/usr/bin/ld: glib/libglib-2.0.a(ggettext.c.o): in function `ensure_gettext_initialized':
/home/xyz/lede/build_dir/hostpkg/glib-2.70.4/openwrt-build/../glib/ggettext.c:107: undefined reference to `libintl_bindtextdomain'
/usr/bin/ld: /home/xyz/lede/build_dir/hostpkg/glib-2.70.4/openwrt-build/../glib/ggettext.c:110: undefined reference to `libintl_bind_textdomain_codeset'
/usr/bin/ld: glib/libglib-2.0.a(ggettext.c.o): in function `g_dgettext':
/home/xyz/lede/build_dir/hostpkg/glib-2.70.4/openwrt-build/../glib/ggettext.c:402: undefined reference to `libintl_dgettext'
/usr/bin/ld: glib/libglib-2.0.a(ggettext.c.o): in function `g_dcgettext':
/home/xyz/lede/build_dir/hostpkg/glib-2.70.4/openwrt-build/../glib/ggettext.c:429: undefined reference to `libintl_dcgettext'
/usr/bin/ld: glib/libglib-2.0.a(ggettext.c.o): in function `g_dngettext':
/home/xyz/lede/build_dir/hostpkg/glib-2.70.4/openwrt-build/../glib/ggettext.c:460: undefined reference to `libintl_dngettext'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
make[3]: *** [Makefile:128: /home/xyz/lede/build_dir/hostpkg/glib-2.70.4/.built] Error 1
make[3]: Leaving directory '/home/xyz/lede/feeds/packages/libs/glib2'
time: package/feeds/packages/glib2/host-compile#24.29#3.30#28.37
    ERROR: package/feeds/packages/glib2 [host] failed to build.
make[2]: *** [package/Makefile:114: package/feeds/packages/glib2/host/compile] Error 1
make[2]: Leaving directory '/home/xyz/lede'
make[1]: *** [package/Makefile:110: /home/xyz/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/xyz/lede'
make: *** [/home/xyz/lede/include/toplevel.mk:230: world] Error 2
```

WYC-2020, 2022-04-02:
等有空看下

mumuou, 2022-04-03:
同样遇到这个问题。

martin-xj, 2022-04-03:
同样编译失败

RodmanWang, 2022-04-03:
编译失败+1

fishbless, 2022-04-04:
编译失败+1

abctel, 2022-04-05:
编译失败+1

SakuraFallingMad, 2022-04-05:
编译失败+1

WYC-2020, 2022-04-05:
我这边今天测试没有任何问题呢，这个是解决了吧

martin-xj, 2022-04-05:
> 我这边今天测试没有任何问题呢，这个是解决了吧

又试了一下，还是不行。
mkdir -p /home/test/lede/dl
SHELL= flock /home/test/lede/tmp/.glib-2.70.4.tar.xz.flock -c '  	/home/test/lede/scripts/download.pl "/home/test/lede/dl" "glib-2.70.4.tar.xz" "ab3d176f3115dcc4e5d02db795984e04e4f4b48d836252e23e8c468e9d423c33" "" "@GNOME/glib/2.70"    '
MAKEFLAGS="" /home/test/lede/staging_dir/host/bin/ninja  -j1 -C /home/test/lede/build_dir/hostpkg/glib-2.70.4/openwrt-build 
ninja: Entering directory `/home/test/lede/build_dir/hostpkg/glib-2.70.4/openwrt-build'
[1/356] Linking target glib/gtester
FAILED: glib/gtester 
gcc  -o glib/gtester glib/gtester.p/gtester.c.o -L/home/test/lede/staging_dir/host/lib -L/home/test/lede/staging_dir/hostpkg/lib -L/home/test/lede/staging_dir/target-x86_64_musl/host/lib -Wl,--as-needed -Wl,--no-undefined -Wl,-rpath,/home/test/lede/staging_dir/hostpkg/lib -Wl,--start-group glib/libglib-2.0.a /home/test/lede/staging_dir/hostpkg/lib/libpcre.so -pthread -liconv -lm -Wl,--end-group
/usr/bin/ld: glib/libglib-2.0.a(ggettext.c.o): in function `_g_dgettext_should_translate':
/home/test/lede/build_dir/hostpkg/glib-2.70.4/openwrt-build/../glib/ggettext.c:315: undefined reference to `libintl_textdomain'
/usr/bin/ld: /home/test/lede/build_dir/hostpkg/glib-2.70.4/openwrt-build/../glib/ggettext.c:316: undefined reference to `libintl_gettext'
/usr/bin/ld: glib/libglib-2.0.a(ggettext.c.o): in function `ensure_gettext_initialized':
/home/test/lede/build_dir/hostpkg/glib-2.70.4/openwrt-build/../glib/ggettext.c:107: undefined reference to `libintl_bindtextdomain'
/usr/bin/ld: /home/test/lede/build_dir/hostpkg/glib-2.70.4/openwrt-build/../glib/ggettext.c:110: undefined reference to `libintl_bind_textdomain_codeset'
/usr/bin/ld: glib/libglib-2.0.a(ggettext.c.o): in function `g_dgettext':
/home/test/lede/build_dir/hostpkg/glib-2.70.4/openwrt-build/../glib/ggettext.c:402: undefined reference to `libintl_dgettext'
/usr/bin/ld: glib/libglib-2.0.a(ggettext.c.o): in function `g_dcgettext':
/home/test/lede/build_dir/hostpkg/glib-2.70.4/openwrt-build/../glib/ggettext.c:429: undefined reference to `libintl_dcgettext'
/usr/bin/ld: glib/libglib-2.0.a(ggettext.c.o): in function `g_dngettext':
/home/test/lede/build_dir/hostpkg/glib-2.70.4/openwrt-build/../glib/ggettext.c:460: undefined reference to `libintl_dngettext'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
make[3]: *** [Makefile:128: /home/test/lede/build_dir/hostpkg/glib-2.70.4/.built] Error 1
make[3]: Leaving directory '/home/test/lede/feeds/packages/libs/glib2'
time: package/feeds/packages/glib2/host-compile#0.28#0.16#0.46
    ERROR: package/feeds/packages/glib2 [host] failed to build.
make[2]: *** [package/Makefile:114: package/feeds/packages/glib2/host/compile] Error 1
make[2]: Leaving directory '/home/test/lede'
make[1]: *** [package/Makefile:110: /home/test/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/test/lede'
make: *** [/home/test/lede/include/toplevel.mk:230：world] 错误 2

WYC-2020, 2022-04-05:
> 又试了一下，还是不行。 mkdir -p /home/test/lede/dl SHELL= flock /home/test/lede/tmp/.glib-2.70.4.tar.xz.flock -c ' /home/test/lede/scripts/download.pl "/home/test/lede/dl" "glib-2.70.4.tar.xz" [...]

那不清楚了我用自己仓库没问题，按理和l区别不大才对

Beginner-Go, 2022-04-05:
> 我这边今天测试没有任何问题呢，这个是解决了吧

似乎用host编译才会报错 
`make package/glib2/{clean,compile} V=s` 没有问题

WYC-2020, 2022-04-05:
> > 我这边今天测试没有任何问题呢，这个是解决了吧
> 
> 似乎用host编译才会报错 `make package/glib2/{clean,compile} V=s` 没有问题

管他编译什么版本。反正能编译出open-vm-tools就行，其他的没测试

HoldOnBro, 2022-04-06:
仍然报错
```
[25/380] Linking target glib/gtester
FAILED: glib/gtester
gcc  -o glib/gtester glib/gtester.p/gtester.c.o -L/home/xyz/lede/staging_dir/host/lib -L/home/xyz/lede/staging_dir/hostpkg/lib -L/home/xyz/lede/staging_dir/target-x86_64_musl/host/lib -Wl,--as-needed -Wl,--no-undefined -Wl,-rpath,/home/xyz/lede/staging_dir/hostpkg/lib -Wl,--start-group glib/libglib-2.0.a /home/xyz/lede/staging_dir/hostpkg/lib/libpcre.so -pthread -liconv -lm -Wl,--end-group
/usr/bin/ld: glib/libglib-2.0.a(ggettext.c.o): in function `_g_dgettext_should_translate':
/home/xyz/lede/build_dir/hostpkg/glib-2.70.4/openwrt-build/../glib/ggettext.c:315: undefined reference to `libintl_textdomain'
/usr/bin/ld: /home/xyz/lede/build_dir/hostpkg/glib-2.70.4/openwrt-build/../glib/ggettext.c:316: undefined reference to `libintl_gettext'
/usr/bin/ld: glib/libglib-2.0.a(ggettext.c.o): in function `ensure_gettext_initialized':
/home/xyz/lede/build_dir/hostpkg/glib-2.70.4/openwrt-build/../glib/ggettext.c:107: undefined reference to `libintl_bindtextdomain'
/usr/bin/ld: /home/xyz/lede/build_dir/hostpkg/glib-2.70.4/openwrt-build/../glib/ggettext.c:110: undefined reference to `libintl_bind_textdomain_codeset'
/usr/bin/ld: glib/libglib-2.0.a(ggettext.c.o): in function `g_dgettext':
/home/xyz/lede/build_dir/hostpkg/glib-2.70.4/openwrt-build/../glib/ggettext.c:402: undefined reference to `libintl_dgettext'
/usr/bin/ld: glib/libglib-2.0.a(ggettext.c.o): in function `g_dcgettext':
/home/xyz/lede/build_dir/hostpkg/glib-2.70.4/openwrt-build/../glib/ggettext.c:429: undefined reference to `libintl_dcgettext'
/usr/bin/ld: glib/libglib-2.0.a(ggettext.c.o): in function `g_dngettext':
/home/xyz/lede/build_dir/hostpkg/glib-2.70.4/openwrt-build/../glib/ggettext.c:460: undefined reference to `libintl_dngettext'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
make[3]: *** [Makefile:128: /home/xyz/lede/build_dir/hostpkg/glib-2.70.4/.built] Error 1
make[3]: Leaving directory '/home/xyz/lede/feeds/packages/libs/glib2'
time: package/feeds/packages/glib2/host-compile#6.83#1.06#11.43
    ERROR: package/feeds/packages/glib2 [host] failed to build.
make[2]: *** [package/Makefile:114: package/feeds/packages/glib2/host/compile] Error 1
make[2]: Leaving directory '/home/xyz/lede'
make[1]: *** [package/Makefile:110: /home/xyz/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/xyz/lede'
make: *** [/home/xyz/lede/include/toplevel.mk:230: world] Error 2
```

SakuraFallingMad, 2022-04-06:
我测试是云编译失败，本机正常

martin-xj, 2022-04-06:
试了半天，错误依旧没法编译 通过，希望大佬们关注一下。

WYC-2020, 2022-04-06:
都不能复现如何关注，自己试下万能大法，我这边x86 一切正常，一次性编译过

WYC-2020, 2022-04-06:
> > 我这边今天测试没有任何问题呢，这个是解决了吧
> 
> 似乎用host编译才会报错 `make package/glib2/{clean,compile} V=s` 没有问题

make package/feeds/packages/glib2/{clean,compile} V=s，这要编译也没出现那个问题，怕是遇到错误的那个地方有坑吧，哈哈

SakuraFallingMad, 2022-04-06:
本地编译没问题，好像云编译会碰到libintl的问题，config里添加了compile with full language support，它安装了libintl-full。暂未确认症结。可以试试取消libintl-full

fakeboboliu, 2022-04-06:
@WYC-2020 你测试的编译参数是错的，open-vm-tools 的依赖是 `glib2/host`

因此要测试复现需要使用 `make package/feeds/packages/glib2/host/{clean,compile} V=s`

另外这问题还是比较容易定位的，build 的时候有 `-lintl`，但是丢失了相关 func ref，也没有提示 not found 类错误，那应该就是有相关的二进制文件工作不太正常。

此时删除相关二进制应当是有用的：
`rm staging_dir/host/lib/libintl.* staging_dir/hostpkg/lib/libintl.* staging_dir/target-x86_64_musl/host/lib/libintl.*`

顺便 https://github.com/openwrt/packages/commit/0ce744202ad2a2feaa3b841ce934765416d1339e 有一个 patch 可以打上试一试

WYC-2020, 2022-04-06:
> @WYC-2020 你测试的编译参数是错的，open-vm-tools 的依赖是 `glib2/host`
> 
> 因此要测试复现需要使用 `make package/feeds/packages/glib2/host/{clean,compile} V=s`
> 
> 另外这问题还是比较容易定位的，build 的时候有 `-lintl`，但是丢失了相关 func ref，也没有提示 not found 类错误，那应该就是有相关的二进制文件工作不太正常。
> 
> 此时删除相关二进制，让编译的时候更新一下或者防止混进来的其他来源文件应当是有用的： `rm staging_dir/host/lib/libintl.* staging_dir/hostpkg/lib/libintl.* staging_dir/target-x86_64_musl/host/lib/libintl.*`
> 
> 顺便 [openwrt/packages@0ce7442](https://github.com/openwrt/packages/commit/0ce744202ad2a2feaa3b841ce934765416d1339e) 有一个 patch 可以打上试一试

这只是另外一个人让我这样测试一下，等会我也测试下你这个吧，我整体编译过也没遇到这个问题，你说的这个patch原来加了的被回退了而已，你可以看仓库的提交记录

HoldOnBro, 2022-04-07:
> 本地编译没问题，好像云编译会碰到libintl的问题，config里添加了compile with full language support，它安装了libintl-full。暂未确认症结。可以试试取消libintl-full
> 
> 尝试了下，依然云编译失败，本地环境正常，这坑过不去了

我跟你是反的，云编译可以通过了，本地还是不行

WYC-2020, 2022-04-07:
> @WYC-2020 你测试的编译参数是错的，open-vm-tools 的依赖是 `glib2/host`
>
> 因此要测试复现需要使用 `make package/feeds/packages/glib2/host/{clean,compile} V=s`
>
> 另外这问题还是比较容易定位的，build 的时候有 `-lintl`，但是丢失了相关 func ref，也没有提示 not found [...]

刚测试了，make package/feeds/packages/glib2/host/{clean,compile} V=s 这样也没问题，哈哈，你们怕是见鬼了，因为我平时也不编译这个open-vm-tools，所以云编译我这边没测试，本地 反正我是测试了 不管怎么编译都不会出现你们这些问题，奇怪

HoldOnBro, 2022-04-07:
> 刚测试了，make package/feeds/packages/glib2/host/{clean,compile} V=s 这样也没问题，哈哈，你们怕是见鬼了，因为我平时也不编译这个open-vm-tools，所以云编译我这边没测试，本地 反正我是测试了 不管怎么编译都不会出现你们这些问题，奇怪

应该是没问题了，云编译环境是干净的，本地估计脏了得clean

W7721, 2022-04-07:
同样的问题，make distclean之后再删除.config 重新搞就没问题了。

SakuraFallingMad, 2022-04-08:
我云编译一直失败，本地一直成功，头疼

abctel, 2022-04-09:
可以close了，昨晚重新同步了代码之后编译该插件，顺利编译，无错误提示。

closed 2022-04-09 by HoldOnBro