coolsnowwolf/lede #9170: mtk闭源驱动在新内核5.15上编译失败，以及一个粗暴的修复方法

everything411, 2022-04-01:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

我将仓库中的mtk闭源驱动代码放到了openwrt master上去编译发现，当开启高版本内核的时候（kernel 5.15），编译报错`mt/drivers/mt7603e/src/mt7603_wifi/os/linux/rt_linux.c`和`mt/drivers/mt7615d/src/mt_wifi/os/linux/rt_linux.c`找不到函数set_fs和get_fs，经搜索发现新版内核中这两个函数已经被删除。

这两个函数用于调整用户空间和内核空间的界限，在使用vfs_read和vfs_open的时候有用，而现在驱动里已经使用了新版函数kernel_read和kernel_write，感觉应该可以安全的删除这些报错的set_fs和get_fs调用。

此issue供各位大神参考。

### (2) 路由器型号和固件版本

A：红米ac2100

删除所有涉及set_fs和get_fs的调用后，从openwrt主线编译了一版固件，内核5.15.31，目前在我自己的机器上运行似乎是完全正常的，下附编译的结果。

[openwrt-ramips-mt7621-xiaomi_redmi-router-ac2100-squashfs-sysupgrade.zip](https://github.com/coolsnowwolf/lede/files/8395102/openwrt-ramips-mt7621-xiaomi_redmi-router-ac2100-squashfs-sysupgrade.zip)

WYC-2020, 2022-04-01:
那老板内核不就不支持了，加宏最好，或者如果原来的函数不复杂，直接内核打补丁

everything411, 2022-04-01:
> 那老板内核不就不支持了，加宏最好，或者如果原来的函数不复杂，直接内核打补丁

嗯，加宏是最好的，在驱动里面判断内核版本选用不同的方式，我只是验证性的粗暴的删除了这些调用，因为可以确认在高版本内核里没有且不再需要set_fs和get_fs。参考https://stackoverflow.com/a/53917617

Henshufu, 2022-04-03:
楼主有红米ac2100 5.10内核的固件吗，我编译过好几次有线连接都无效。

everything411, 2022-04-03:
> 楼主有红米ac2100 5.10内核的固件吗，我编译过好几次有线连接都无效。

目前openwrt snapshot官方编译默认5.10内核，打开测试版是5.15。 现在从https://downloads.openwrt.org/snapshots/targets/ramips/mt7621/ 下载的就是5.10内核的，不过这里的固件没有luci，也没有各种插件

lean大固件目前默认是5.4，打开测试版是5.10，我现在没有现成的版本，因为我只用来做ap不需要插件，所以我都是从lean大这里复制一份闭源驱动包放到openwrt主线上编译的固件。

Henshufu, 2022-04-03:
> > 楼主有红米ac2100 5.10内核的固件吗，我编译过好几次有线连接都无效。
> 
> 目前openwrt snapshot官方编译默认5.10内核，打开测试版是5.15。 现在从https://downloads.openwrt.org/snapshots/targets/ramips/mt7621/ 下载的就是5.10内核的，不过这里的固件没有luci，也没有各种插件
> 
> lean大固件目前默认是5.4，打开测试版是5.10，我现在没有现成的版本，因为我只用来做ap不需要插件，所以我都是从lean大这里复制一份闭源驱动包放到openwrt主线上编译的固件。
> 
> update: 刚刚发现lean大的最新源码ramips target测试内核已经从5.10升级到5.15了

好的，谢谢！

liwenjie119, 2022-04-09:
尝试编译5.15的cr660x，驱动报错：
/home/liwenjie/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-04-07-506bb060/mt7915/pci.c: In function 'mt7915_pci_probe':
/home/liwenjie/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-04-07-506bb060/mt7915/pci.c:209:43: error: 'PCI_IRQ_ALL_TYPES' undeclared (first use in this function); did you mean 'NUM_RFKILL_TYPES'?
   ret = pci_alloc_irq_vectors(pdev, 1, 1, PCI_IRQ_ALL_TYPES);
                                           ^~~~~~~~~~~~~~~~~
                                           NUM_RFKILL_TYPES
/home/liwenjie/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-04-07-506bb060/mt7915/pci.c:209:43: note: each undeclared identifier is reported only once for each function it appears in
/home/liwenjie/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-04-07-506bb060/mt7915/pci.c:260:3: error: implicit declaration of function 'pci_free_irq_vectors'; did you mean 'pci_alloc_irq_vectors'? [-Werror=implicit-function-declaration]
   pci_free_irq_vectors(pdev);
   ^~~~~~~~~~~~~~~~~~~~
   pci_alloc_irq_vectors
cc1: some warnings being treated as errors
make[6]: *** [scripts/Makefile.build:277: /home/liwenjie/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-04-07-506bb060/mt7915/pci.o] Error 1
make[5]: *** [scripts/Makefile.build:540: /home/liwenjie/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-04-07-506bb060/mt7915] Error 2
make[4]: *** [Makefile:1868: /home/liwenjie/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-04-07-506bb060] Error 2
make[4]: Leaving directory '/home/liwenjie/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.15.32'
make[3]: *** [Makefile:469: /home/liwenjie/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-04-07-506bb060/.built] Error 2
make[3]: Leaving directory '/home/liwenjie/lede/package/kernel/mt76'

everything411, 2022-04-10:
> 尝试编译5.15的cr660x，驱动报错： /home/liwenjie/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-04-07-506bb060/mt7915/pci.c: In function 'mt7915_pci_probe': [...]

这两个标识符均在linux/pci.h中定义，理论上不应该报错才对，不太清楚为什么

https://github.com/openwrt/mt76/blob/506bb0605e3e2fa4c5285e6c8866167691a6d71b/mt7915/pci.c#L9
https://elixir.bootlin.com/linux/v5.15.33/source/include/linux/pci.h#L1501
https://elixir.bootlin.com/linux/v5.15.33/source/include/linux/pci.h#L1544

zixing131, 2022-04-15:
感谢，把get_fs和set_fs删了后不报错了

uniessey, 2022-04-16:
> > 那老板内核不就不支持了，加宏最好，或者如果原来的函数不复杂，直接内核打补丁
> 
> 嗯，加宏是最好的，在驱动里面判断内核版本选用不同的方式，我只是验证性的粗暴的删除了这些调用，因为可以确认在高版本内核里没有且不再需要set_fs和get_fs。参考https://stackoverflow.com/a/53917617

能否麻烦您上传一个删除set_fs和get_fs之后的rt_linux.c，我试着删除与set_fs和get_fs有关的调用，结果问题变得更复杂了，估计我还是技术太差了。

zixing131, 2022-04-16:
> > > 那老板内核不就不支持了，加宏最好，或者如果原来的函数不复杂，直接内核打补丁
> > 
> > 
> > 嗯，加宏是最好的，在驱动里面判断内核版本选用不同的方式，我只是验证性的粗暴的删除了这些调用，因为可以确认在高版本内核里没有且不再需要set_fs和get_fs。参考https://stackoverflow.com/a/53917617
> 
> 能否麻烦您上传一个删除set_fs和get_fs之后的rt_linux.c，我试着删除与set_fs和get_fs有关的调用，结果问题变得更复杂了，估计我还是技术太差了。

注释掉get_fs和set_fs之后，7615还有2个地方找不到函数定义的，改成返回0，然后eip-93也有报错，添加对应的头文件就行了，linux/sha.h改成linux/sha1.h和linux/sha2.h ，还有2个地方头文件找不到，你搜索一下这个结构在哪里定义的把头文件加上就行。

但是，我是红米ac2100，5.15内核编译成功了，刷入进去之后连接不上路由器，5.10的不需要改动就直接可以编译成功，刷5.10的有线lan无法访问，无线可以访问，wan不分配ip。最后还是5.4养老。

liwenjie119, 2022-04-16:
> 
看来这个报错还不是个例，不知道为啥调用出错
https://github.com/coolsnowwolf/lede/issues/9217#issue-1198836546
https://github.com/coolsnowwolf/lede/issues/9231#issue-1199440303

everything411, 2022-04-17:
我确实只修改了get_fs和set_fs，然后在openwrt主线上则可以正常编译和工作。eip-93因为用不到所以我没有放进去编译，上面的问题就不太清楚了
```
everything411@builder $ diff /home/everything411/openwrt/package/mtwifi/mtwifi/drivers/mt7615d/src/mt_wifi/os/linux/rt_linux.c /home/everything411/lede/package/lean/mt/drivers/mt7615d/src/mt_wifi/os/linux/rt_linux.c
868,869c868,869
<               //pOSFSInfo->fs = get_fs();
<               //set_fs(KERNEL_DS);
---
>               pOSFSInfo->fs = get_fs();
>               set_fs(KERNEL_DS);
871c871
<               //set_fs(pOSFSInfo->fs);
---
>               set_fs(pOSFSInfo->fs);
1928,1929c1928,1929
<       //orig_fs = get_fs();
<       //set_fs(KERNEL_DS);
---
>       orig_fs = get_fs();
>       set_fs(KERNEL_DS);
1971c1971
<       //set_fs(orig_fs);
---
>       set_fs(orig_fs);
1994,1995c1994,1995
<       //orig_fs = get_fs();
<       //set_fs(KERNEL_DS);
---
>       orig_fs = get_fs();
>       set_fs(KERNEL_DS);
2038c2038
<       //set_fs(orig_fs);
---
>       set_fs(orig_fs);
2052,2053c2052,2053
<       //orig_fs = get_fs();
<       //set_fs(KERNEL_DS);
---
>       orig_fs = get_fs();
>       set_fs(KERNEL_DS);
2084c2084
<       //set_fs(orig_fs);
---
>       set_fs(orig_fs);
```

uniessey, 2022-04-17:
> 我确实只修改了get_fs和set_fs，然后在openwrt主线上则可以正常编译和工作。eip-93因为用不到所以我没有放进去编译，上面的问题就不太清楚了
>
> ```
> everything411@builder $ diff [...]

谢谢您百忙之中的回复，这非常有用。

everything411, 2022-04-18:
https://github.com/everything411/lede/tree/fix_kernel5.15_mtwifi_build 创建了一个新的分支，我编译测试后会向主仓库发起PR以解决这个问题。mt7612e的代码目前还没有动，对于mt7612e，有地方还没有使用新的kernel_read和kernel_write函数代替老的函数，代码修改量可能比较多，而且我没有mt7612e的设备，无法进行测试

everything411, 2022-04-23:
close this as my PR has been merged

closed 2022-04-23 by everything411