coolsnowwolf/lede #9173: 5.4.188内核和mac80211更新后，R7800的非CT常规无线驱动失效

loydcheung, 2022-04-02:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：具体表现为：路由器能正常启动，但2.4G和5G的无线显示未关联或未开启，设置中工作频率、模式、信道、频宽部分都为空且下拉菜单空白，国家地区的下拉菜单也消失，无法更改和设置。

### (2) 路由器型号和固件版本

A：路由器型号：R7800     
内核版本：5.4.188 
无线驱动：ath10k-firmware-qca9984（不带CT）

### (3) 详细日志

A：

loydcheung, 2022-04-02:
选择不带CT驱动的原因是，经我大量测试，用此驱动，设置发射功率2.4G在14dbm左右、5G在21dbm左右最稳定，几乎不会出现断流，CT驱动会有断流现象，特别是在2.4G频段几乎每天发生。

coolsnowwolf, 2022-04-02:
请提供日志，否则无法分析

coolsnowwolf, 2022-04-02:
[image]
[image]

loydcheung, 2022-04-02:
> [image] [image]

OpenWrt_R7800 login: root
Password: 

BusyBox v1.35.0 (2022-04-01 04:26:18 UTC) built-in shell (ash)

     _________
    /        /\      _    ___ ___  ___
   /  LE    /  \    | |  | __|   \| __|
  /    DE  /    \   | |__| _|| |) | _|
 /________/  LE  \  |____|___|___/|___|
 \        \   DE /
  \    LE  \    /  -------------------------------------------
   \  DE    \  /    OpenWrt SNAPSHOT, r4403+36-a77cc5f1e
    \________\/    -------------------------------------------

root@OpenWrt_R7800:~# logread
Sat Apr  2 19:41:06 2022 kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0
Sat Apr  2 19:41:06 2022 kern.notice kernel: [    0.000000] Linux version 5.4.188 (loyd@loyd-Standard-PC-i440FX-PIIX-1996) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r4386+30-7a7f9d647)) #0 SMP Fri Apr 1 17:15:55 2022
Sat Apr  2 19:41:06 2022 kern.info kernel: [    0.000000] CPU: ARMv7 Processor [512f04d0] revision 0 (ARMv7), cr=10c5787d
Sat Apr  2 19:41:06 2022 kern.info kernel: [    0.000000] CPU: div instructions available: patching division code
Sat Apr  2 19:41:06 2022 kern.info kernel: [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
Sat Apr  2 19:41:06 2022 kern.info kernel: [    0.000000] OF: fdt: Machine model: Netgear Nighthawk X4S R7800
Sat Apr  2 19:41:06 2022 kern.info kernel: [    0.000000] Memory policy: Data cache writealloc
Sat Apr  2 19:41:06 2022 kern.debug kernel: [    0.000000] On node 0 totalpages: 122880
Sat Apr  2 19:41:06 2022 kern.debug kernel: [    0.000000]   Normal zone: 1080 pages used for memmap
Sat Apr  2 19:41:06 2022 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Sat Apr  2 19:41:06 2022 kern.debug kernel: [    0.000000]   Normal zone: 122880 pages, LIFO batch:31
Sat Apr  2 19:41:06 2022 kern.info kernel: [    0.000000] percpu: Embedded 15 pages/cpu s30220 r8192 d23028 u61440
Sat Apr  2 19:41:06 2022 kern.debug kernel: [    0.000000] pcpu-alloc: s30220 r8192 d23028 u61440 alloc=15*4096
Sat Apr  2 19:41:06 2022 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1
Sat Apr  2 19:41:06 2022 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 121800
Sat Apr  2 19:41:06 2022 kern.notice kernel: [    0.000000] Kernel command line:
Sat Apr  2 19:41:06 2022 kern.notice kernel: [    0.000000] Bootloader command line (ignored): console=ttyHSL1,115200n8
Sat Apr  2 19:41:06 2022 kern.info kernel: [    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
Sat Apr  2 19:41:06 2022 kern.info kernel: [    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
Sat Apr  2 19:41:06 2022 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Sat Apr  2 19:41:06 2022 kern.info kernel: [    0.000000] Memory: 474820K/491520K available (6269K kernel code, 198K rwdata, 1572K rodata, 1024K init, 232K bss, 16700K reserved, 0K cma-reserved, 0K highmem)
Sat Apr  2 19:41:06 2022 kern.info kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
Sat Apr  2 19:41:06 2022 kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Sat Apr  2 19:41:06 2022 kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
Sat Apr  2 19:41:06 2022 kern.info kernel: [    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
Sat Apr  2 19:41:06 2022 kern.notice kernel: [    0.000000] random: get_random_bytes called from start_kernel+0x340/0x4ec with crng_init=0
Sat Apr  2 19:41:06 2022 kern.info kernel: [    0.000000] clocksource: dg_timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 305801671480 ns
Sat Apr  2 19:41:06 2022 kern.info kernel: [    0.000006] sched_clock: 32 bits at 6MHz, resolution 160ns, wraps every 343597383600ns
Sat Apr  2 19:41:06 2022 kern.info kernel: [    0.000019] Switching to timer-based delay loop, resolution 160ns
Sat Apr  2 19:41:06 2022 kern.info kernel: [    0.000251] Calibrating delay loop (skipped), value calculated using timer frequency.. 12.50 BogoMIPS (lpj=62500)
Sat Apr  2 19:41:06 2022 kern.info kernel: [    0.000274] pid_max: default: 32768 minimum: 301
Sat Apr  2 19:41:06 2022 kern.info kernel: [    0.000473] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Sat Apr  2 19:41:06 2022 kern.info kernel: [    0.000491] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Sat Apr  2 19:41:06 2022 kern.info kernel: [    0.001546] CPU: Testing write buffer coherency: ok
Sat Apr  2 19:41:06 2022 kern.info kernel: [    0.002684] Setting up static identity map for 0x42300000 - 0x42300060
Sat Apr  2 19:41:06 2022 kern.info kernel: [    0.002858] rcu: Hierarchical SRCU implementation.
Sat Apr  2 19:41:06 2022 kern.info kernel: [    0.003344] smp: Bringing up secondary CPUs ...
Sat Apr  2 19:41:06 2022 kern.info kernel: [    0.005280] smp: Brought up 1 node, 2 CPUs
Sat Apr  2 19:41:06 2022 kern.info kernel: [    0.005298] SMP: Total of 2 processors activated (25.00 BogoMIPS).
Sat Apr  2 19:41:06 2022 kern.info kernel: [    0.005310] CPU: All CPU(s) started in SVC mode.
Sat Apr  2 19:41:06 2022 kern.info kernel: [    0.017143] VFP support v0.3: implementor 51 architecture 64 part 4d variant 2 rev 0
Sat Apr  2 19:41:06 2022 kern.info kernel: [    0.017293] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[... cut ...]
Sat Apr  2 20:05:44 2022 daemon.notice ttyd[3903]: started process, pid: 5353
Sat Apr  2 20:05:58 2022 auth.warn login[5353]: invalid password for 'UNKNOWN' on 'pts/0'
Sat Apr  2 20:06:08 2022 auth.info login[5354]: root login on 'pts/0'
Sat Apr  2 20:07:29 2022 daemon.notice ttyd[3903]: HTTP / - 192.168.2.1
Sat Apr  2 20:07:29 2022 daemon.notice ttyd[3903]: HTTP /token - 192.168.2.1
Sat Apr  2 20:07:29 2022 daemon.notice ttyd[3903]: WS   /ws - 192.168.2.1, clients: 2
Sat Apr  2 20:07:29 2022 daemon.notice ttyd[3903]: started process, pid: 5597

loydcheung, 2022-04-02:
[image]
[image]

loydcheung, 2022-04-02:
[image]
[image]

coolsnowwolf, 2022-04-02:
Sat Apr 2 19:41:06 2022 kern.info kernel: [ 23.015666] ath10k 5.15 driver, optimized for CT firmware, probing pci device: 0x46.

你应该还是要选 ct 版本的 firmware

loydcheung, 2022-04-02:
> Sat Apr 2 19:41:06 2022 kern.info kernel: [ 23.015666] ath10k 5.15 driver, optimized for CT firmware, probing pci device: 0x46.
> 
> 你应该还是要选 ct 版本的 firmware

太遗憾了，我估计只能抱着5.4.187版本用很久了，还是谢谢大佬花时间。

coolsnowwolf, 2022-04-03:
不是，我的意思是，你可以用 ath10k 配 ct 的 firmware

loydcheung, 2022-04-03:
> 不是，我的意思是，你可以用 ath10k 配 ct 的 firmware

我之前测试下来不带ct的firmware比较稳定好用，所以一直用的不带ct的，ct的2.4G经常有各种断流问题或者速度奇慢无比。

顺便问下我们qq群的进群问题“请正确回答：你用的什么路由器”该怎么回答才能进群。

loydcheung, 2022-04-04:
已经解决，需要手动另外在kernel module里面也相应选择不带ct的，大佬可以close了
[image]

zbluekuma, 2022-04-04:
麻烦请问大师：
Mon Apr  4 21:07:58 2022 kern.info kernel: [  120.421289] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
Mon Apr  4 21:07:59 2022 daemon.err odhcpd[2148]: Failed to send to ff02::1%lan@br-lan (Address not available)
这个错误怎么处理。

loydcheung, 2022-04-05:
> 麻烦请问大师： Mon Apr 4 21:07:58 2022 kern.info kernel: [ 120.421289] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready Mon Apr 4 21:07:59 2022 daemon.err odhcpd[2148]: Failed to send to ff02::1%lan@br-lan (Address not available) 这个错误怎么处理。

看起来像是ipv6的DHCP设置问题，你可以把你ipv6相关的设置都贴上来看看。

closed 2022-04-05 by loydcheung