coolsnowwolf/lede #9176: 关于最新源码编译失败

chiyu521, 2022-04-03:
Applying ./patches/rt2x00/651-rt2x00-driver-compile-with-kernel-5.15.patch using plaintext: 
patching file drivers/net/wireless/ralink/rt2x00/rt2x00dev.c
Hunk #1 FAILED at 997.
1 out of 1 hunk FAILED -- saving rejects to file drivers/net/wireless/ralink/rt2x00/rt2x00dev.c.rej
patch unexpectedly ends in middle of line
Patch failed!  Please fix ./patches/rt2x00/651-rt2x00-driver-compile-with-kernel-5.15.patch!
make[3]: *** [Makefile:568: /data/home/chiyu/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.15.8-1/.prepared_028c5ebf71d97a6985bbc21f46f45431_6664517399ebbbc92a37c5bb081b5c53] Error 1
make[3]: Leaving directory '/data/home/chiyu/lede/package/kernel/mac80211'
time: package/kernel/mac80211/regular/compile#1.33#0.72#2.10
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
make[2]: *** [package/Makefile:116: package/kernel/mac80211/compile] Error 1
make[2]: Leaving directory '/data/home/chiyu/lede'
make[1]: *** [package/Makefile:110: /data/home/chiyu/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/data/home/chiyu/lede'
make: *** [/data/home/chiyu/lede/include/toplevel.mk:230：world] 错误 2

WYC-2020, 2022-04-03:
删除这个patch不需要了

WYC-2020, 2022-04-03:
https://github.com/WYC-2020/lede/commit/c3369e495d586bc584c648523bdda01c0fde6b6b

chiyu521, 2022-04-03:
去掉了以后依然失败
Applying ./patches/rt2x00/651-rt2x00-driver-compile-with-kernel-5.15.patch using plaintext: 
patching file drivers/net/wireless/ralink/rt2x00/rt2x00dev.c
Hunk #1 FAILED at 997.
1 out of 1 hunk FAILED -- saving rejects to file drivers/net/wireless/ralink/rt2x00/rt2x00dev.c.rej
patch unexpectedly ends in middle of line
Patch failed!  Please fix ./patches/rt2x00/651-rt2x00-driver-compile-with-kernel-5.15.patch!
make[3]: *** [Makefile:568: /data/home/chiyu/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.15.8-1/.prepared_e9405f2b0833fe84981f4f4f25f7b23b_6664517399ebbbc92a37c5bb081b5c53] Error 1
make[3]: Leaving directory '/data/home/chiyu/lede/package/kernel/mac80211'
time: package/kernel/mac80211/regular/compile#1.77#1.09#3.70
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
make[2]: *** [package/Makefile:116: package/kernel/mac80211/compile] Error 1
make[2]: Leaving directory '/data/home/chiyu/lede'
make[1]: *** [package/Makefile:110: /data/home/chiyu/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/data/home/chiyu/lede'
make: *** [/data/home/chiyu/lede/include/toplevel.mk:230：world] 错误 2

WYC-2020, 2022-04-03:
额，你确定你去掉了，这明显就还在打651补丁

chiyu521, 2022-04-03:
对不起 我只删了001 现在删651

closed 2022-07-25 by aiamadeus