coolsnowwolf/lede #9180: bandwidthd依赖冲突

rnzipg, 2022-04-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述
bandwidthd-sqlite选择的php8会选择libgd-full，和bandwidthd-sqlite本身选择的libgd冲突。
### (1) 具体问题
A：
define Package/bandwidthd-sqlite
$(call Package/bandwidthd/Default)
  TITLE += (with sqlite enabled)
  VARIANT:=sqlite
  DEPENDS:=+libgd +libpcap +libsqlite3 +php8 +php8-cgi +php8-mod-gd +php8-mod-pdo +php8-mod-pdo-sqlite
endef

WYC-2020, 2022-04-05:
那就移除一个，或者改成一样的即可

Her0R0cket, 2022-04-06:
把这边的libgd改成libgd-full吧,php那边变了好几次了.
bandwidthd的几个包都要改下,不push的话就算了.

coolsnowwolf, 2026-09-05:
已修复当前 `feeds/packages/utils/bandwidthd/Makefile` 的实际依赖冲突：`bandwidthd-sqlite` 原来同时依赖 `libgd` 和 `php8-mod-gd`（后者依赖 `libgd-full`），现改为直接依赖 `libgd-full`。独立提交 `39bdfc6352842bcff3d0de592da6971ec2faf13a` 已 push 到 packages 仓库的 `master`。

closed 2026-09-05 by coolsnowwolf