coolsnowwolf/lede #9184: 5.4内核编译失败

jlzsss, 2022-04-03:
报错内容如下：
x86_64-openwrt-linux-musl-gcc -o gpsmon.o -c --sysroot=/media/jlz/FIRMWARE/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl -pthread -Wall -Wcast-align -Wextra -Wimplicit-fallthrough -Wmissing-declarations -Wmissing-prototypes -Wno-missing-field-initializers -Wno-uninitialized -Wpointer-arith -Wreturn-type -Wstrict-prototypes -Wvla -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/media/jlz/FIRMWARE/openwrt/build_dir/target-x86_64_musl/gpsd-3.21=gpsd-3.21 -Wformat -Werror=format-security -fstack-protector -znow -zrelro -O2 -pthread -D_FORTIFY_SOURCE=1 -D_DEFAULT_SOURCE -D_XOPEN_SOURCE=600 -I/media/jlz/FIRMWARE/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/media/jlz/FIRMWARE/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/media/jlz/FIRMWARE/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include -I/media/jlz/FIRMWARE/openwrt/staging_dir/target-x86_64_musl/usr/include/libusb-1.0 -I/media/jlz/FIRMWARE/openwrt/staging_dir/hostpkg/include/ncurses -I/media/jlz/FIRMWARE/openwrt/staging_dir/hostpkg/include gpsmon.c
gpsmon.c: In function 'refresh_cmdwin':
gpsmon.c:457:5: error: format not a string literal and no format arguments [-Werror=format-security]
     (void)wprintw(cmdwin, type_name);
     ^
cc1: some warnings being treated as errors
scons: *** [gpsmon.o] Error 1
scons: building terminated because of errors.
WARNING: asciidoctor not found.
WARNING: Some documentation and html will not be built.
make[2]: *** [Makefile:167: /media/jlz/FIRMWARE/openwrt/build_dir/target-x86_64_musl/gpsd-3.21/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 2
make[2]: Leaving directory '/media/jlz/FIRMWARE/openwrt/feeds/packages/utils/gpsd'
time: package/feeds/packages/gpsd/compile#11.29#1.24#12.09
    ERROR: package/feeds/packages/gpsd failed to build.
make[1]: *** [package/Makefile:116: package/feeds/packages/gpsd/compile] Error 1
make[1]: Leaving directory '/media/jlz/FIRMWARE/openwrt'
make: *** [/media/jlz/FIRMWARE/openwrt/include/toplevel.mk:230：package/gpsd/compile] 错误 2
请老大修复一下。

WYC-2020, 2022-04-05:
今天刚编译没任何问题

azgh, 2022-04-05:
R2S编译发生同样的问题，编译2次均失败。

WYC-2020, 2022-04-05:
> 今天R2S编译发生同样的问题，编译2次均失败。 [image]

那不清楚了没法修，无法复现

chenpapa, 2022-04-05:
解决了吗？树莓派4B遇到同样的问题了

EnnawYang, 2022-04-06:
前两天编译的并无错误， ~~除了无线寄了~~

zbluekuma, 2022-04-08:
同样5.4编译失败,5.10 ok 5.15 ok

azgh, 2022-04-09:
> > 今天R2S编译发生同样的问题，编译2次均失败。 [image]
> 
> 那不清楚了没法修，无法复现

重新配置环境，R2S编译成功了。

closed 2022-04-10 by jlzsss