coolsnowwolf/lede #9185: 最新版本NEWIFI D2编译不选择luci-app-mtwifi会没有“无线”设置，选择后wifi会自动停止

iowi, 2022-04-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

zhoujinshi, 2022-04-14:
我测试过，不是自动启动问题，是设置lan口IP后就自动启动失败了，通过ssh进去修改的lan的IP就是正常的。

zhoujinshi, 2022-04-14:
不过mtwifi这个插件不能设置[802.11k/v/r] 移动域无法配置

iowi, 2022-04-15:
不要mtwifi如何设置才能有无线选项？

zhoujinshi, 2022-04-16:
不选好像是通用wifi管理模块

iowi, 2022-04-16:
> 不选好像是通用wifi管理模块，通用模块有这些配置

不选“网络”那就没有“无线”

zhoujinshi, 2022-04-17:
是的，很纠结😣，不知道通知wifi模块怎么配置，网络一片空白，我复制别的配置使用云编译计算中。

zhoujinshi, 2022-04-17:
lede以外其他的分支可以openwrt-immortalwrt，要对比下参数

zhoujinshi, 2022-04-17:
6in4 - 26
adbyby - 2.7-2
appfilter - 5.0.1-2021-06-28-81f2e7ab-1
automount - 3
banip - 0.7.10-3
base-files - 1521-r19235-e80786fa7e
blkid - 2.36.1-2
block-mount - 2021-01-04-c53b1882-1
boost - 1.75.0-3
boost-date_time - 1.75.0-3
boost-program_options - 1.75.0-3
boost-system - 1.75.0-3
btrfs-progs - 5.11-1
busybox - 1.33.2-32
ca-bundle - 20210119-1
cgi-io - 2021-09-08-98cef9dd-20
cjdns - v21-3
collectd - 5.12.0-10
collectd-mod-cpu - 5.12.0-10
collectd-mod-interface - 5.12.0-10
collectd-mod-iwinfo - 5.12.0-10
collectd-mod-load - 5.12.0-10
collectd-mod-memory - 5.12.0-10
collectd-mod-network - 5.12.0-10
collectd-mod-rrdtool - 5.12.0-10
coreutils - 8.32-6
coreutils-base64 - 8.32-6
curl - 7.80.0-1
ddns-scripts - 2.8.2-12
ddns-scripts-services - 2.8.2-12
default-settings - 1.1-12
default-settings-chn - 1.1-12
dkjson - 2.5-4
dns2socks - 2.1-2
dnsmasq-full - 2.85-50
dropbear - 2020.81-2
e2fsprogs - 1.45.6-2
firewall - 2021-03-23-61db17ed-1
frpc - 0.38.0-1
fstools - 2021-01-04-c53b1882-1
fwtool - 2019-11-12-8f7fe925-1
getrandom - 2020-10-25-9ef88681-2
hostapd-common - 2020-06-08-5a8b3662-40
ip-full - 5.11.0-3
ip6tables - 1.8.7-1
ip6tables-mod-nat - 1.8.7-1
ipset - 7.6-1
ipt2socks - 1.1.3-2
iptables - 1.8.7-1
iptables-mod-conntrack-extra - 1.8.7-1
iptables-mod-fullconenat - 2019-10-21-0cf3b48f-1
iptables-mod-ipopt - 1.8.7-1
iptables-mod-tproxy - 1.8.7-1
iptvhelper - 0.1.1-1
ipv6helper - 1.0-4
iw - 5.9-8fab0c9e-3
iwinfo - 2021-04-30-c45f0b58-2.1
jshn - 2021-05-16-b14c4688-2
jsonfilter - 2018-02-04-c7e938d6-1
kernel - 5.4.182-1-bc0dff8c2e556008127cbce7b8bf977b
kmod-asn1-decoder - 5.4.182-1
kmod-cfg80211 - 5.4.182+5.10.85-1-1
kmod-crypto-acompress - 5.4.182-1
kmod-crypto-aead - 5.4.182-1
kmod-crypto-arc4 - 5.4.182-1
kmod-crypto-authenc - 5.4.182-1
kmod-crypto-crc32c - 5.4.182-1
kmod-crypto-ecb - 5.4.182-1
kmod-crypto-hash - 5.4.182-1
kmod-crypto-kpp - 5.4.182-1
[... 299 lines cut ...]
tcping - 0.3-1
terminfo - 6.2-3
trojan - 1.16.0-2
ttyd - 1.6.3-3
ubi-utils - 2.1.2-1
ubox - 2020-10-25-9ef88681-2
ubus - 2021-06-30-4fc532c8-2
ubusd - 2021-06-30-4fc532c8-2
uci - 2021-04-14-4b3db117-5
uclibcxx - 0.2.5-3
uclient-fetch - 2021-05-14-6a6011df-1
uhttpd - 2021-03-21-15346de8-2
uhttpd-mod-ubus - 2021-03-21-15346de8-2
urandom-seed - 3
urngd - 2020-01-21-c7f7b6b6-2
usign - 2020-05-23-f1f65026-1
vnstat2 - 2.9-1
vnstati2 - 2.9-1
wget-ssl - 1.21.1-1
wireguard-tools - 1.0.20210223-2
wireless-regdb - 2021.08.28-1
wpad-basic-wolfssl - 2020-06-08-5a8b3662-40
xray-core - 1.5.4-1
zerotier - 1.6.6-2
zlib - 1.2.11-3

closed 2022-07-25 by aiamadeus