coolsnowwolf/lede #9186: frr编译失败

avrileader, 2022-04-04:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：不选frr相关的选项正常，选了frr-staticd frr-ospfd frr-vtysh frr-watchfrr frr-zebra frr必报错

### (2) 路由器型号和固件版本

A：x86，最新lede：4.4

### (3) 详细日志

A：gcc -std=gnu99 -Wall -Wshadow -Wformat=2 -Wold-style-definition -Wstrict-prototypes -Wtrampolines -Wlogical-op -Wduplicated-cond -Wnull-dereference -Wimplicit-fallthrough=5 -Werror -Wunused -Wextra -Wstack-usage=262144    -D_FORTIFY_SOURCE=2 -O2 -I/home/avrileader/openwrt/lede/staging_dir/host/include -I/home/avrileader/openwrt/lede/staging_dir/hostpkg/include -I/home/avrileader/openwrt/lede/staging_dir/target-x86_64_musl/host/include  -L/home/avrileader/openwrt/lede/staging_dir/host/lib -L/home/avrileader/openwrt/lede/staging_dir/hostpkg/lib -L/home/avrileader/openwrt/lede/staging_dir/target-x86_64_musl/host/lib -o libdw.so -shared -Wl,-z,defs -Wl,-z,relro -o libdw.so \
        -Wl,--soname,libdw.so.1,--enable-new-dtags \
        -Wl,--version-script,libdw.map,--no-undefined \
        -Wl,--whole-archive ../libebl/libebl_pic.a ../backends/libebl_backends_pic.a ../libcpu/libcpu_pic.a libdw_pic.a ../libdwelf/libdwelf_pic.a ../libdwfl/libdwfl_pic.a -Wl,--no-whole-archive \
        ../lib/libeu.a ../libelf/libelf.so -ldl -lz    -lbz2  -pthread
/usr/bin/ld: libdw_pic.a(dwarf_error.os): in function `dwarf_errmsg':
dwarf_error.c:(.text+0xa2): undefined reference to `libintl_dgettext'
/usr/bin/ld: dwarf_error.c:(.text+0xd0): undefined reference to `libintl_dgettext'
/usr/bin/ld: libdw_pic.a(libdw_alloc.os): in function `__libdw_oom':
libdw_alloc.c:(.text+0x2ae): undefined reference to `libintl_gettext'
collect2: error: ld returned 1 exit status
make[7]: *** [Makefile:1081: libdw.so] Error 1
make[6]: *** [Makefile:583: all] Error 2
make[5]: *** [Makefile:526: all-recursive] Error 1
make[4]: *** [Makefile:442: all] Error 2
make[4]: Leaving directory '/home/avrileader/openwrt/lede/build_dir/hostpkg/elfutils-0.186'
make[3]: *** [Makefile:114: /home/avrileader/openwrt/lede/build_dir/hostpkg/elfutils-0.186/.built] Error 2
make[3]: Leaving directory '/home/avrileader/openwrt/lede/package/libs/elfutils'
time: package/libs/elfutils/host-compile#1.76#0.61#2.24
    ERROR: package/libs/elfutils [host] failed to build.
make[2]: *** [package/Makefile:114: package/libs/elfutils/host/compile] Error 1
make[2]: Leaving directory '/home/avrileader/openwrt/lede'
make[1]: *** [package/Makefile:110: /home/avrileader/openwrt/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/avrileader/openwrt/lede'
make: *** [/home/avrileader/openwrt/lede/include/toplevel.mk:230: world] Error 2

WYC-2020, 2022-04-05:
刚测试没任何问题

avrileader, 2022-04-05:
> 刚测试没任何问题

还是报错

WYC-2020, 2022-04-06:
> > 刚测试没任何问题
> 
> 还是报错

那不清楚了我这边按你那个选择没有报错

avrileader, 2022-04-06:
> > > 刚测试没任何问题
> > 
> > 
> > 还是报错
> 
> 那不清楚了我这边按你那个选择没有报错

好的 谢谢 ，我再尝试

closed 2022-04-06 by avrileader