coolsnowwolf/lede #9190: 编译固件加入luci-app-hd-idle编译没有问题，但硬盘不能休眠，这几天编译的固件都有这个问题

adwzlad, 2022-04-04:
编译固件加入luci-app-hd-idle编译没有问题，但硬盘不能休眠，这几天编译的固件都有这个问题

adwzlad, 2022-04-04:
X86

WYC-2020, 2022-04-07:
先用指令测试能否休眠，确定是那方的问题再说

adwzlad, 2022-04-07:
hd-idle -a /dev/sda -i 180
可以成功休眠

adwzlad, 2022-04-07:
sda是要休眠硬盘

adwzlad, 2022-04-07:
系统日志不会查看，没有找到相关字的日志

adwzlad, 2022-04-07:
日志截取
[    3.345246] usb 1-1: new high-speed USB device number 2 using xhci_hcd
[    3.347124] ata2.00: ATA Identify Device Log not supported
[    3.357466] ata2.00: configured for UDMA/133
[    3.364682] ata1.00: Features: Dev-Sleep
[    3.379699] ata1.00: configured for UDMA/133
[    3.384138] scsi 0:0:0:0: Direct-Access     ATA      eVtran V26SX     8A   PQ: 0 ANSI: 5
[    3.392652] sd 0:0:0:0: [sda] 247463936 512-byte logical blocks: (127 GB/118 GiB)
[    3.392670] scsi 1:0:0:0: Direct-Access     ATA      Hitachi HUS72403 A5F0 PQ: 0 ANSI: 5
[    3.400262] sd 0:0:0:0: [sda] Write Protect is off
[    3.408721] sd 1:0:0:0: [sdb] 5860533168 512-byte logical blocks: (3.00 TB/2.73 TiB)
[    3.413324] sd 0:0:0:0: [sda] Mode Sense: 00 3a 00 00
[    3.413387] sd 0:0:0:0: [sda] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[    3.421234] sd 1:0:0:0: [sdb] 4096-byte physical blocks
[    3.421274] sd 1:0:0:0: [sdb] Write Protect is off
[    3.440596] sd 1:0:0:0: [sdb] Mode Sense: 00 3a 00 00
[    3.440614]  sda: sda1
[    3.440621] sd 1:0:0:0: [sdb] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[    3.443140] sd 0:0:0:0: [sda] Attached SCSI removable disk
[    3.485635]  sdb: sdb1
[    3.488213] sd 1:0:0:0: [sdb] Attached SCSI disk
[    3.492931] Waiting for root device PARTUUID=49a780bf-d7b2-2a2b-079e-4edad8104802...
[    3.510532] usb-storage 1-1:1.0: USB Mass Storage device detected
[    3.516811] scsi host2: usb-storage 1-1:1.0
[    3.761427] tsc: Refined TSC clocksource calibration: 1607.999 MHz
[    3.768183] clocksource: tsc: mask: 0xffffffffffffffff max_cycles: 0x172da94f0e2, max_idle_ns: 440795278248 ns
[    3.779102] clocksource: Switched to clocksource tsc
[    4.529588] scsi 2:0:0:0: Direct-Access     AI       Mass Storage          PQ: 0 ANSI: 4
[    4.538384] sd 2:0:0:0: [sdc] 31490176 512-byte logical blocks: (16.1 GB/15.0 GiB)
[    4.546446] sd 2:0:0:0: [sdc] Write Protect is off
[    4.551511] sd 2:0:0:0: [sdc] Mode Sense: 03 00 00 00
[    4.551651] sd 2:0:0:0: [sdc] No Caching mode page found
[    4.557247] sd 2:0:0:0: [sdc] Assuming drive cache: write through
[    4.571680] GPT:Primary header thinks Alt. header is not at the end of the disk.
[    4.579165] GPT:1081888 != 31490175
[    4.582698] GPT:Alternate GPT header not at the end of the disk.
[    4.588792] GPT:1081888 != 31490175
[    4.592335] GPT: Use GNU Parted to correct GPT errors.
[    4.597553]  sdc: sdc1 sdc2 sdc128
[    4.601596] sd 2:0:0:0: [sdc] Attached SCSI removable disk
[    4.629765] VFS: Mounted root (squashfs filesystem) readonly on device 8:34.
[    4.637274] Freeing unused kernel image (initmem) memory: 1244K
[    4.645332] Write protecting the kernel read-only data: 18432k
[    4.652038] Freeing unused kernel image (text/rodata gap) memory: 2044K
[    4.659320] Freeing unused kernel image (rodata/data gap) memory: 1248K
[    4.666252] Run /sbin/init as init process
[    4.670414]   with arguments:
[    4.670414]     /sbin/init

adwzlad, 2022-04-07:
2022.04.07   22:00编译成功固件，luci-app-hd-idle也不起作用，也是有指令才可休眠

adwzlad, 2022-04-07:
其它不知道要要查看什么

WYC-2020, 2022-04-08:
那服务脚本里去掉-d -i 0这个我记得以前有一个人说过，也说是不能睡眠

coolsnowwolf, 2026-09-05:
硬盘休眠是否生效取决于磁盘、USB 转接、挂载、进程和 hd-idle 参数；报告没有磁盘型号、配置和 hdparm/日志证据，关闭为“不计划处理”。

closed 2026-09-05 by coolsnowwolf