coolsnowwolf/lede #9192: 最新的版本会导致x86 br-lan下子端口产生环路

GreatMichaelLee, 2022-04-05:
rt, 4.5的代码版本
问题描述：
1. x86 路由，eth3-eth5为br-lan端口，开启了stp生成树协议防止环路
2. eth3-eth5下有线方式的接了3个路由器当AP，这些线用ethtool看有 MDI-X on也有off的，TP类型 自协商1000M full, 之前的版本都OK
3. 升级了4.5编译的版本后，发现Br-lan下端口不断地进入blocking->disable->listening的flap状态，原因应该是STP协议在工作，但实际物理线我用测线仪测了没任何问题。关闭STP协议后，故障消失，但是端口明显降速了，路由下测试只有200-300M（正常时940M).
4. 回退到3.30（软路由只保留了这个）的代码编译版本，使用的一模一样的配置，故障消失
5. 又重新升级到4.5版本，问题复现

应该是最近两日的代码合入有哪里底层相关的有bug.  附软路由日志及网卡信息

端口日志：
	行 786: Tue Apr  5 11:07:14 2022 kern.info kernel: [   38.183462] br-lan: port 1(eth3) entered blocking state
	行 787: Tue Apr  5 11:07:14 2022 kern.info kernel: [   38.188940] br-lan: port 1(eth3) entered disabled state
	行 791: Tue Apr  5 11:07:14 2022 kern.info kernel: [   38.250165] br-lan: port 2(eth4) entered blocking state
	行 792: Tue Apr  5 11:07:14 2022 kern.info kernel: [   38.255625] br-lan: port 2(eth4) entered disabled state
	行 795: Tue Apr  5 11:07:14 2022 kern.info kernel: [   38.312677] br-lan: port 3(eth5) entered blocking state
	行 796: Tue Apr  5 11:07:14 2022 kern.info kernel: [   38.318130] br-lan: port 3(eth5) entered disabled state
	行 821: Tue Apr  5 11:07:17 2022 kern.info kernel: [   41.218499] br-lan: port 2(eth4) entered blocking state
	行 822: Tue Apr  5 11:07:17 2022 kern.info kernel: [   41.223939] br-lan: port 2(eth4) entered listening state
	行 825: Tue Apr  5 11:07:17 2022 kern.info kernel: [   41.239485] br-lan: port 1(eth3) entered blocking state
	行 826: Tue Apr  5 11:07:17 2022 kern.info kernel: [   41.244937] br-lan: port 1(eth3) entered listening state
	行 828: Tue Apr  5 11:07:17 2022 kern.info kernel: [   41.264002] br-lan: port 3(eth5) entered blocking state
	行 829: Tue Apr  5 11:07:17 2022 kern.info kernel: [   41.269427] br-lan: port 3(eth5) entered listening state
	行 1080: Tue Apr  5 11:07:25 2022 kern.info kernel: [   49.279956] br-lan: port 3(eth5) entered learning state
	行 1081: Tue Apr  5 11:07:25 2022 kern.info kernel: [   49.285536] br-lan: port 2(eth4) entered learning state
	行 1082: Tue Apr  5 11:07:25 2022 kern.info kernel: [   49.290975] br-lan: port 1(eth3) entered learning state
	行 1112: Tue Apr  5 11:07:34 2022 kern.info kernel: [   57.472007] br-lan: port 2(eth4) entered forwarding state
	行 1113: Tue Apr  5 11:07:34 2022 kern.info kernel: [   57.477587] br-lan: topology change detected, propagating
	行 1114: Tue Apr  5 11:07:34 2022 kern.info kernel: [   57.483214] br-lan: port 1(eth3) entered forwarding state
	行 1115: Tue Apr  5 11:07:34 2022 kern.info kernel: [   57.483483] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
	行 1116: Tue Apr  5 11:07:34 2022 kern.info kernel: [   57.488807] br-lan: topology change detected, propagating
	行 1117: Tue Apr  5 11:07:34 2022 kern.info kernel: [   57.488819] br-lan: port 3(eth5) entered forwarding state
	行 1118: Tue Apr  5 11:07:34 2022 daemon.notice netifd: bridge 'br-lan' link is up
	行 1121: Tue Apr  5 11:07:34 2022 daemon.info avahi-daemon[7576]: Joining mDNS multicast group on interface br-lan.IPv4 with address 192.168.2.1.
	行 1122: Tue Apr  5 11:07:34 2022 daemon.info avahi-daemon[7576]: New relevant interface br-lan.IPv4 for mDNS.
	行 1123: Tue Apr  5 11:07:34 2022 daemon.info avahi-daemon[7576]: Registering new address record for 192.168.2.1 on br-lan.IPv4.
	行 1124: Tue Apr  5 11:07:34 2022 kern.info kernel: [   57.506675] br-lan: topology change detected, propagating
	行 1131: Tue Apr  5 11:07:35 2022 daemon.info avahi-daemon[7576]: Joining mDNS multicast group on interface br-lan.IPv6 with address fe80::290:27ff:fee5:652d.
	行 1132: Tue Apr  5 11:07:35 2022 daemon.info avahi-daemon[7576]: New relevant interface br-lan.IPv6 for mDNS.
	行 1133: Tue Apr  5 11:07:35 2022 daemon.info avahi-daemon[7576]: Registering new address record for fe80::290:27ff:fee5:652d on br-lan.*.
	行 1150: Tue Apr  5 11:07:39 2022 kern.info kernel: [   62.463716] br-lan: port 2(eth4) entered disabled state
	行 1169: Tue Apr  5 11:07:40 2022 kern.info kernel: [   63.488217] br-lan: port 1(eth3) entered disabled state
	行 1172: Tue Apr  5 11:07:41 2022 kern.info kernel: [   64.512186] br-lan: port 3(eth5) entered disabled state
	行 1176: Tue Apr  5 11:07:42 2022 daemon.notice netifd: bridge 'br-lan' link is down
	行 1179: Tue Apr  5 11:07:42 2022 daemon.info avahi-daemon[7576]: Withdrawing address record for 192.168.2.1 on br-lan.
	行 1180: Tue Apr  5 11:07:42 2022 daemon.info avahi-daemon[7576]: Leaving mDNS multicast group on interface br-lan.IPv4 with address 192.168.2.1.
	行 1181: Tue Apr  5 11:07:42 2022 daemon.info avahi-daemon[7576]: Interface br-lan.IPv4 no longer relevant for mDNS.
	行 1182: Tue Apr  5 11:07:42 2022 daemon.info avahi-daemon[7576]: Interface br-lan.IPv6 no longer relevant for mDNS.
	行 1183: Tue Apr  5 11:07:42 2022 daemon.info avahi-daemon[7576]: Leaving mDNS multicast group on interface br-lan.IPv6 with address fe80::290:27ff:fee5:652d.
	行 1185: Tue Apr  5 11:07:42 2022 kern.info kernel: [   65.549888] br-lan: port 1(eth3) entered disabled state
	行 1201: Tue Apr  5 11:07:42 2022 daemon.info avahi-daemon[7576]: Withdrawing address record for fe80::290:27ff:fee5:652d on br-lan.
	行 1210: Tue Apr  5 11:07:42 2022 kern.info kernel: [   65.663572] br-lan: port 2(eth4) entered disabled state
	行 1214: Tue Apr  5 11:07:42 2022 kern.info kernel: [   65.716045] br-lan: port 3(eth5) entered disabled state
	行 1217: Tue Apr  5 11:07:42 2022 kern.info kernel: [   65.830998] br-lan: port 1(eth3) entered blocking state
	行 1218: Tue Apr  5 11:07:42 2022 kern.info kernel: [   65.836453] br-lan: port 1(eth3) entered disabled state
	行 1221: Tue Apr  5 11:07:42 2022 kern.info kernel: [   65.878993] br-lan: port 2(eth4) entered blocking state
	行 1222: Tue Apr  5 11:07:42 2022 kern.info kernel: [   65.884471] br-lan: port 2(eth4) entered disabled state
	行 1225: Tue Apr  5 11:07:42 2022 kern.info kernel: [   65.927001] br-lan: port 3(eth5) entered blocking state
	行 1226: Tue Apr  5 11:07:42 2022 kern.info kernel: [   65.932458] br-lan: port 3(eth5) entered disabled state
	行 1428: Tue Apr  5 11:07:45 2022 kern.info kernel: [   68.694493] br-lan: port 1(eth3) entered blocking state
	行 1429: Tue Apr  5 11:07:45 2022 kern.info kernel: [   68.699916] br-lan: port 1(eth3) entered listening state
	行 1432: Tue Apr  5 11:07:45 2022 kern.info kernel: [   68.798531] br-lan: port 2(eth4) entered blocking state
	行 1433: Tue Apr  5 11:07:45 2022 kern.info kernel: [   68.804094] br-lan: port 2(eth4) entered listening state
	行 1520: Tue Apr  5 11:07:45 2022 kern.info kernel: [   69.132379] br-lan: port 3(eth5) entered blocking state
	行 1521: Tue Apr  5 11:07:45 2022 kern.info kernel: [   69.137789] br-lan: port 3(eth5) entered listening state
	行 1549: Tue Apr  5 11:07:47 2022 daemon.info vnstatd[19064]: Info: Monitoring: eth3 (1000 Mbit) eth1 (1000 Mbit) pppoe-vwan2 (1000 Mbit) br-lan (1000 Mbit) pppoe-WAN3 (1000 Mbit) eth0 (1000 Mbit) pppoe-vwan1 (1000 Mbit) eth4 (1000 Mbit) eth5 (1000 Mbit) eth2 (1000 Mbit) tun0 (10 Mbit)
	行 1585: Tue Apr  5 11:07:54 2022 kern.info kernel: [   76.927619] br-lan: port 2(eth4) entered learning state
	行 1586: Tue Apr  5 11:07:54 2022 kern.info kernel: [   76.933112] br-lan: port 1(eth3) entered learning state
	行 1587: Tue Apr  5 11:07:54 2022 kern.info kernel: [   77.183535] br-lan: port 3(eth5) entered learning state
	行 1620: Tue Apr  5 11:08:02 2022 kern.info kernel: [   85.118769] br-lan: port 1(eth3) entered forwarding state
[... 305 lines cut ...]
00:00.0 Host bridge: Intel Corporation Xeon E3-1200 v6/7th Gen Core Processor Host Bridge/DRAM Registers (rev 02)
00:02.0 VGA compatible controller: Intel Corporation Device 5906 (rev 02)
00:08.0 System peripheral: Intel Corporation Xeon E3-1200 v5/v6 / E3-1500 v5 / 6th/7th/8th Gen Core Processor Gaussian Mixture Model
00:14.0 USB controller: Intel Corporation Sunrise Point-LP USB 3.0 xHCI Controller (rev 21)
00:14.2 Signal processing controller: Intel Corporation Sunrise Point-LP Thermal subsystem (rev 21)
00:15.0 Signal processing controller: Intel Corporation Sunrise Point-LP Serial IO I2C Controller #0 (rev 21)
00:15.1 Signal processing controller: Intel Corporation Sunrise Point-LP Serial IO I2C Controller #1 (rev 21)
00:16.0 Communication controller: Intel Corporation Sunrise Point-LP CSME HECI #1 (rev 21)
00:17.0 SATA controller: Intel Corporation Sunrise Point-LP SATA Controller [AHCI mode] (rev 21)
00:1c.0 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #3 (rev f1)
00:1c.3 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #4 (rev f1)
[... cut ...]
01:00.0 Ethernet controller: Intel Corporation I211 Gigabit Network Connection (rev 03)
02:00.0 Ethernet controller: Intel Corporation I211 Gigabit Network Connection (rev 03)
03:00.0 Ethernet controller: Intel Corporation I211 Gigabit Network Connection (rev 03)
04:00.0 Ethernet controller: Intel Corporation I211 Gigabit Network Connection (rev 03)
05:00.0 Ethernet controller: Intel Corporation I211 Gigabit Network Connection (rev 03)
06:00.0 Ethernet controller: Intel Corporation I211 Gigabit Network Connection (rev 03)

GreatMichaelLee, 2022-04-05:
找了个4.3的编译版本，问题一样，3.30版本OK, 应该是跟这期间commit的代码相关

GreatMichaelLee, 2022-04-05:
确认了时netifd最新版本的问题，其他code都用最新的基线，单纯回退netifd的版本（修改makefile里的版本到去年6月份的），问题消失，麻烦大佬看看netifd今年二月份这个版本是否改了mac/phy/driver配置相关的？或者说先回退一下？
@coolsnowwolf 

[image]

loydcheung, 2022-04-05:
我也有同样的问题，我是r8125的2.5G口软*由，之前都是和下游NAS自动协商成2.5G用，这两天现在日志里显示网口entered promiscuous mode. 首页显示无连接，网口指示灯也不亮。需要手动敲代码ethtool -s eth* speed 2500 duplex full才能和下游协商2.5G（下游的intel X550卡在TrueNAS上用也不支持autoneg，暂时常年固定在2.5G速度下，正常使用）

GreatMichaelLee, 2022-04-05:
可能是不是最新的netifd会自己设置混杂模式，导致协商或端口还回等问题？纯猜测。

loydcheung, 2022-04-05:
netifd已回退，经测试问题已修复。

GreatMichaelLee, 2022-04-06:
> netifd已回退，经测试问题已修复。

是的，回退好了，新版的看来有问题

WYC-2020, 2022-04-07:
其实这个可以试下升级stp 看下 ，有可能 新版netifd 需要配套新版的stp

closed 2022-04-07 by coolsnowwolf