coolsnowwolf/lede #9195: openssh-sftp segment fault issue

GreatMichaelLee, 2022-04-05:
x86/arm都有这个问题，最新基线(4.5)，之前一直存在，提了没人修，用之前腾讯云源上的8.8p1-5包替换就没问题（匪夷所思），用8.9p1-1（代码编译出来的版本）的就是不行。
[image]

发起sftp连接时，软路由显示内核段错误

Tue Apr  5 19:10:09 2022 kern.info kernel: [ 1632.750030] sftp-server[24157]: segfault at ffffffff87331450 ip 00005654eeff3711 sp 00007ffe0e491e78 error 7 in sftp-server[5654eeff3000+11000]
Tue Apr  5 19:10:09 2022 kern.info kernel: [ 1632.763913] Code: c4 88 04 00 00 5b 5d 41 5c 41 5d 41 5e 41 5f c3 8b 15 0b a9 01 00 48 63 c7 89 3d 02 a9 01 00 48 6b c0 38 48 03 05 d7 aa 01 00 <c7> 00 00 00 00 00 89 50 30 c3 31 c0 85 ff 78 1d 3b 3d b9 aa 01 00

server端回复128错误。
[image]

WYC-2020, 2022-04-05:
我以前就说过，我修复了的，用我仓库的

GreatMichaelLee, 2022-04-05:
> 我以前就说过，我修复了的，用我仓库的

哪个，给个commit

closed 2022-04-06 by coolsnowwolf