coolsnowwolf/lede #9197: 编译passwall报错

xiaotan8, 2022-04-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
Process completed with exit code 2.
### (1) 具体问题

A：github.com/marten-seemann/qtls-go1-18@v0.1.1/conn.go:973:14: too many errors

### (2) 路由器型号和固件版本

A：X86

### (3) 详细日志

A：
2022-04-05T22:06:23.1520766Z SHELL= flock /home/runner/work/Openwrt-actions/Openwrt-actions/openwrt/tmp/.v2ray-core-5.0.3.tar.gz.flock -c '  	/home/runner/work/Openwrt-actions/Openwrt-actions/openwrt/scripts/download.pl "/home/runner/work/Openwrt-actions/Openwrt-actions/openwrt/dl" "v2ray-core-5.0.3.tar.gz" "c0fe91f715293cfc39a5afeef71e1ff43d379ae0faa139c560fdc3ede88a458a" "" "https://codeload.github.com/v2fly/v2ray-core/tar.gz/v5.0.3?"    '
2022-04-05T22:06:23.2060667Z rm -f /home/runner/work/Openwrt-actions/Openwrt-actions/openwrt/build_dir/target-x86_64_musl/v2ray-core-5.0.3/.built
2022-04-05T22:06:23.2086878Z touch /home/runner/work/Openwrt-actions/Openwrt-actions/openwrt/build_dir/target-x86_64_musl/v2ray-core-5.0.3/.built_check
2022-04-05T22:06:23.2119058Z CONFIG_GOLANG_MOD_CACHE_WORLD_READABLE="" GO_BUILD_CACHE_DIR="/home/runner/work/Openwrt-actions/Openwrt-actions/openwrt/tmp/go-build" GO_MOD_CACHE_DIR="/home/runner/work/Openwrt-actions/Openwrt-actions/openwrt/dl/go-mod-cache" GO_MOD_ARGS="-modcacherw" GO_PKG="github.com/v2fly/v2ray-core/v5" GO_INSTALL_EXTRA="" GO_INSTALL_ALL="" GO_SOURCE_ONLY="" GO_BUILD_PKG="github.com/v2fly/v2ray-core/v5/..." GO_EXCLUDES="" GO_GO_GENERATE="" GO_INSTALL_BIN_PATH="/usr/bin" BUILD_DIR="/home/runner/work/Openwrt-actions/Openwrt-actions/openwrt/build_dir/target-x86_64_musl/v2ray-core-5.0.3" GO_BUILD_DIR="/home/runner/work/Openwrt-actions/Openwrt-actions/openwrt/build_dir/target-x86_64_musl/v2ray-core-5.0.3/.go_work/build" GO_BUILD_BIN_DIR="/home/runner/work/Openwrt-actions/Openwrt-actions/openwrt/build_dir/target-x86_64_musl/v2ray-core-5.0.3/.go_work/build/bin" GO_BUILD_DEPENDS_PATH="/usr/share/gocode" GO_BUILD_DEPENDS_SRC="/home/runner/work/Openwrt-actions/Openwrt-actions/openwrt/staging_dir/target-x86_64_musl/usr/share/gocode/src" GOOS="linux" GOARCH="amd64" GO386="" GOARM="" GOMIPS="" GOMIPS64="" CGO_ENABLED=1 CC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" CGO_CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/runner/work/Openwrt-actions/Openwrt-actions/openwrt/build_dir/target-x86_64_musl/v2ray-core-5.0.3=v2ray-core-5.0.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_CPPFLAGS="-I/home/runner/work/Openwrt-actions/Openwrt-actions/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/home/runner/work/Openwrt-actions/Openwrt-actions/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/home/runner/work/Openwrt-actions/Openwrt-actions/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include" CGO_CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/runner/work/Openwrt-actions/Openwrt-actions/openwrt/build_dir/target-x86_64_musl/v2ray-core-5.0.3=v2ray-core-5.0.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_LDFLAGS="-L/home/runner/work/Openwrt-actions/Openwrt-actions/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/runner/work/Openwrt-actions/Openwrt-actions/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -znow -zrelro" GOPATH="/home/runner/work/Openwrt-actions/Openwrt-actions/openwrt/build_dir/target-x86_64_musl/v2ray-core-5.0.3/.go_work/build" GOCACHE="/home/runner/work/Openwrt-actions/Openwrt-actions/openwrt/tmp/go-build" GOMODCACHE="/home/runner/work/Openwrt-actions/Openwrt-actions/openwrt/dl/go-mod-cache" GOENV=off /usr/bin/env bash /home/runner/work/Openwrt-actions/Openwrt-actions/openwrt/feeds/packages/lang/golang//golang-build.sh build -v -trimpath -ldflags "all=-buildid '1649178380' -linkmode external -extldflags '-L/home/runner/work/Openwrt-actions/Openwrt-actions/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/runner/work/Openwrt-actions/Openwrt-actions/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -Wl,-z,now -Wl,-z,relro'"        -ldflags " -X github.com/v2fly/v2ray-core/v5.build=OpenWrt -X github.com/v2fly/v2ray-core/v5.version=5.0.3 -buildid '1649178380' -linkmode external -extldflags '-L/home/runner/work/Openwrt-actions/Openwrt-actions/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/runner/work/Openwrt-actions/Openwrt-actions/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -Wl,-z,now -Wl,-z,relro'"  
2022-04-05T22:06:23.2169546Z Finding targets
2022-04-05T22:06:24.4120095Z 
2022-04-05T22:06:24.4120528Z Building targets
2022-04-05T22:06:25.0136420Z github.com/marten-seemann/qtls-go1-18
2022-04-05T22:06:25.0700843Z # github.com/marten-seemann/qtls-go1-18
2022-04-05T22:06:25.0704436Z ../../../../../dl/go-mod-cache/github.com/marten-seemann/qtls-go1-18@v0.1.1/cipher_suites.go:141:43: undefined: any
2022-04-05T22:06:25.0705446Z ../../../../../dl/go-mod-cache/github.com/marten-seemann/qtls-go1-18@v0.1.1/cipher_suites.go:399:45: undefined: any
2022-04-05T22:06:25.0706355Z ../../../../../dl/go-mod-cache/github.com/marten-seemann/qtls-go1-18@v0.1.1/cipher_suites.go:404:46: undefined: any
2022-04-05T22:06:25.0707210Z ../../../../../dl/go-mod-cache/github.com/marten-seemann/qtls-go1-18@v0.1.1/cipher_suites.go:412:45: undefined: any
2022-04-05T22:06:25.0707963Z ../../../../../dl/go-mod-cache/github.com/marten-seemann/qtls-go1-18@v0.1.1/common.go:1496:41: undefined: any
2022-04-05T22:06:25.0708696Z ../../../../../dl/go-mod-cache/github.com/marten-seemann/qtls-go1-18@v0.1.1/conn.go:177:10: undefined: any
2022-04-05T22:06:25.0709418Z ../../../../../dl/go-mod-cache/github.com/marten-seemann/qtls-go1-18@v0.1.1/conn.go:183:13: undefined: any
2022-04-05T22:06:25.0710127Z ../../../../../dl/go-mod-cache/github.com/marten-seemann/qtls-go1-18@v0.1.1/conn.go:211:62: undefined: any
2022-04-05T22:06:25.0710835Z ../../../../../dl/go-mod-cache/github.com/marten-seemann/qtls-go1-18@v0.1.1/conn.go:973:14: undefined: any
2022-04-05T22:06:25.0711467Z ../../../../../dl/go-mod-cache/github.com/marten-seemann/qtls-go1-18@v0.1.1/conn.go:1056:33: undefined: any
2022-04-05T22:06:25.0712057Z ../../../../../dl/go-mod-cache/github.com/marten-seemann/qtls-go1-18@v0.1.1/conn.go:973:14: too many errors
2022-04-05T22:06:25.0712526Z note: module requires Go 1.18
2022-04-05T22:06:26.3193141Z 
2022-04-05T22:06:26.4499470Z make[3]: *** [Makefile:111: /home/runner/work/Openwrt-actions/Openwrt-actions/openwrt/build_dir/target-x86_64_musl/v2ray-core-5.0.3/.built] Error 2
2022-04-05T22:06:26.4500302Z make[3]: Leaving directory '/home/runner/work/Openwrt-actions/Openwrt-actions/openwrt/package/passwall-packages/v2ray-core'
2022-04-05T22:06:26.4507764Z time: package/passwall-packages/v2ray-core/compile#3.18#1.18#3.47
2022-04-05T22:06:26.4511141Z     ERROR: package/passwall-packages/v2ray-core failed to build.
2022-04-05T22:06:26.4525507Z make[2]: *** [package/Makefile:116: package/passwall-packages/v2ray-core/compile] Error 1
2022-04-05T22:06:26.4526115Z make[2]: Leaving directory '/home/runner/work/Openwrt-actions/Openwrt-actions/openwrt'
2022-04-05T22:06:26.4530150Z make[1]: *** [package/Makefile:110: /home/runner/work/Openwrt-actions/Openwrt-actions/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
2022-04-05T22:06:26.4534652Z make[1]: Leaving directory '/home/runner/work/Openwrt-actions/Openwrt-actions/openwrt'
2022-04-05T22:06:26.4542523Z make: *** [/home/runner/work/Openwrt-actions/Openwrt-actions/openwrt/include/toplevel.mk:230: world] Error 2
2022-04-05T22:06:26.4743078Z ##[error]Process completed with exit code 2.
2022-04-05T22:06:26.5284441Z ##[group]Run df -hT
2022-04-05T22:06:26.5284720Z [36;1mdf -hT[0m
2022-04-05T22:06:26.5390063Z shell: /usr/bin/bash -e {0}
2022-04-05T22:06:26.5390321Z env:
2022-04-05T22:06:26.5390538Z   DATE: 2022-04-05
2022-04-05T22:06:26.5390852Z   REPO_URL: https://github.com/coolsnowwolf/lede
2022-04-05T22:06:26.5391156Z   REPO_BRANCH: master
2022-04-05T22:06:26.5391401Z   UPLOAD_PACKAGES_DIR: true
2022-04-05T22:06:26.5391666Z   UPLOAD_TARGETS_DIR: true
2022-04-05T22:06:26.5391919Z   UPLOAD_FIRMWARE: true
2022-04-05T22:06:26.5392166Z   UPLOAD_TO_ARTIFACTS: falsee
[... cut ...]
2022-04-05T22:06:26.9912277Z [command]/usr/bin/git config --local --name-only --get-regexp http\.https\:\/\/github\.com\/\.extraheader
2022-04-05T22:06:27.0011690Z http.https://github.com/.extraheader
2022-04-05T22:06:27.0028352Z [command]/usr/bin/git config --local --unset-all http.https://github.com/.extraheader
2022-04-05T22:06:27.0154091Z [command]/usr/bin/git submodule foreach --recursive git config --local --name-only --get-regexp 'http\.https\:\/\/github\.com\/\.extraheader' && git config --local --unset-all 'http.https://github.com/.extraheader' || :
2022-04-05T22:06:27.0933868Z Cleaning up orphan processes

WYC-2020, 2022-04-06:
隔壁

WYC-2020, 2022-04-06:
> 反馈bug/问题模板，提建议请删除
> 
> ## 1.关于你要提交的问题
> Q：是否搜索了issue (使用 "x" 选择)
> 
> * [ ]  没有类似的issue
> 
> ## 2. 详细叙述
> Process completed with exit code 2.
> 
> ### (1) 具体问题
> A：github.com/marten-seemann/qtls-go1-18@v0.1.1/conn.go:973:14: too many errors
> 
> ### (2) 路由器型号和固件版本
> A：X86
> 
> ### (3) 详细日志
> A： 2022-04-05T22:06:23.1520766Z SHELL= flock /home/runner/work/Openwrt-actions/Openwrt-actions/openwrt/tmp/.v2ray-core-5.0.3.tar.gz.flock -c ' /home/runner/work/Openwrt-actions/Openwrt-actions/openwrt/scripts/download.pl "/home/runner/work/Openwrt-actions/Openwrt-actions/openwrt/dl" "v2ray-core-5.0.3.tar.gz" "c0fe91f715293cfc39a5afeef71e1ff43d379ae0faa139c560fdc3ede88a458a" "" "https://codeload.github.com/v2fly/v2ray-core/tar.gz/v5.0.3?" ' 2022-04-05T22:06:23.2060667Z rm -f /home/runner/work/Openwrt-actions/Openwrt-actions/openwrt/build_dir/target-x86_64_musl/v2ray-core-5.0.3/.built 2022-04-05T22:06:23.2086878Z touch /home/runner/work/Openwrt-actions/Openwrt-actions/openwrt/build_dir/target-x86_64_musl/v2ray-core-5.0.3/.built_check 2022-04-05T22:06:23.2119058Z CONFIG_GOLANG_MOD_CACHE_WORLD_READABLE="" GO_BUILD_CACHE_DIR="/home/runner/work/Openwrt-actions/Openwrt-actions/openwrt/tmp/go-build" GO_MOD_CACHE_DIR="/home/runner/work/Openwrt-actions/Openwrt-actions/openwrt/dl/go-mod-cache" GO_MOD_ARGS="-modcacherw" GO_PKG="github.com/v2fly/v2ray-core/v5" GO_INSTALL_EXTRA="" GO_INSTALL_ALL="" GO_SOURCE_ONLY="" GO_BUILD_PKG="github.com/v2fly/v2ray-core/v5/..." GO_EXCLUDES="" GO_GO_GENERATE="" GO_INSTALL_BIN_PATH="/usr/bin" BUILD_DIR="/home/runner/work/Openwrt-actions/Openwrt-actions/openwrt/build_dir/target-x86_64_musl/v2ray-core-5.0.3" GO_BUILD_DIR="/home/runner/work/Openwrt-actions/Openwrt-actions/openwrt/build_dir/target-x86_64_musl/v2ray-core-5.0.3/.go_work/build" GO_BUILD_BIN_DIR="/home/runner/work/Openwrt-actions/Openwrt-actions/openwrt/build_dir/target-x86_64_musl/v2ray-core-5.0.3/.go_work/build/bin" GO_BUILD_DEPENDS_PATH="/usr/share/gocode" GO_BUILD_DEPENDS_SRC="/home/runner/work/Openwrt-actions/Openwrt-actions/openwrt/staging_dir/target-x86_64_musl/usr/share/gocode/src" GOOS="linux" GOARCH="amd64" GO386="" GOARM="" GOMIPS="" GOMIPS64="" CGO_ENABLED=1 CC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" CGO_CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/runner/work/Openwrt-actions/Openwrt-actions/openwrt/build_dir/target-x86_64_musl/v2ray-core-5.0.3=v2ray-core-5.0.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_CPPFLAGS="-I/home/runner/work/Openwrt-actions/Openwrt-actions/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/home/runner/work/Openwrt-actions/Openwrt-actions/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/home/runner/work/Openwrt-actions/Openwrt-actions/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include" CGO_CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/runner/work/Openwrt-actions/Openwrt-actions/openwrt/build_dir/target-x86_64_musl/v2ray-core-5.0.3=v2ray-core-5.0.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_LDFLAGS="-L/home/runner/work/Openwrt-actions/Openwrt-actions/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/runner/work/Openwrt-actions/Openwrt-actions/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -znow -zrelro" GOPATH="/home/runner/work/Openwrt-actions/Openwrt-actions/openwrt/build_dir/target-x86_64_musl/v2ray-core-5.0.3/.go_work/build" GOCACHE="/home/runner/work/Openwrt-actions/Openwrt-actions/openwrt/tmp/go-build" GOMODCACHE="/home/runner/work/Openwrt-actions/Openwrt-actions/openwrt/dl/go-mod-cache" GOENV=off /usr/bin/env bash /home/runner/work/Openwrt-actions/Openwrt-actions/openwrt/feeds/packages/lang/golang//golang-build.sh build -v -trimpath -ldflags "all=-buildid '1649178380' -linkmode external -extldflags '-L/home/runner/work/Openwrt-actions/Openwrt-actions/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/runner/work/Openwrt-actions/Openwrt-actions/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -Wl,-z,now -Wl,-z,relro'" -ldflags " -X github.com/v2fly/v2ray-core/v5.build=OpenWrt -X github.com/v2fly/v2ray-core/v5.version=5.0.3 -buildid '1649178380' -linkmode external -extldflags '-L/home/runner/work/Openwrt-actions/Openwrt-actions/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/runner/work/Openwrt-actions/Openwrt-actions/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -Wl,-z,now -Wl,-z,relro'" 2022-04-05T22:06:23.2169546Z Finding targets 2022-04-05T22:06:24.4120095Z 2022-04-05T22:06:24.4120528Z Building targets 2022-04-05T22:06:25.0136420Z github.com/marten-seemann/qtls-go1-18 2022-04-05T22:06:25.0700843Z # github.com/marten-seemann/qtls-go1-18 2022-04-05T22:06:25.0704436Z ../../../../../dl/go-mod-cache/github.com/marten-seemann/qtls-go1-18@v0.1.1/cipher_suites.go:141:43: undefined: any 2022-04-05T22:06:25.0705446Z ../../../../../dl/go-mod-cache/github.com/marten-seemann/qtls-go1-18@v0.1.1/cipher_suites.go:399:45: undefined: any 2022-04-05T22:06:25.0706355Z ../../../../../dl/go-mod-cache/github.com/marten-seemann/qtls-go1-18@v0.1.1/cipher_suites.go:404:46: undefined: any 2022-04-05T22:06:25.0707210Z ../../../../../dl/go-mod-cache/g [... cut ...] 022-04-05T22:06:27.0028352Z [command]/usr/bin/git config --local --unset-all http.https://github.com/.extraheader 2022-04-05T22:06:27.0154091Z [command]/usr/bin/git submodule foreach --recursive git config --local --name-only --get-regexp 'http.https://github.com/.extraheader' && git config --local --unset-all 'http.https://github.com/.extraheader' || : 2022-04-05T22:06:27.0933868Z Cleaning up orphan processes

以后这种问题去隔壁提，这个是提交了1.8 go那个patch有问题，吧那个提交撤销，继续用原来的1.17，这种升级go依赖v2ray-core后续都会升级的，何必急于此刻

WYC-2020, 2022-04-07:
这个去隔壁，我给了更好的解决方案了 @xiaotan8

closed 2022-04-07 by coolsnowwolf