coolsnowwolf/lede #9199: 最新原编译失败

a7502, 2022-04-06:
刚更新源编译失败，前几天还可以编译。老大看看是那里的问题。。。
SHELL= flock /home/koko/lede/tmp/.root-copy.flock -c 'cp -fpR /home/koko/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/.pkgdir/kmod-drm-i915/. /home/koko/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/koko/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.kmod-drm-i915_installed
echo "kmod-drm-i915" >> /home/koko/lede/staging_dir/target-x86_64_musl/pkginfo/linux.default.install
mkdir -p /home/koko/lede/staging_dir/target-x86_64_musl/root-x86/stamp

SHELL= flock /home/koko/lede/tmp/.root-copy.flock -c 'cp -fpR /home/koko/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/.pkgdir/kmod-amazon-ena/. /home/koko/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/koko/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.kmod-amazon-ena_installed
echo "kmod-amazon-ena" >> /home/koko/lede/staging_dir/target-x86_64_musl/pkginfo/linux.default.install
mkdir -p /home/koko/lede/staging_dir/target-x86_64_musl/root-x86/stamp

SHELL= flock /home/koko/lede/tmp/.root-copy.flock -c 'cp -fpR /home/koko/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/.pkgdir/kmod-amd-xgbe/. /home/koko/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/koko/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.kmod-amd-xgbe_installed
echo "kmod-amd-xgbe" >> /home/koko/lede/staging_dir/target-x86_64_musl/pkginfo/linux.default.install
make[3]: Leaving directory '/home/koko/lede/package/kernel/linux'
time: package/kernel/linux/compile#16.57#6.90#55.70
make[2]: Leaving directory '/home/koko/lede'
make[1]: *** [package/Makefile:110: /home/koko/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/koko/lede'
make: *** [/home/koko/lede/include/toplevel.mk:230: world] Error 2
koko@koko:~/lede$

WYC-2020, 2022-04-06:
日志贴的不对

a7502, 2022-04-06:
日志在那呢，我是小白。。。

a7502, 2022-04-06:
小白，不知道日志在那，大神指导下

GreatMichaelLee, 2022-04-06:
> 小白，不知道日志在那，大神指导下

-j1

WYC-2020, 2022-04-06:
单线程日志说了几千万次了，难道非要打一个公告

a7502, 2022-04-07:
znow -zrelro" GOPATH="/home/koko/lede/build_dir/target-x86_64_musl/v2ray-core-5.0.3/.go_work/build" GOCACHE="/home/koko/lede/tmp/go-build" GOMODCACHE="/home/koko/lede/dl/go-mod-cache" GOENV=off /usr/bin/env bash /home/koko/lede/feeds/packages/lang/golang//golang-build.sh build -v -trimpath -ldflags "all=-buildid '1649264465' -linkmode external -extldflags '-L/home/koko/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/koko/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -Wl,-z,now -Wl,-z,relro'"        -ldflags " -X github.com/v2fly/v2ray-core/v5.build=OpenWrt -X github.com/v2fly/v2ray-core/v5.version=5.0.3 -buildid '1649264465' -linkmode external -extldflags '-L/home/koko/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/koko/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -Wl,-z,now -Wl,-z,relro'"
Finding targets

Building targets
github.com/marten-seemann/qtls-go1-18
# github.com/marten-seemann/qtls-go1-18
../../../../../dl/go-mod-cache/github.com/marten-seemann/qtls-go1-18@v0.1.1/cipher_suites.go:141:43: undefined: any
../../../../../dl/go-mod-cache/github.com/marten-seemann/qtls-go1-18@v0.1.1/cipher_suites.go:399:45: undefined: any
../../../../../dl/go-mod-cache/github.com/marten-seemann/qtls-go1-18@v0.1.1/cipher_suites.go:404:46: undefined: any
../../../../../dl/go-mod-cache/github.com/marten-seemann/qtls-go1-18@v0.1.1/cipher_suites.go:412:45: undefined: any
../../../../../dl/go-mod-cache/github.com/marten-seemann/qtls-go1-18@v0.1.1/common.go:1496:41: undefined: any
../../../../../dl/go-mod-cache/github.com/marten-seemann/qtls-go1-18@v0.1.1/conn.go:177:10: undefined: any
../../../../../dl/go-mod-cache/github.com/marten-seemann/qtls-go1-18@v0.1.1/conn.go:183:13: undefined: any
../../../../../dl/go-mod-cache/github.com/marten-seemann/qtls-go1-18@v0.1.1/conn.go:211:62: undefined: any
../../../../../dl/go-mod-cache/github.com/marten-seemann/qtls-go1-18@v0.1.1/conn.go:973:14: undefined: any
../../../../../dl/go-mod-cache/github.com/marten-seemann/qtls-go1-18@v0.1.1/conn.go:1056:33: undefined: any
../../../../../dl/go-mod-cache/github.com/marten-seemann/qtls-go1-18@v0.1.1/conn.go:973:14: too many errors
note: module requires Go 1.18

make[3]: *** [Makefile:111: /home/koko/lede/build_dir/target-x86_64_musl/v2ray-core-5.0.3/.built] Error 2
make[3]: Leaving directory '/home/koko/lede/feeds/small/v2ray-core'
time: package/feeds/small/v2ray-core/compile#1.80#3.13#1.54
    ERROR: package/feeds/small/v2ray-core failed to build.
make[2]: *** [package/Makefile:116: package/feeds/small/v2ray-core/compile] Error 1
make[2]: Leaving directory '/home/koko/lede'
make[1]: *** [package/Makefile:110: /home/koko/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/koko/lede'
make: *** [/home/koko/lede/include/toplevel.mk:230：world] 错误 2
koko@koko-virtual-machine:~/lede$
用的是https://github.com/WYC-2020/openwrt-packages插件库，新手多多指教，应该怎么搞

WYC-2020, 2022-04-07:
我仓库没升级go到1.8的，自己升级或者去用官方的仓库

> znow -zrelro" GOPATH="/home/koko/lede/build_dir/target-x86_64_musl/v2ray-core-5.0.3/.go_work/build" GOCACHE="/home/koko/lede/tmp/go-build" GOMODCACHE="/home/koko/lede/dl/go-mod-cache" GOENV=off [...]

WYC-2020, 2022-04-07:
> znow -zrelro" GOPATH="/home/koko/lede/build_dir/target-x86_64_musl/v2ray-core-5.0.3/.go_work/build" GOCACHE="/home/koko/lede/tmp/go-build" GOMODCACHE="/home/koko/lede/dl/go-mod-cache" GOENV=off [...]

你这是是编译v2出错了，你的更新下你的passwalll代码或者用我的openwrt-passwall和我的仓库配套，即可

WYC-2020, 2022-04-07:
> znow -zrelro" GOPATH="/home/koko/lede/build_dir/target-x86_64_musl/v2ray-core-5.0.3/.go_work/build" GOCACHE="/home/koko/lede/tmp/go-build" GOMODCACHE="/home/koko/lede/dl/go-mod-cache" GOENV=off [...]

以后用我仓库出问题，可以去我那边提，免得别人吐槽你，我也不想被吐槽