coolsnowwolf/lede #9201: qnap-301w最新源码编译的固件pppoe无法拨号

zyh6111, 2022-04-07:
详细叙述：
qnap-301w最新源码编译的固件pppoe无法拨号（wan口dhcp也无法获取ip地址），万兆网卡用的是AQR_ethphyfw_5.5.6.mbn公版驱动，已硬改2G内存。

 详细日志：
=~=~=~=~=~=~=~=~=~=~=~= PuTTY log 2022.04.07 21:06:38 =~=~=~=~=~=~=~=~=~=~=~=
 
Format: Log Type - Time(microsec) - Message - Optional Info
Log Type: B - Since Boot(Power On Reset),  D - Delta,  S - Statistic
S - QC_IMAGE_VERSION_STRING=BOOT.BF.3.3.1-00158
S - IMAGE_VARIANT_STRING=HAABANAZA
S - OEM_IMAGE_VERSION_STRING=CRM
S - Boot Config, 0x000002e1
B -       201 - PBL, Start
B -      2735 - bootable_media_detect_entry, Start
B -      2881 - bootable_media_detect_success, Start
B -      2886 - elf_loader_entry, Start
B -     11375 - auth_hash_seg_entry, Start
B -     11617 - auth_hash_seg_exit, Start
B -    250653 - elf_segs_hash_verify_entry, Start
B -    311323 - PBL, End
B -    330528 - SBL1, Start
B -    383659 - GCC [RstStat:0x10, RstDbg:0x600000] WDog Stat : 0x4
B -    390247 - pm_device_init, Start
B -    513894 - PM_SET_VAL:Skip
D -    123006 - pm_device_init, Delta
B -    516304 - pm_driver_init, Start
D -      5337 - pm_driver_init, Delta
B -    522617 - clock_init, Start
D -      2104 - clock_init, Delta
B -    526613 - boot_flash_init, Start
D -     15524 - boot_flash_init, Delta
B -    545919 - boot_config_data_table_init, Start
D -      1006 - boot_config_data_table_init, Delta - (575 Bytes)
B -    553575 - Boot Setting :  0x00000618
B -    557326 - CDT version:2,Platform ID:8,Major ID:1,Minor ID:0,Subtype:0
B -    564250 - sbl1_ddr_set_params, Start
B -    568062 - CPR configuration: 0x30c
B -    571509 - cpr_init, Start
B -    574284 - Rail:0 Mode: 5 Voltage: 784000
B -    579500 - CL CPR settled at 736000mV
B -    582336 - Rail:1 Mode: 5 Voltage: 880000
B -    586515 - Rail:1 Mode: 7 Voltage: 888000
D -     16531 - cpr_init, Delta
B -    593408 - Pre_DDR_clock_init, Start
B -    597403 - Pre_DDR_clock_init, End
B -    600697 - DDR Type : PCDDR3
B -    606431 - do ddr sanity test, Start
D -      1067 - do ddr sanity test, Delta
B -    611281 - DDR: Start of HAL DDR Boot Training
B -    615917 - DDR: End of HAL DDR Boot Training
B -    621620 - DDR: Checksum to be stored on flash is -1623478298
B -    632112 - Image Load, Start
D -    336202 - QSEE Image Loaded, Delta - (1376448 Bytes)
B -    968405 - Image Load, Start
D -        30 - SEC Image Loaded, Delta - (0 Bytes)
B -    976091 - Image Load, Start
D -      9760 - DEVCFG Image Loaded, Delta - (26008 Bytes)
B -    985943 - Image Load, Start
D -     25955 - RPM Image Loaded, Delta - (86584 Bytes)
B -   1011990 - Image Load, Start
D -    136670 - APPSBL Image Loaded, Delta - (556008 Bytes)
B -   1148782 - QSEE Execution, Start
D -        61 - QSEE Execution, Delta
B -   1154608 - USB D+ check, Start
D -         0 - USB D+ check, Delta
B -   1160982 - SBL1, End
D -    832772 - SBL1, Delta
S - Flash Throughput, 4430 KB/s  (2045623 Bytes,  461762 us)
S - DDR Frequency, 466 MHz
S - Core 0 Frequency, 1651 MHz
[... 481 lines cut ...]
[   18.157487] br-lan: port 6(wlan1) entered forwarding state
[   18.163640] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   18.267649] squashfs image failed sanity check
[   18.270018] EXT4-fs (mmcblk0p8): warning: mounting fs with errors, running e2fsck is recommended
[   18.272509] EXT4-fs (mmcblk0p8): recovery complete
[   18.279946] EXT4-fs (mmcblk0p8): mounted filesystem with ordered data mode. Opts: 
[   18.284573] ext4 filesystem being mounted at /mnt/mmcblk0p8 supports timestamps until 2038 (0x7fffffff)
[   18.322388] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[   18.395719] br-lan: port 7(wlan0) entered blocking state
[   18.395753] br-lan: port 7(wlan0) entered disabled state
[   18.400423] device wlan0 entered promiscuous mode
[   18.405563] br-lan: port 7(wlan0) entered blocking state
[   18.410066] br-lan: port 7(wlan0) entered forwarding state
[   19.098225] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   21.504811] nss-dp 3a007000.dp6 eth5: PHY Link up speed: 1000
[   21.504885] IPv6: ADDRCONF(NETDEV_CHANGE): eth5: link becomes ready
[   23.520855] nss-dp 3a001800.dp5 eth4: PHY Link up speed: 2500
[   23.520931] br-lan: port 5(eth4) entered blocking state
[   23.525649] br-lan: port 5(eth4) entered forwarding state
[   87.325390] pppoe-wan: renamed from ppp0
[  160.161353] pppoe-wan: renamed from ppp0
[  291.533337] pppoe-wan: renamed from ppp0
[  332.768423] EXT4-fs (mmcblk0p8): error count since last fsck: 1
[  332.768460] EXT4-fs (mmcblk0p8): initial error at time 1603937682: ext4_has_uninit_itable:2830
[  332.773170] EXT4-fs (mmcblk0p8): last error at time 1603937682: ext4_has_uninit_itable:2830

WYC-2020, 2022-04-08:
十之八九硬改的锅哈哈

zyh6111, 2022-04-08:
> 十之八九硬改的锅哈哈

换回原厂1G的cdt也不行。

0x5826, 2022-04-09:
应该跟防火墙规则下发有问题，配置完网口保存生效之后，可以尝试重启路由器或者`/etc/init.d/firewall restart`看看？

yyleis, 2022-11-12:
最新编译固件PPPOE无法拨号的现象好像还在

closed 2023-12-22 by aiamadeus