coolsnowwolf/lede #9205: glib error

naoki66, 2022-04-08:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述
编译X64固件失败

### (1) 具体问题

A：编译失败

### (2) 路由器型号和固件版本

A：X64

### (3) 详细日志

A：/HOME/lede/staging_dir/host/bin/ninja  -j1 -C /HOME/lede/build_dir/hostpkg/glib-2.70.4/openwrt-build
ninja: Entering directory `/HOME/lede/build_dir/hostpkg/glib-2.70.4/openwrt-build'
[1/356] Linking target glib/gtester
FAILED: glib/gtester
gcc  -o glib/gtester glib/gtester.p/gtester.c.o -L/HOME/lede/staging_dir/host/lib -L/HOME/lede/staging_dir/hostpkg/lib -L/HOME/lede/staging_dir/target-x86_64_musl/host/lib -Wl,--as-needed -Wl,--no-undefined -Wl,-rpath,/HOME/lede/staging_dir/hostpkg/lib -Wl,--start-group glib/libglib-2.0.a /HOME/lede/staging_dir/hostpkg/lib/libpcre.so -pthread -liconv -lm -Wl,--end-group
/usr/bin/ld: glib/libglib-2.0.a(ggettext.c.o): in function `_g_dgettext_should_translate':
/HOME/lede/build_dir/hostpkg/glib-2.70.4/openwrt-build/../glib/ggettext.c:315: undefined reference to `g_libintl_textdomain'
/usr/bin/ld: /HOME/lede/build_dir/hostpkg/glib-2.70.4/openwrt-build/../glib/ggettext.c:316: undefined reference to `g_libintl_gettext'
/usr/bin/ld: glib/libglib-2.0.a(ggettext.c.o): in function `ensure_gettext_initialized':
/HOME/lede/build_dir/hostpkg/glib-2.70.4/openwrt-build/../glib/ggettext.c:107: undefined reference to `g_libintl_bindtextdomain'
/usr/bin/ld: /HOME/lede/build_dir/hostpkg/glib-2.70.4/openwrt-build/../glib/ggettext.c:110: undefined reference to `g_libintl_bind_textdomain_codeset'
/usr/bin/ld: glib/libglib-2.0.a(ggettext.c.o): in function `g_dgettext':
/HOME/lede/build_dir/hostpkg/glib-2.70.4/openwrt-build/../glib/ggettext.c:402: undefined reference to `g_libintl_dgettext'
/usr/bin/ld: glib/libglib-2.0.a(ggettext.c.o): in function `g_dcgettext':
/HOME/lede/build_dir/hostpkg/glib-2.70.4/openwrt-build/../glib/ggettext.c:429: undefined reference to `g_libintl_dcgettext'
/usr/bin/ld: glib/libglib-2.0.a(ggettext.c.o): in function `g_dngettext':
/HOME/lede/build_dir/hostpkg/glib-2.70.4/openwrt-build/../glib/ggettext.c:460: undefined reference to `g_libintl_dngettext'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
make[3]: *** [Makefile:128: /HOME/lede/build_dir/hostpkg/glib-2.70.4/.built] Error 1
make[3]: Leaving directory '/HOME/lede/feeds/packages/libs/glib2'
time: package/feeds/packages/glib2/host-compile#0.33#0.08#0.39
    ERROR: package/feeds/packages/glib2 [host] failed to build.
make[2]: *** [package/Makefile:114: package/feeds/packages/glib2/host/compile] Error 1
make[2]: Leaving directory '/HOME/lede'
make[1]: *** [package/Makefile:110: /HOME/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/HOME/lede'
make: *** [/HOME/lede/include/toplevel.mk:230: world] Error 2

WYC-2020, 2022-04-08:
万能大法

naoki66, 2022-04-08:
已经rm -rf了

| |
***@***.***
|
|
***@***.***
|

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 日期 | 2022年04月08日 20:39 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.******@***.***> |
| 主题 | Re: [coolsnowwolf/lede] glib error (Issue #9205) |

万能大法

closed 2022-07-25 by aiamadeus