coolsnowwolf/lede #9206: grub2报错

guochan2019, 2022-04-08:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：4月2日的更新编译正常，4月8日更新貌似报grub2错误

### (2) 路由器型号和固件版本

A：x86

### (3) 详细日志

A：
make[7]: Leaving directory '/home/hellomax/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.06/grub-core'
make[6]: Leaving directory '/home/hellomax/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.06/grub-core'
Making all in po
make[6]: Entering directory '/home/hellomax/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.06/po'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/hellomax/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.06/po'
Making all in docs
make[6]: Entering directory '/home/hellomax/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.06/docs'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/hellomax/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.06/docs'
Making all in util/bash-completion.d
make[6]: Entering directory '/home/hellomax/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.06/util/bash-completion.d'
../../config.status --file=grub:grub-completion.bash.in
config.status: creating grub
make[6]: Leaving directory '/home/hellomax/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.06/util/bash-completion.d'
make[5]: Leaving directory '/home/hellomax/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.06'
make[4]: Leaving directory '/home/hellomax/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.06'
touch /home/hellomax/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.06/.built
mkdir -p /home/hellomax/lede/bin/targets/x86/64/packages /home/hellomax/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.06/ipkg-x86_64/grub2-efi/CONTROL /home/hellomax/lede/staging_dir/target-x86_64_musl/pkginfo
sed 's#msdos1#gpt1#g' ./files/grub-early.cfg >/home/hellomax/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.06/grub-early.cfg
/home/hellomax/lede/staging_dir/host/bin/grub-mkimage -d /home/hellomax/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.06/grub-core -p /boot/grub -O "x86_64"-efi -c /home/hellomax/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.06/grub-early.cfg -o /home/hellomax/lede/staging_dir/target-x86_64_musl/image/grub2/bootx64.efi at_keyboard boot chain configfile fat linux ls part_gpt reboot serial efi_gop efi_uga
/home/hellomax/lede/staging_dir/host/bin/grub-mkimage -d /home/hellomax/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.06/grub-core -p /boot/grub -O "x86_64"-efi -c ./files/grub-early.cfg -o /home/hellomax/lede/staging_dir/target-x86_64_musl/image/grub2/iso-bootx64.efi at_keyboard boot chain configfile fat iso9660 linux ls part_msdos part_gpt reboot serial test efi_gop efi_uga
find /home/hellomax/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.06/ipkg-x86_64/grub2-efi -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
export CROSS="x86_64-openwrt-linux-musl-"  NO_RENAME=1 ; NM="x86_64-openwrt-linux-musl-nm" STRIP="/home/hellomax/lede/staging_dir/host/bin/sstrip -z" STRIP_KMOD="/home/hellomax/lede/scripts/strip-kmod.sh" PATCHELF="/home/hellomax/lede/staging_dir/host/bin/patchelf" /home/hellomax/lede/scripts/rstrip.sh /home/hellomax/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.06/ipkg-x86_64/grub2-efi
(cd /home/hellomax/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.06/ipkg-x86_64/grub2-efi/CONTROL; ( echo "$CONTROL"; printf "Description: "; echo "$DESCRIPTION" | sed -e 's,^[[:space:]]*, ,g'; ) > control; chmod 644 control; ( echo "#!/bin/sh"; echo "[ \"\${IPKG_NO_SCRIPT}\" = \"1\" ] && exit 0"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_postinst \$0 \$@"; ) > postinst; ( echo "#!/bin/sh"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_prerm \$0 \$@"; ) > prerm; chmod 0755 postinst prerm;  )
install -d -m0755 /home/hellomax/lede/bin/targets/x86/64/packages
/home/hellomax/lede/staging_dir/host/bin/fakeroot /home/hellomax/lede/scripts/ipkg-build -m "" /home/hellomax/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.06/ipkg-x86_64/grub2-efi /home/hellomax/lede/bin/targets/x86/64/packages
Packaged contents of /home/hellomax/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.06/ipkg-x86_64/grub2-efi into /home/hellomax/lede/bin/targets/x86/64/packages/grub2-efi_2.06-1_x86_64.ipk
rm -rf /home/hellomax/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.06/.pkgdir/grub2-efi.installed /home/hellomax/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.06/.pkgdir/grub2-efi
mkdir -p /home/hellomax/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.06/.pkgdir/grub2-efi
sed 's#msdos1#gpt1#g' ./files/grub-early.cfg >/home/hellomax/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.06/grub-early.cfg
/home/hellomax/lede/staging_dir/host/bin/grub-mkimage -d /home/hellomax/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.06/grub-core -p /boot/grub -O "x86_64"-efi -c /home/hellomax/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.06/grub-early.cfg -o /home/hellomax/lede/staging_dir/target-x86_64_musl/image/grub2/bootx64.efi at_keyboard boot chain configfile fat linux ls part_gpt reboot serial efi_gop efi_uga
/home/hellomax/lede/staging_dir/host/bin/grub-mkimage -d /home/hellomax/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.06/grub-core -p /boot/grub -O "x86_64"-efi -c ./files/grub-early.cfg -o /home/hellomax/lede/staging_dir/target-x86_64_musl/image/grub2/iso-bootx64.efi at_keyboard boot chain configfile fat iso9660 linux ls part_msdos part_gpt reboot serial test efi_gop efi_uga
touch /home/hellomax/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.06/.pkgdir/grub2-efi.installed
mkdir -p /home/hellomax/lede/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/hellomax/lede/tmp/.root-copy.flock -c 'cp -fpR /home/hellomax/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.06/.pkgdir/grub2-efi/. /home/hellomax/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/hellomax/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.grub2-efi_installed
echo "grub2-efi" >> /home/hellomax/lede/staging_dir/target-x86_64_musl/pkginfo/grub2.efi.install
make[3]: Leaving directory '/home/hellomax/lede/package/boot/grub2'
time: package/boot/grub2/efi/compile#77.10#13.83#86.87
make[2]: Leaving directory '/home/hellomax/lede'
make[1]: *** [package/Makefile:110: /home/hellomax/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/hellomax/lede'
make: *** [/home/hellomax/lede/include/toplevel.mk:230: world] Error 2

WYC-2020, 2022-04-08:
万能大法

hcym, 2022-04-08:
今天早上，不是，是昨天上午编译的正常，以前也从来没有这个问题

guochan2019, 2022-04-08:
> 万能大法

老大请明示

minlang112, 2022-04-09:
> > 万能大法
> 
> 老大请明示
make clean
make distclean

guochan2019, 2022-04-09:
> > > 万能大法
> > 
> > 
> > 老大请明示
> > make clean
> > make distclean

真心不想重心来过，make dirclean解决了，谢谢

closed 2022-04-10 by guochan2019