coolsnowwolf/lede #9207: 路由器每天晚上自动重启（断开）无线

pk49800, 2022-04-08:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

路由器每到晚上10点38分会自动重启（断开）无线网络，计划任务和定时重启并未开启此重启任务

### (2) 路由器型号和固件版本

Netgear Nighthawk X4S R7800 

固件版本  OpenWrt R22.4.1 / LuCI Master (git-22.095.24346-72703c8)
内核版本  5.4.188

### (3) 详细日志

Fri Apr  8 18:00:58 2022 daemon.notice hostapd: wlan1: STA-OPMODE-N_SS-CHANGED 94:65:2d:8f:f3:57 2
Fri Apr  8 18:00:58 2022 daemon.notice hostapd: wlan1: STA-OPMODE-N_SS-CHANGED 94:65:2d:8f:f3:57 1
Fri Apr  8 18:00:58 2022 daemon.notice hostapd: wlan1: STA-OPMODE-N_SS-CHANGED 94:65:2d:8f:f3:57 2
Fri Apr  8 18:00:58 2022 daemon.notice hostapd: wlan1: STA-OPMODE-N_SS-CHANGED 94:65:2d:8f:f3:57 1
Fri Apr  8 18:01:28 2022 daemon.notice hostapd: wlan1: STA-OPMODE-N_SS-CHANGED 94:65:2d:8f:f3:57 2
Fri Apr  8 18:01:49 2022 daemon.notice hostapd: wlan1: STA-OPMODE-N_SS-CHANGED 94:65:2d:8f:f3:57 1
Fri Apr  8 18:01:49 2022 daemon.notice hostapd: wlan1: STA-OPMODE-N_SS-CHANGED 94:65:2d:8f:f3:57 2
Fri Apr  8 18:03:21 2022 daemon.notice hostapd: wlan1: STA-OPMODE-N_SS-CHANGED 94:65:2d:8f:f3:57 1
Fri Apr  8 18:03:22 2022 daemon.notice hostapd: wlan1: STA-OPMODE-N_SS-CHANGED 94:65:2d:8f:f3:57 2
Fri Apr  8 18:04:23 2022 daemon.notice hostapd: wlan1: STA-OPMODE-N_SS-CHANGED 94:65:2d:8f:f3:57 1
Fri Apr  8 18:04:23 2022 daemon.notice hostapd: wlan1: STA-OPMODE-N_SS-CHANGED 94:65:2d:8f:f3:57 2
Fri Apr  8 18:05:01 2022 daemon.notice hostapd: wlan1: STA-OPMODE-N_SS-CHANGED 94:65:2d:8f:f3:57 1
Fri Apr  8 18:05:19 2022 daemon.notice hostapd: wlan1: STA-OPMODE-N_SS-CHANGED 94:65:2d:8f:f3:57 2
Fri Apr  8 18:07:31 2022 daemon.notice hostapd: wlan1: STA-OPMODE-N_SS-CHANGED 94:65:2d:8f:f3:57 1
Fri Apr  8 18:07:31 2022 daemon.notice hostapd: wlan1: STA-OPMODE-N_SS-CHANGED 94:65:2d:8f:f3:57 2
Fri Apr  8 18:07:46 2022 daemon.notice hostapd: wlan1: STA-OPMODE-N_SS-CHANGED 94:65:2d:8f:f3:57 1
Fri Apr  8 18:07:46 2022 daemon.notice hostapd: wlan1: STA-OPMODE-N_SS-CHANGED 94:65:2d:8f:f3:57 2
Fri Apr  8 18:07:48 2022 daemon.notice hostapd: wlan1: STA-OPMODE-N_SS-CHANGED 94:65:2d:8f:f3:57 1
Fri Apr  8 18:07:48 2022 daemon.notice hostapd: wlan1: STA-OPMODE-N_SS-CHANGED 94:65:2d:8f:f3:57 2
Fri Apr  8 18:07:49 2022 daemon.notice hostapd: wlan1: STA-OPMODE-N_SS-CHANGED 94:65:2d:8f:f3:57 1
Fri Apr  8 18:08:31 2022 daemon.notice hostapd: wlan1: STA-OPMODE-N_SS-CHANGED 94:65:2d:8f:f3:57 2
Fri Apr  8 18:08:55 2022 daemon.notice hostapd: wlan1: STA-OPMODE-N_SS-CHANGED 94:65:2d:8f:f3:57 1
Fri Apr  8 18:08:55 2022 daemon.notice hostapd: wlan1: STA-OPMODE-N_SS-CHANGED 94:65:2d:8f:f3:57 2
Fri Apr  8 18:08:55 2022 daemon.notice hostapd: wlan1: STA-OPMODE-N_SS-CHANGED 94:65:2d:8f:f3:57 1
Fri Apr  8 18:08:55 2022 daemon.notice hostapd: wlan1: STA-OPMODE-N_SS-CHANGED 94:65:2d:8f:f3:57 2
Fri Apr  8 18:10:00 2022 daemon.notice hostapd: wlan1: STA-OPMODE-N_SS-CHANGED 94:65:2d:8f:f3:57 1
Fri Apr  8 18:13:37 2022 daemon.notice hostapd: wlan1: STA-OPMODE-N_SS-CHANGED 94:65:2d:8f:f3:57 2
Fri Apr  8 18:13:39 2022 daemon.notice hostapd: wlan1: STA-OPMODE-N_SS-CHANGED 94:65:2d:8f:f3:57 1
Fri Apr  8 18:13:39 2022 daemon.notice hostapd: wlan1: STA-OPMODE-N_SS-CHANGED 94:65:2d:8f:f3:57 2
Fri Apr  8 18:14:30 2022 daemon.notice hostapd: wlan1: STA-OPMODE-N_SS-CHANGED 94:65:2d:8f:f3:57 1
Fri Apr  8 18:17:58 2022 daemon.notice hostapd: wlan1: STA-OPMODE-N_SS-CHANGED 94:65:2d:8f:f3:57 2
Fri Apr  8 18:18:15 2022 daemon.notice hostapd: wlan1: STA-OPMODE-N_SS-CHANGED 94:65:2d:8f:f3:57 1
Fri Apr  8 18:18:16 2022 daemon.notice hostapd: wlan1: STA-OPMODE-N_SS-CHANGED 94:65:2d:8f:f3:57 2
Fri Apr  8 18:18:16 2022 daemon.notice hostapd: wlan1: STA-OPMODE-N_SS-CHANGED 94:65:2d:8f:f3:57 1
Fri Apr  8 18:18:16 2022 daemon.notice hostapd: wlan1: STA-OPMODE-N_SS-CHANGED 94:65:2d:8f:f3:57 2
Fri Apr  8 18:18:39 2022 daemon.notice hostapd: wlan1: STA-OPMODE-N_SS-CHANGED 94:65:2d:8f:f3:57 1
Fri Apr  8 18:18:41 2022 daemon.notice hostapd: wlan1: STA-OPMODE-N_SS-CHANGED 94:65:2d:8f:f3:57 2
Fri Apr  8 18:18:42 2022 daemon.notice hostapd: wlan1: STA-OPMODE-N_SS-CHANGED 94:65:2d:8f:f3:57 1
Fri Apr  8 18:18:44 2022 daemon.notice hostapd: wlan1: STA-OPMODE-N_SS-CHANGED 94:65:2d:8f:f3:57 2
Fri Apr  8 18:18:44 2022 daemon.notice hostapd: wlan1: STA-OPMODE-N_SS-CHANGED 94:65:2d:8f:f3:57 1
Fri Apr  8 18:18:44 2022 daemon.notice hostapd: wlan1: STA-OPMODE-N_SS-CHANGED 94:65:2d:8f:f3:57 2
Fri Apr  8 18:18:58 2022 daemon.notice hostapd: wlan1: STA-OPMODE-N_SS-CHANGED 94:65:2d:8f:f3:57 1
Fri Apr  8 18:18:58 2022 daemon.notice hostapd: wlan1: STA-OPMODE-N_SS-CHANGED 94:65:2d:8f:f3:57 2
Fri Apr  8 18:18:58 2022 daemon.notice hostapd: wlan1: STA-OPMODE-N_SS-CHANGED 94:65:2d:8f:f3:57 1
Fri Apr  8 18:18:58 2022 daemon.notice hostapd: wlan1: STA-OPMODE-N_SS-CHANGED 94:65:2d:8f:f3:57 2
Fri Apr  8 18:19:02 2022 daemon.notice hostapd: wlan1: STA-OPMODE-N_SS-CHANGED 94:65:2d:8f:f3:57 1
Fri Apr  8 18:19:02 2022 daemon.notice hostapd: wlan1: STA-OPMODE-N_SS-CHANGED 94:65:2d:8f:f3:57 2
Fri Apr  8 18:19:02 2022 daemon.notice hostapd: wlan1: STA-OPMODE-N_SS-CHANGED 94:65:2d:8f:f3:57 1
[... 252 lines cut ...]
Fri Apr  8 22:38:09 2022 daemon.info avahi-daemon[2494]: Joining mDNS multicast group on interface pppoe-wan.IPv4 with address 119.144.14.95.
Fri Apr  8 22:38:09 2022 daemon.info avahi-daemon[2494]: New relevant interface pppoe-wan.IPv4 for mDNS.
Fri Apr  8 22:38:09 2022 daemon.info avahi-daemon[2494]: Registering new address record for 119.144.14.95 on pppoe-wan.IPv4.
Fri Apr  8 22:38:09 2022 daemon.notice netifd: Network device 'pppoe-wan' link is up
Fri Apr  8 22:38:09 2022 daemon.notice netifd: Network alias 'pppoe-wan' link is up
Fri Apr  8 22:38:09 2022 daemon.notice netifd: Interface 'wan' is now up
Fri Apr  8 22:38:09 2022 daemon.notice netifd: Interface 'wan_6' is enabled
Fri Apr  8 22:38:09 2022 daemon.notice netifd: Interface 'wan_6' has link connectivity
Fri Apr  8 22:38:09 2022 daemon.notice netifd: Interface 'wan_6' is setting up now
Fri Apr  8 22:38:09 2022 daemon.notice netifd: Interface 'wan_6' is now up
Fri Apr  8 22:38:10 2022 daemon.info hostapd: wlan0: STA 38:00:25:aa:91:a5 WPA: group key handshake failed (RSN) after 4 tries
Fri Apr  8 22:38:10 2022 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 38:00:25:aa:91:a5
Fri Apr  8 22:38:10 2022 daemon.info hostapd: wlan0: STA 38:00:25:aa:91:a5 IEEE 802.11: authenticated
Fri Apr  8 22:38:10 2022 daemon.info hostapd: wlan0: STA 38:00:25:aa:91:a5 IEEE 802.11: associated (aid 4)
Fri Apr  8 22:38:11 2022 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 38:00:25:aa:91:a5
Fri Apr  8 22:38:11 2022 daemon.info hostapd: wlan0: STA 38:00:25:aa:91:a5 RADIUS: starting accounting session 4DD294DF1DA6CD86
Fri Apr  8 22:38:11 2022 daemon.info hostapd: wlan0: STA 38:00:25:aa:91:a5 WPA: pairwise key handshake completed (RSN)
Fri Apr  8 22:38:11 2022 daemon.notice hostapd: wlan0: EAPOL-4WAY-HS-COMPLETED 38:00:25:aa:91:a5
Fri Apr  8 22:38:19 2022 user.notice firewall: Reloading firewall due to ifup of wan (pppoe-wan)
Fri Apr  8 22:38:21 2022 cron.err crond[25993]: crond (busybox 1.35.0) started, log level 8
Fri Apr  8 22:38:22 2022 cron.err crond[26211]: crond (busybox 1.35.0) started, log level 8
Fri Apr  8 22:38:22 2022 user.notice nlbwmon: Reloading nlbwmon due to ifup of wan (pppoe-wan)
Fri Apr  8 22:38:23 2022 user.notice firewall: Reloading firewall due to ifupdate of wan (pppoe-wan)
Fri Apr  8 22:38:24 2022 cron.err crond[26907]: crond (busybox 1.35.0) started, log level 8
Fri Apr  8 22:38:25 2022 cron.err crond[27130]: crond (busybox 1.35.0) started, log level 8

GreatMichaelLee, 2022-04-09:
Fri Apr 8 22:38:05 2022 daemon.info pppd[8374]: LCP terminated by peer
Fri Apr 8 22:38:05 2022 daemon.info pppd[8374]: Connect time 1440.0 minutes.
Fri Apr 8 22:38:05 2022 daemon.info pppd[8374]: Sent 379816285 bytes, received 2359114880 bytes.

每1440分钟你被运营商结束一下LCP会话，运营商的策略而已。不过我们这都是48或者72的，你这个24小时有点夸张了。

pk49800, 2022-04-09:
> Fri Apr 8 22:38:05 2022 daemon.info pppd[8374]: LCP terminated by peer Fri Apr 8 22:38:05 2022 daemon.info pppd[8374]: Connect time 1440.0 minutes. Fri Apr 8 22:38:05 2022 daemon.info pppd[8374]: Sent 379816285 bytes, received 2359114880 bytes.
> 
> 每1440分钟你被运营商结束一下PPP/LCP会话，让你重新拨号而已，运营商为了防止你长期占用固定公网IP的下作策略而已。不过我们这都是48或者72的，你这个24小时有点夸张了。

那如果我定时重启路由器，就相当于重新拨号了吧？那就不会在我要用的时候重新拨号？

GreatMichaelLee, 2022-04-13:
> 那如果我定时重启路由器，就相当于重新拨号了吧？那就不会在我要用的时候重新拨号？

是的，看你重启时间和他踢你下线时间哪个先到了。

coolsnowwolf, 2026-09-05:
R7800 每晚无线中断没有对应时间点的完整 hostapd/kernel/netifd 日志或定时任务证据，无法确认是驱动还是外部计划任务，关闭为“不计划处理”。

closed 2026-09-05 by coolsnowwolf