coolsnowwolf/lede #9211: 報告編譯出錯

alan21j, 2022-04-09:
-- Build files have been written to: /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-03-29-eed44048/tools
touch /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-03-29-eed44048/.configured_b1c6ae92f8e974e60b47b3e5ad3573cc
rm -f /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-03-29-eed44048/.built
touch /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-03-29-eed44048/.built_check
make  -C "/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.15.32" KCFLAGS="-fmacro-prefix-map=/workdir/openwrt/build_dir/target-mipsel_24kc_musl=target-mipsel_24kc_musl" HOSTCFLAGS="-O2 -I/workdir/openwrt/staging_dir/host/include -I/workdir/openwrt/staging_dir/hostpkg/include -I/workdir/openwrt/staging_dir/target-mipsel_24kc_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="mipsel-openwrt-linux-musl-" ARCH="mips" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Thu Apr  7 15:37:19 2022" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/workdir/openwrt/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/workdir/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KBUILD_EXTRA_SYMBOLS="/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/symvers/cryptodev-linux.symvers /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/symvers/gpio-button-hotplug.symvers /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/symvers/mac80211.symvers" KERNELRELEASE=5.15.32 CONFIG_MT76_CONNAC_LIB=m CONFIG_MT7615_COMMON=m CONFIG_MT7915E=m M="/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-03-29-eed44048" NOSTDINC_FLAGS="-nostdinc  -isystem /workdir/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/lib/gcc/mipsel-openwrt-linux-musl/8.4.0/include -I/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-03-29-eed44048 -I/workdir/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/mac80211-backport/uapi -I/workdir/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/mac80211-backport -I/workdir/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/mac80211/uapi -I/workdir/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/mac80211 -include backport/autoconf.h -include backport/backport.h -DCONFIG_MAC80211_MESH" modules
make[4]: Entering directory '/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.15.32'
  CC [M]  /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-03-29-eed44048/mt7915/pci.o
/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-03-29-eed44048/mt7915/pci.c: In function 'mt7915_pci_probe':
/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-03-29-eed44048/mt7915/pci.c:132:42: error: 'PCI_IRQ_ALL_TYPES' undeclared (first use in this function); did you mean 'NUM_RFKILL_TYPES'?
  ret = pci_alloc_irq_vectors(pdev, 1, 1, PCI_IRQ_ALL_TYPES);
                                          ^~~~~~~~~~~~~~~~~
                                          NUM_RFKILL_TYPES
/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-03-29-eed44048/mt7915/pci.c:132:42: note: each undeclared identifier is reported only once for each function it appears in
/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-03-29-eed44048/mt7915/pci.c:178:2: error: implicit declaration of function 'pci_free_irq_vectors'; did you mean 'pci_alloc_irq_vectors'? [-Werror=implicit-function-declaration]
  pci_free_irq_vectors(pdev);
  ^~~~~~~~~~~~~~~~~~~~
  pci_alloc_irq_vectors
cc1: some warnings being treated as errors
make[6]: *** [scripts/Makefile.build:277: /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-03-29-eed44048/mt7915/pci.o] Error 1
make[5]: *** [scripts/Makefile.build:540: /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-03-29-eed44048/mt7915] Error 2
make[4]: *** [Makefile:1868: /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-03-29-eed44048] Error 2
make[4]: Leaving directory '/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.15.32'
make[3]: *** [Makefile:469: /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-03-29-eed44048/.built] Error 2
make[3]: Leaving directory '/workdir/openwrt/package/kernel/mt76'
time: package/kernel/mt76/compile#1.76#0.33#1.96
    ERROR: package/kernel/mt76 failed to build.
make[2]: *** [package/Makefile:116: package/kernel/mt76/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.

gswangmd-gssi, 2022-04-11:
编译依然是这种错误，通不过编译。

closed 2022-04-12 by coolsnowwolf

liwenjie119, 2022-04-21:
编译5.15内核的cr660x同样问题，重新拉酷也没解决，请问大家是如何解决的，还是需要向上游反馈？

anjue39, 2022-09-16:
同样问题，复议

Tearknight, 2022-09-25:
同样问题。