coolsnowwolf/lede #9217: 小米路由器3G在内核5.15编译失败

danuonuo, 2022-04-10:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：mir3g在内核5.15编译失败

### (2) 路由器型号和固件版本

A：小米路由器3G MT7621A

### (3) 详细日志

A：/home/openwrt/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-03-29-eed44048/mt76x2/pci.c:181:1: warning: data definition has no type or storage class
 module_pci_driver(mt76pci_driver);
 ^~~~~~~~~~~~~~~~~
/home/openwrt/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-03-29-eed44048/mt76x2/pci.c:181:1: error: type defaults to 'int' in declaration of 'module_pci_driver' [-Werror=implicit-int]
/home/openwrt/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-03-29-eed44048/mt76x2/pci.c:181:1: warning: parameter names (without types) in function declaration
/home/openwrt/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-03-29-eed44048/mt76x2/pci.c:170:26: warning: 'mt76pci_driver' defined but not used [-Wunused-variable]
 static struct pci_driver mt76pci_driver = {
                          ^~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[6]: *** [scripts/Makefile.build:277: /home/openwrt/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-03-29-eed44048/mt76x2/pci.o] Error 1
make[5]: *** [scripts/Makefile.build:540: /home/openwrt/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-03-29-eed44048/mt76x2] Error 2
make[4]: *** [Makefile:1868: /home/openwrt/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-03-29-eed44048] Error 2
make[4]: Leaving directory '/home/openwrt/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.15.32'
make[3]: *** [Makefile:469: /home/openwrt/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-03-29-eed44048/.built] Error 2make[3]: Leaving directory '/home/openwrt/lede/package/kernel/mt76'
time: package/kernel/mt76/compile#10.49#2.54#13.04
    ERROR: package/kernel/mt76 failed to build.
make[2]: *** [package/Makefile:116: package/kernel/mt76/compile] Error 1
make[2]: Leaving directory '/home/openwrt/lede'
make[1]: *** [package/Makefile:110: /home/openwrt/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/openwrt/lede'
make: *** [/home/openwrt/lede/include/toplevel.mk:230：world] 错误 2
openwrt@dann:~/lede$ ad

GreenHT, 2022-04-10:
同问，也是  package/kernel/mt76 failed to build

yuanlun1983, 2022-04-11:
怎么选择5.15的内核编译呢？默认是5.4的内核

WizardQAQ, 2022-04-11:
> 同问，也是 package/kernel/mt76 failed to build

make menuconfig
Global  build settings -> Use the testing kernel version

yuanlun1983, 2022-04-11:
这是来自QQ邮箱的假期自动回复邮件。
 
您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。

WizardQAQ, 2022-04-11:
> 怎么选择5.15的内核编译呢？默认是5.4的内核

make menuconfig
Global build settings -> Use the testing kernel version

yuanlun1983, 2022-04-11:
> > 怎么选择5.15的内核编译呢？默认是5.4的内核
> 
> make menuconfig Global build settings -> Use the testing kernel version
谢谢，ax3600,ax6之类的路由，在ipq807这个处理器下没有可供编译的设备，要怎么选呢？

danuonuo, 2022-04-23:
实测此bug在此消息提交时仍未修复，日志如下：
make[3]: *** [Makefile:469: /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-04-20-eecbb499/.built] Error 2
[27237](https://github.com/danuonuo/actions-openwrt-mir3g/runs/6137171672?check_suite_focus=true#step:11:27237)
make[3]: Leaving directory '/workdir/openwrt/package/kernel/mt76'
[27238](https://github.com/danuonuo/actions-openwrt-mir3g/runs/6137171672?check_suite_focus=true#step:11:27238)
time: package/kernel/mt76/compile#3.93#0.72#4.48
[27239](https://github.com/danuonuo/actions-openwrt-mir3g/runs/6137171672?check_suite_focus=true#step:11:27239)
    ERROR: package/kernel/mt76 failed to build.
[27240](https://github.com/danuonuo/actions-openwrt-mir3g/runs/6137171672?check_suite_focus=true#step:11:27240)
make[2]: *** [package/Makefile:116: package/kernel/mt76/compile] Error 1
[27241](https://github.com/danuonuo/actions-openwrt-mir3g/runs/6137171672?check_suite_focus=true#step:11:27241)
make[2]: Leaving directory '/workdir/openwrt'
[27242](https://github.com/danuonuo/actions-openwrt-mir3g/runs/6137171672?check_suite_focus=true#step:11:27242)
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
[27243](https://github.com/danuonuo/actions-openwrt-mir3g/runs/6137171672?check_suite_focus=true#step:11:27243)
make[1]: Leaving directory '/workdir/openwrt'
[27244](https://github.com/danuonuo/actions-openwrt-mir3g/runs/6137171672?check_suite_focus=true#step:11:27244)
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2

closed 2022-07-25 by aiamadeus

Cathgao, 2022-10-11:
放过MIPS吧