coolsnowwolf/lede #9219: R4S kernel 5.15.13编译失败

GreatMichaelLee, 2022-04-10:
Applying /home/builder/lede_r4s/target/linux/generic/pending-5.15/850-0008-PCI-aardvark-Use-dev_fwnode-instead-of-of_node_to_fw.patch using plaintext:
patching file drivers/pci/controller/pci-aardvark.c

Applying /home/builder/lede_r4s/target/linux/generic/pending-5.15/850-0009-PCI-aardvark-Refactor-unmasking-summary-MSI-interrup.patch using plaintext:
patching file drivers/pci/controller/pci-aardvark.c

Applying /home/builder/lede_r4s/target/linux/generic/pending-5.15/850-0010-PCI-aardvark-Add-support-for-masking-MSI-interrupts.patch using plaintext:
patching file drivers/pci/controller/pci-aardvark.c

Applying /home/builder/lede_r4s/target/linux/generic/pending-5.15/850-0011-PCI-aardvark-Fix-setting-MSI-address.patch using plaintext:
patching file drivers/pci/controller/pci-aardvark.c

Applying /home/builder/lede_r4s/target/linux/generic/pending-5.15/850-0012-PCI-aardvark-Enable-MSI-X-support.patch using plaintext:
patching file drivers/pci/controller/pci-aardvark.c

Applying /home/builder/lede_r4s/target/linux/generic/pending-5.15/850-0013-PCI-aardvark-Add-support-for-ERR-interrupt-on-emulat.patch using plaintext:
patching file drivers/pci/controller/pci-aardvark.c
patch: **** malformed patch at line 100:                advk_pcie_handle_msi(pcie);

Patch failed!  Please fix /home/builder/lede_r4s/target/linux/generic/pending-5.15/850-0013-PCI-aardvark-Add-support-for-ERR-interrupt-on-emulat.patch!
make[4]: *** [Makefile:33: /home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.33/.prepared_17abe7e916aa2912ee4d0dffc9ccf30f] Error 1
make[4]: Leaving directory '/home/builder/lede_r4s/target/linux/rockchip'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/builder/lede_r4s/target/linux'
time: target/linux/compile#4.28#17.61#15.52
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:25: target/linux/compile] Error 1
make[2]: Leaving directory '/home/builder/lede_r4s'
make[1]: *** [target/Makefile:18: /home/builder/lede_r4s/staging_dir/target-aarch64_generic_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/builder/lede_r4s'
make: *** [/home/builder/lede_r4s/include/toplevel.mk:230：world] 错误 2
builder@Build-Server:~/lede_r4s$

GreatMichaelLee, 2022-04-10:
x86一样问题

dsyo2008, 2022-04-10:
同样问题

LPDDR6-10000MHz, 2022-04-10:
git pull....
[image]

LPDDR6-10000MHz, 2022-04-10:
删了源码重新拉

Boos4721, 2022-04-10:
Na, Works TY [My Firwame](https://drive.google.com/file/d/1vM9QyUtbv9rJ-Auq1Eb_QZvLSnGSCBqD/view?usp=sharing) passwd boos.

closed 2022-04-11 by coolsnowwolf