coolsnowwolf/lede #9220: 关于 ``` x86_64 ``` 重启丢配置，``` Hyper-V + UEFI + SquashFS ```

signigelchan, 2022-04-10:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [*] 有类似的issue，但实际解决说明很模糊

## 2. 详细叙述

### (1) 具体问题

A：自行编译的 ``` x86_64 ``` 固件，跑在 Hyper-V 虚拟机上面，正常启动后配置openwt，重启后配置全部丢失，恢复到了原始状态，初步怀疑是磁盘挂载点问题，或者可能是编译选项未正确配置。

### (2) 路由器型号和固件版本

A：Hyper-V Gen2    ---    x86_64
```
openwrt-x86-64-generic-squashfs-combined-efi.img.gz
openwrt-x86-64-generic-squashfs-combined-efi.vhdx
```

### (3) 详细日志

A：以下是 ``` Lean-lede 自行编译镜像 ``` 及 ``` OpenWrt 官方镜像 ``` 挂载点信息

- Lean lede

```
root@OpenWrt:~# lsblk 
NAME     MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
sda        8:0    0 544.3M  0 disk 
├─sda1     8:1    0    32M  0 part /boot
│                                  /boot
├─sda2     8:2    0   512M  0 part /rom
└─sda128 259:0    0   239K  0 part 
```

```
root@OpenWrt:~# df -hT
Filesystem           Type            Size      Used Available Use% Mounted on
/dev/root            squashfs       57.0M     57.0M         0 100% /rom
tmpfs                tmpfs         493.8M     11.9M    481.9M   2% /tmp
tmpfs                tmpfs         493.8M      1.2M    492.6M   0% /tmp/root
overlayfs:/tmp/root  overlay       493.8M      1.2M    492.6M   0% /
/dev/sda1            vfat           31.9M      5.1M     26.8M  16% /boot
/dev/sda1            vfat           31.9M      5.1M     26.8M  16% /boot
tmpfs                tmpfs         512.0K         0    512.0K   0% /dev
```

```
root@OpenWrt:~# mount
/dev/root              on /rom                 type     squashfs    (ro,relatime)
tmpfs                  on /tmp                 type     tmpfs       (rw,nosuid,nodev,noatime)
tmpfs                  on /tmp/root            type     tmpfs       (rw,noatime,mode=755)
overlayfs:/tmp/root    on /                    type     overlay     (rw,noatime,lowerdir=/,upperdir=/tmp/root/upper,workdir=/tmp/root/work)
tmpfs                  on /dev                 type     tmpfs       (rw,nosuid,relatime,size=512k,mode=755)
/dev/sda1              on /boot                type     vfat        (rw,noatime,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
/dev/sda1              on /boot                type     vfat        (rw,noatime,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
----------------------
devpts                 on /dev/pts             type     devpts      (rw,nosuid,noexec,relatime,mode=600,ptmxmode=000)
debugfs                on /sys/kernel/debug    type     debugfs     (rw,noatime)
sysfs                  on /sys                 type     sysfs       (rw,nosuid,nodev,noexec,noatime)
proc                   on /proc                type     proc        (rw,nosuid,nodev,noexec,noatime)
cgroup2                on /sys/fs/cgroup       type     cgroup2     (rw,nosuid,nodev,noexec,relatime,nsdelegate)
none                   on /sys/fs/bpf          type     bpf         (rw,nosuid,nodev,noexec,noatime,mode=700)
```

```
root@OpenWrt:/etc# cat /etc/os-release 
NAME="OpenWrt"
VERSION="SNAPSHOT"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt SNAPSHOT"
VERSION_ID="snapshot"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r4456-178212db2"
OPENWRT_BOARD="x86/64"
OPENWRT_ARCH="x86_64"
OPENWRT_TAINTS="no-all"
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt SNAPSHOT r4456-178212db2"
```

- OpenWrt 官方

```
root@OpenWrt:~# lsblk 
NAME     MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
loop0      7:0    0   100M  0 loop /overlay
sda        8:0    0 120.5M  0 disk 
├─sda1     8:1    0    16M  0 part /boot
├─sda2     8:2    0   104M  0 part /rom
└─sda128 259:0    0   239K  0 part
```

```
root@OpenWrt:~# df -hT
Filesystem           Type            Size      Used Available Use% Mounted on
/dev/root            squashfs        4.0M      4.0M         0 100% /rom
tmpfs                tmpfs         494.2M     52.0K    494.1M   0% /tmp
/dev/loop0           ext4           92.8M      1.5M     84.3M   2% /overlay
overlayfs:/overlay   overlay        92.8M      1.5M     84.3M   2% /
/dev/sda1            vfat           16.0M      5.3M     10.7M  33% /boot
/dev/sda1            vfat           16.0M      5.3M     10.7M  33% /boot
tmpfs                tmpfs         512.0K         0    512.0K   0% /dev
```

```
root@OpenWrt:~# mount
/dev/root               on /rom                 type    squashfs    ro,relatime)
proc                    on /proc                type    proc        rw,nosuid,nodev,noexec,noatime)
sysfs                   on /sys                 type    sysfs       rw,nosuid,nodev,noexec,noatime)
cgroup2                 on /sys/fs/cgroup       type    cgroup2     rw,nosuid,nodev,noexec,relatime,nsdelegate)
tmpfs                   on /tmp                 type    tmpfs       rw,nosuid,nodev,noatime)
/dev/loop0              on /overlay             type    ext4        rw,noatime)
overlayfs:/overlay      on /                    type    overlay     rw,noatime,lowerdir=/,upperdir=/overlay/upper,workdir=/overlay/work)
/dev/sda1               on /boot                type    vfat        rw,noatime,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
/dev/sda1               on /boot                type    vfat        rw,noatime,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
tmpfs                   on /dev                 type    tmpfs       rw,nosuid,relatime,size=512k,mode=755)
devpts                  on /dev/pts             type    devpts      rw,nosuid,noexec,relatime,mode=600,ptmxmode=000)
debugfs                 on /sys/kernel/debug    type    debugfs     rw,noatime)
none                    on /sys/fs/bpf          type    bpf         rw,nosuid,nodev,noexec,noatime,mode=700)
```

```
root@OpenWrt:~# cat /etc/os-release 
NAME="OpenWrt"
VERSION="21.02.2"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 21.02.2"
VERSION_ID="21.02.2"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r16495-bf0c965af0"
OPENWRT_BOARD="x86/64"
OPENWRT_ARCH="x86_64"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 21.02.2 r16495-bf0c965af0"
```

signigelchan, 2022-04-10:
### ``` dmesg ``` 日志

- Lean lede

```
root@OpenWrt:~# dmesg 
[    0.000000] Linux version 5.10.109 (instuser@CHANYOGA) (x86_64-openwrt-linux-musl-gcc (OpenWrt GCC 8.4.0 r4456-178212db2) 8.4.0, GNU ld (GNU Binutils) 2.34) #0 SMP Sat Apr 9 09:07:58 2022
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=5d2802f0-6939-a1e7-1503-8afaea3cb802 rootwait console=tty0 console=ttyS0,115200n8 noinitrd
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x020: 'AVX-512 opmask'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x040: 'AVX-512 Hi256'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x080: 'AVX-512 ZMM_Hi256'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: xstate_offset[5]:  832, xstate_sizes[5]:   64
[    0.000000] x86/fpu: xstate_offset[6]:  896, xstate_sizes[6]:  512
[    0.000000] x86/fpu: xstate_offset[7]: 1408, xstate_sizes[7]: 1024
[    0.000000] x86/fpu: Enabled xstate features 0xe7, context size is 2432 bytes, using 'compacted' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000000c0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000003ed93fff] usable
[    0.000000] BIOS-e820: [mem 0x000000003ed94000-0x000000003ed94fff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000003ed95000-0x000000003eecbfff] usable
[    0.000000] BIOS-e820: [mem 0x000000003eecc000-0x000000003eeeafff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000003eeeb000-0x000000003eef1fff] type 20
[    0.000000] BIOS-e820: [mem 0x000000003eef2000-0x000000003ef1afff] reserved
[    0.000000] BIOS-e820: [mem 0x000000003ef1b000-0x000000003ff9afff] usable
[    0.000000] BIOS-e820: [mem 0x000000003ff9b000-0x000000003ffcafff] type 20
[    0.000000] BIOS-e820: [mem 0x000000003ffcb000-0x000000003fff2fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000003fff3000-0x000000003fffafff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000003fffb000-0x000000003fffefff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000003ffff000-0x000000003fffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.70 by Microsoft
[    0.000000] efi: ACPI=0x3fffa000 ACPI 2.0=0x3fffa014 SMBIOS=0x3ffd8000 SMBIOS 3.0=0x3ffd6000 MEMATTR=0x3fa2ba98 
[    0.000000] SMBIOS 3.1.0 present.
[    0.000000] DMI: Microsoft Corporation Virtual Machine/Virtual Machine, BIOS Hyper-V UEFI Release v4.0 11/01/2019
[    0.000000] Hypervisor detected: Microsoft Hyper-V
[    0.000000] Hyper-V: features 0x2e7f, hints 0x20e24, misc 0xbed7b2
[    0.000000] Hyper-V Host Build:19041-10.0-1-0.1620
[    0.000000] Hyper-V: LAPIC Timer Frequency: 0x1e8480
[    0.000000] Hyper-V: Using hypercall for remote TLB flush
[    0.000000] clocksource: hyperv_clocksource_tsc_page: mask: 0xffffffffffffffff max_cycles: 0x24e6a1710, max_idle_ns: 440795202120 ns
[    0.000000] tsc: Marking TSC unstable due to running on Hyper-V
[    0.000002] tsc: Detected 2419.210 MHz processor
[    0.000006] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000008] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000011] last_pfn = 0x40000 max_arch_pfn = 0x400000000
[    0.000025] MTRR default type: uncachable
[    0.000025] MTRR fixed ranges enabled:
[    0.000026]   00000-9FFFF write-back
[    0.000026]   A0000-FFFFF uncachable
[    0.000027] MTRR variable ranges enabled:
[    0.000027]   0 base 0000000000 mask 7F00000000 write-back
[    0.000028]   1 disabled
[    0.000028]   2 disabled
[    0.000028]   3 disabled
[    0.000029]   4 disabled
[    0.000029]   5 disabled
[    0.000029]   6 disabled
[    0.000029]   7 disabled
[    0.000034] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.007839] Using GB pages for direct mapping
[    0.008007] Secure boot disabled
[    0.008009] ACPI: Early table checksum verification disabled
[    0.008011] ACPI: RSDP 0x000000003FFFA014 000024 (v02 VRTUAL)
[    0.008013] ACPI: XSDT 0x000000003FFF90E8 00005C (v01 VRTUAL MICROSFT 00000001 MSFT 00000001)
[    0.008017] ACPI: FACP 0x000000003FFF8000 000114 (v06 VRTUAL MICROSFT 00000001 MSFT 00000001)
[... 638 lines cut ...]
[    4.462309] ext4 filesystem being mounted at /tmp/overlay supports timestamps until 2038 (0x7fffffff)
[    4.463286] mount_root: switching to ext4 overlay
[    4.463924] overlayfs: "xino" feature enabled using 32 upper inode bits.
[    4.488996] urandom-seed: Seeding with /etc/urandom.seed
[    4.512500] procd: - early -
[    5.032325] procd: - ubus -
[    5.041159] random: ubusd: uninitialized urandom read (4 bytes read)
[    5.084039] procd: - init -
[    5.145558] urngd: jent-rng init failed, err: 2
[    5.155424] kmodloader: loading kernel modules from /etc/modules.d/*
[    5.157598] e1000: Intel(R) PRO/1000 Network Driver - version 7.3.21-k8-NAPI
[    5.158009] e1000: Copyright (c) 1999-2006 Intel Corporation.
[    5.158615] igb: Intel(R) Gigabit Ethernet Network Driver - version 5.6.0-k
[    5.159019] igb: Copyright (c) 2007-2014 Intel Corporation.
[    5.159728] ixgbe: Intel(R) 10 Gigabit PCI Express Network Driver - version 5.1.0-k
[    5.160219] ixgbe: Copyright (c) 1999-2016 Intel Corporation.
[    5.161196] i2c /dev entries driver
[    5.162050] e1000e: Intel(R) PRO/1000 Network Driver - 3.2.6-k
[    5.162439] e1000e: Copyright(c) 1999 - 2015 Intel Corporation.
[    5.163882] xt_time: kernel timezone is -0000
[    5.164900] PPP generic driver version 2.4.2
[    5.165370] NET: Registered protocol family 24
[    5.166340] kmodloader: done loading kernel modules from /etc/modules.d/*
[    8.015790] crng init done
```

ntlf9t, 2022-04-10:
/dev/root            squashfs       57.0M     57.0M         0 100% /rom
overlayfs:/tmp/root  overlay       493.8M      1.2M    492.6M   0% /

满了

signigelchan, 2022-04-10:
搜索相关报错问题及多次尝试，问题已解决，以下是最终定位的原因：

``` lede/target/linux/x86/Makefile ``` 编译文件内容里，依赖了 ``` mkf2fs ``` 包，自行配置编译选项过程中手动取消选中了此包，导致 ``` /dev/loop0 ``` 设备初始化异常，启动流程 overlay 挂载了 ``` tmpfs ``` 而非 ``` sda2 ``` 所致。

> DEFAULT_PACKAGES += partx-utils **mkf2fs** e2fsprogs kmod-button-hotplug ... ...

closed 2022-04-10 by signigelchan