coolsnowwolf/lede #9221: 更新最新内核后编译出错

mitsukileung, 2022-04-10:
X86

Patch failed!  Please fix /home/koyko/lede/target/linux/generic/pending-5.15/850-0013-PCI-aardvark-Add-support-for-ERR-interrupt-on-emulat.patch!
make[4]: *** [Makefile:35: /home/koyko/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.33/.prepared_ee9386265a89370f5a4f9c4104d91883] Error 1
make[4]: Leaving directory '/home/koyko/lede/target/linux/x86'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/koyko/lede/target/linux'
time: target/linux/compile#9.10#5.15#11.86
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:25: target/linux/compile] Error 1
make[2]: Leaving directory '/home/koyko/lede'
make[1]: *** [target/Makefile:18: /home/koyko/lede/staging_dir/target-x86_64_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/koyko/lede'
make: *** [/home/koyko/lede/include/toplevel.mk:230：world] 错误 2

另外执行  ./scripts/feeds update -a && ./scripts/feeds install -a
出现以下提示
WARNING: Makefile 'package/feeds/packages/baidupcs-web/Makefile' has a build dependency on 'go-rice/host', which does not exist

chenxuan5858, 2022-04-10:
同样问题

wang1zhen, 2022-04-10:
云编译可以复现

breakings, 2022-04-10:
这个和新内核无关吧，关键是这个`go-rice/host` 依赖，这个缺失。

LPDDR6-10000MHz, 2022-04-10:
git pull....
[image]

LPDDR6-10000MHz, 2022-04-10:
删了源码重新拉

closed 2022-04-10 by mitsukileung