coolsnowwolf/lede #9223: /mtk_wed.h: No such file or directory

WizardQAQ, 2022-04-10:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译失败，显示mtk_wed.h不存在，重新拉库好像也不行

### (2) 路由器型号和固件版本

A：极路由B70(HC5962）

### (3) 详细日志

A：
```bash
-- Build files have been written to: /home/gll/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-04-07-506bb060/tools
touch /home/gll/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-04-07-506bb060/.configured_9e276c5a547b8fdc8d20e786cae50ea1
rm -f /home/gll/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-04-07-506bb060/.built
touch /home/gll/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-04-07-506bb060/.built_check
make  -C "/home/gll/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.188" KCFLAGS="-fmacro-prefix-map=/home/gll/lede/build_dir/target-mipsel_24kc_musl=target-mipsel_24kc_musl" HOSTCFLAGS="-O2 -I/home/gll/lede/staging_dir/host/include -I/home/gll/lede/staging_dir/hostpkg/include -I/home/gll/lede/staging_dir/target-mipsel_24kc_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="mipsel-openwrt-linux-musl-" ARCH="mips" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Sun Apr 10 08:32:25 2022" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/gll/lede/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/home/gll/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KBUILD_EXTRA_SYMBOLS="/home/gll/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/symvers/cryptodev-linux.symvers /home/gll/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/symvers/gpio-button-hotplug.symvers /home/gll/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/symvers/mac80211.symvers" KERNELRELEASE=5.4.188 CONFIG_MT76x02_LIB=m CONFIG_MT76x2_COMMON=m CONFIG_MT76x2E=m CONFIG_MT7603E=m M="/home/gll/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-04-07-506bb060" NOSTDINC_FLAGS="-nostdinc  -isystem /home/gll/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/lib/gcc/mipsel-openwrt-linux-musl/8.4.0/include -I/home/gll/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-04-07-506bb060 -I/home/gll/lede/staging_dir/target-mipsel_24kc_musl/usr/include/mac80211-backport/uapi -I/home/gll/lede/staging_dir/target-mipsel_24kc_musl/usr/include/mac80211-backport -I/home/gll/lede/staging_dir/target-mipsel_24kc_musl/usr/include/mac80211/uapi -I/home/gll/lede/staging_dir/target-mipsel_24kc_musl/usr/include/mac80211 -include backport/autoconf.h -include backport/backport.h -DCONFIG_MAC80211_MESH" modules
make[4]: Entering directory '/home/gll/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.188'
  CC [M]  /home/gll/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-04-07-506bb060/mt7603/pci.o
In file included from /home/gll/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-04-07-506bb060/mt7603/mt7603.h:8,
                 from /home/gll/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-04-07-506bb060/mt7603/pci.c:7:
/home/gll/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-04-07-506bb060/mt7603/../mt76.h:16:10: fatal error: linux/soc/mediatek/mtk_wed.h: No such file or directory
 #include <linux/soc/mediatek/mtk_wed.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[6]: *** [scripts/Makefile.build:262: /home/gll/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-04-07-506bb060/mt7603/pci.o] Error 1
make[5]: *** [scripts/Makefile.build:497: /home/gll/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-04-07-506bb060/mt7603] Error 2
make[4]: *** [Makefile:1734: /home/gll/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-04-07-506bb060] Error 2
make[4]: Leaving directory '/home/gll/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.188'
make[3]: *** [Makefile:469: /home/gll/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-04-07-506bb060/.built] Error 2
make[3]: Leaving directory '/home/gll/lede/package/kernel/mt76'
time: package/kernel/mt76/compile#1.30#0.23#1.33

    ERROR: package/kernel/mt76 failed to build.
make[2]: *** [package/Makefile:116: package/kernel/mt76/compile] Error 1
make[2]: Leaving directory '/home/gll/lede'
make[1]: *** [package/Makefile:110: /home/gll/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/gll/lede'
make: *** [/home/gll/lede/include/toplevel.mk:230: world] Error 2
```

Boos4721, 2022-04-10:
kk, revert mt76 update or sync commits.

WizardQAQ, 2022-04-11:
> kk, revert mt76 update or sync commits.https://github.com/Boos4721/openwrt/blob/f8e3f9e3e6a64b94a676dc68bab87c6169ffbaf2/target/linux/generic/pending-5.10/701-02-net-ethernet-mtk_eth_soc-add-support-for-Wireless-Et.patch#L973

感谢大佬，我还是revert吧。
Thanks for replying, I'm gonna revert mt76 update.

closed 2022-04-11 by WizardQAQ

WYC-2020, 2022-04-11:
别轻易去revert 在没确认这个patch 没作用的情况下

Boos4721, 2022-04-11:
> 别轻易去revert 在没确认这个patch 没作用的情况下

lmfao,do u actually know meme mean should revert **MT76** commit

WYC-2020, 2022-04-11:
> > 别轻易去revert 在没确认这个patch 没作用的情况下
> 
> lmfao,r u actually know meme mean should revert **MT76** commit

https://github.com/WYC-2020/lede/commit/4edaacd364e4c15ba450f3ebbec007982ba4ad02
功能自测

Boos4721, 2022-04-11:
> > > 别轻易去revert 在没确认这个patch 没作用的情况下
> > 
> > 
> > lmfao,r u actually know meme mean should revert **MT76** commit
> 
> [WYC-2020/lede@4edaacd](https://github.com/WYC-2020/lede/commit/4edaacd364e4c15ba450f3ebbec007982ba4ad02) 功能自测

lmao, any other contributor info

Boos4721, 2022-04-11:
btw,it`s missing header file https://github.com/openwrt/mt76/commit/87a962e0608f5d599ea091994ae65569bc16c0b4#diff-99df055b0ea8b37173f0d8fea7e7964bd6d9685904c60d5d4d40d815226a7d99R16, so revert mt76 update or pick [this commit](https://github.com/openwrt/openwrt/commit/0f029b3d2b505b40aca9a24a002838ed1060f83d)

WYC-2020, 2022-04-11:
> btw,it`s missing header file [openwrt/mt76@87a962e#diff-99df055b0ea8b37173f0d8fea7e7964bd6d9685904c60d5d4d40d815226a7d99R16](https://github.com/openwrt/mt76/commit/87a962e0608f5d599ea091994ae65569bc16c0b4#diff-99df055b0ea8b37173f0d8fea7e7964bd6d9685904c60d5d4d40d815226a7d99R16), so revert mt76 update or pick [this commit](https://github.com/openwrt/openwrt/commit/0f029b3d2b505b40aca9a24a002838ed1060f83d)

没有缺少头文件，更新mt76 到2022-04-07  并更新我这个提交即可，我看了头文件的
[image]
[image]

Boos4721, 2022-04-11:
> 没有缺少头文件，更新mt76 到2022-04-07 并更新我这个提交即可，我看了头文件的 [image] [image]

i said pick commits mt76 for works, or give up update mt76