coolsnowwolf/lede #9227: MT7622 編譯失败

aaronchin, 2022-04-10:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：MT7622系列

### (2) 路由器型号和固件版本

A：Linksys 8450 

### (3) 详细日志

A：
/home/aaronchin/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/mt76-2022-04-07-506bb060/mt7615/../mt76.h:16:10: fatal error: linux/soc/mediatek/mtk_wed.h: No such file or directory
 #include <linux/soc/mediatek/mtk_wed.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[6]: *** [scripts/Makefile.build:280: /home/aaronchin/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/mt76-2022-04-07-506bb060/mt7615/main.o] Error 1

Linksys E8450编译会出现以上问题,尝试换过其他mt7622机种都是一样问题

WYC-2020, 2022-04-11:
https://github.com/WYC-2020/lede/commit/4edaacd364e4c15ba450f3ebbec007982ba4ad02
功能自测

aaronchin, 2022-04-11:
调整后 出现以下错误
arch/arm64/boot/dts/mediatek/mt7622.dtsi:947.26-950.4: ERROR (duplicate_node_names): /syscon@1af00000: Duplicate node name
ERROR: Input tree has errors, aborting (use -f to force output)
make[7]: *** [scripts/Makefile.lib:326: arch/arm64/boot/dts/mediatek/mt7622-rfb1.dtb] Error 2
make[6]: *** [scripts/Makefile.build:497: arch/arm64/boot/dts/mediatek] Error 2
make[5]: *** [Makefile:1359: dtbs] Error 2

WYC-2020, 2022-04-12:
> 调整后 出现以下错误 arch/arm64/boot/dts/mediatek/mt7622.dtsi:947.26-950.4: ERROR (duplicate_node_names): /syscon@1af00000: Duplicate node name ERROR: Input tree has errors, aborting (use -f to force output) [...]

这是因为有一个补丁重复打了