coolsnowwolf/lede #9231: 极路由4(HC5962) 5.15内核编译失败

KongGuoguang, 2022-04-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：使用5.15内核编译报错

### (2) 路由器型号和固件版本

A：极路由4(HC5962)，最新代码

### (3) 详细日志

A：
/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-04-07-506bb060/mt76x2/pci.c:181:1: warning: data definition has no type or storage class
 module_pci_driver(mt76pci_driver);
 ^~~~~~~~~~~~~~~~~
/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-04-07-506bb060/mt76x2/pci.c:181:1: error: type defaults to 'int' in declaration of 'module_pci_driver' [-Werror=implicit-int]
/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-04-07-506bb060/mt76x2/pci.c:181:1: warning: parameter names (without types) in function declaration
/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-04-07-506bb060/mt76x2/pci.c:170:26: warning: 'mt76pci_driver' defined but not used [-Wunused-variable]
 static struct pci_driver mt76pci_driver = {
                          ^~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[6]: *** [scripts/Makefile.build:277: /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-04-07-506bb060/mt76x2/pci.o] Error 1
make[5]: *** [scripts/Makefile.build:540: /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-04-07-506bb060/mt76x2] Error 2
make[4]: *** [Makefile:1868: /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-04-07-506bb060] Error 2
make[4]: Leaving directory '/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.15.32'
make[3]: *** [Makefile:469: /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-04-07-506bb060/.built] Error 2
make[3]: Leaving directory '/workdir/openwrt/package/kernel/mt76'
time: package/kernel/mt76/compile#3.85#0.65#4.35
    ERROR: package/kernel/mt76 failed to build.
make[2]: *** [package/Makefile:116: package/kernel/mt76/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2

modelsun, 2022-04-12:
同5962 ，编 5.10内核的，也是fail

 make[3] -C feeds/packages/lang/lua-cjson compile
 make[3] -C feeds/packages/lang/luasocket compile
 make[3] -C package/utils/busybox compile
 make[3] -C package/system/opkg host-compile
 make[3] -C package/kernel/linux compile
    ERROR: package/kernel/linux failed to build.

KongGuoguang, 2022-04-13:
2033-4-13  最新报错日志
/home/kong/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mtk-eip93-2020-12-10-bb5fd1c8/src/eip93-core.c:547:34: error: array type has incomplete element type 'struct of_device_id'
 static const struct of_device_id mtk_crypto_of_match[] = {
                                  ^~~~~~~~~~~~~~~~~~~
/home/kong/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mtk-eip93-2020-12-10-bb5fd1c8/src/eip93-core.c:548:4: error: field name not in record or union initializer
  { .compatible = "mediatek,mtk-eip93", },
    ^
/home/kong/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mtk-eip93-2020-12-10-bb5fd1c8/src/eip93-core.c:548:4: note: (near initialization for 'mtk_crypto_of_match')
/home/kong/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mtk-eip93-2020-12-10-bb5fd1c8/src/eip93-core.c:547:34: warning: 'mtk_crypto_of_match' defined but not used [-Wunused-variable]
 static const struct of_device_id mtk_crypto_of_match[] = {
                                  ^~~~~~~~~~~~~~~~~~~
make[5]: *** [scripts/Makefile.build:277: /home/kong/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mtk-eip93-2020-12-10-bb5fd1c8/src/eip93-core.o] Error 1
make[4]: *** [Makefile:1868: /home/kong/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mtk-eip93-2020-12-10-bb5fd1c8/src] Error 2
make[4]: Leaving directory '/home/kong/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.15.33'
make[3]: *** [Makefile:61: /home/kong/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mtk-eip93-2020-12-10-bb5fd1c8/.built] Error 2
make[3]: Leaving directory '/home/kong/lede/package/lean/mtk-eip93'
time: package/lean/mtk-eip93/compile#1.23#0.41#1.50
    ERROR: package/lean/mtk-eip93 failed to build.
make[2]: *** [package/Makefile:116: package/lean/mtk-eip93/compile] Error 1
make[2]: Leaving directory '/home/kong/lede'
make[1]: *** [package/Makefile:110: /home/kong/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/kong/lede'
make: *** [/home/kong/lede/include/toplevel.mk:230：world] 错误 2

xiangfeigao, 2022-04-14:
please fix package/lean/mt/luci-app-mtwifi/Makefile - see logs/package/lean/mt/luci-app-mtwifi/dump.txt for details

modelsun, 2022-04-14:
今天编过了

WizardQAQ, 2022-04-14:
> 今天编过了

看了下你的配置，替换了版本，编译5.10过了。但是好像如果打开test kernel，编译5.15还是不行欸

KongGuoguang, 2022-04-15:
> 今天编过了

我这儿还是不行

zhoujinshi, 2022-04-18:
openmesh有点鸡肋，必须配置免dhcp和静态不冲突的IP，LDDEwifi自带AP功能，寻找后填写WIFI密码即可，也可以静态配置，也可以dhcp配置，我之前编译openmesh没问题，今天出编译问题，导致编译失败。

smomop, 2022-04-21:
想在这个议题问下大佬，我编译这个机器的固件，wifi驱动速度不理想，有没有什么文献可以参考一下，提升一下无线速度

zhengmz, 2022-04-21:
> openmesh有点鸡肋，必须配置免dhcp和静态不冲突的IP，LDDEwifi自带AP功能，寻找后填写WIFI密码即可，也可以静态配置，也可以dhcp配置，我之前编译openmesh没问题，今天出编译问题，导致编译失败。

```shell
# 如果已经编译过
rm -fr build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2022.0
rm -fr feeds/routing/batman-adv
svn co https://github.com/openwrt/routing/branches/openwrt-22.03/batman-adv/ feeds/routing/batman-adv
```

解决思路参见：<https://githubhot.com/repo/immortalwrt/immortalwrt/issues/494>

closed 2022-07-25 by aiamadeus