coolsnowwolf/lede #9236: macOS 编译出现 Build dependency: OpenWrt can only be built on a case-sensitive filesystem

yicko, 2022-04-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X ] 没有类似的issue

## 2. 详细叙述
macos 执行 make -j1 V=s 出现错误

### (1) 具体问题

A：Build dependency: OpenWrt can only be built on a case-sensitive filesystem

Prerequisite check failed. Use FORCE=1 to override.
make[2]: *** [/Users/cunpengwang/lede/include/toplevel.mk:180: staging_dir/host/.prereq-build] Error 1
make[1]: *** [/Users/cunpengwang/lede/include/toplevel.mk:77: prepare-tmpinfo] Error 2
make: *** [/Users/cunpengwang/lede/include/toplevel.mk:228: world] Error 2

iKindle, 2022-04-12:
要在区分大小的宗卷运行。加一个区分大小写的宗卷即可。

feisan, 2022-04-14:
你可以试试修改 include/prereq-build.mk 文件，注释掉 case-sensitive-fs 相关的那几行

```
# $(eval $(call TestHostCommand,case-sensitive-fs, \
#	OpenWrt can only be built on a case-sensitive filesystem, \
#	rm -f $(TMP_DIR)/test.*; touch $(TMP_DIR)/test.fs; \
#		test ! -f $(TMP_DIR)/test.FS))
```

WXR2727, 2022-04-15:
cd ~
hdiutil create -size 20g -type SPARSE -fs "Case-sensitive HFS+" -volname OpenWrt OpenWrt.sparseimage
hdiutil attach OpenWrt.sparseimage

cd /Volumes/OpenWrt

开一个partition就可以解决了

closed 2022-10-31 by aiamadeus

miku1958, 2022-12-05:
APFS 不就是case-sensitive的吗, 这看起来更像是test ! -f $(TMP_DIR)/test.FS的问题