coolsnowwolf/lede #9243: MacOS 12 golang 问题

iKindle, 2022-04-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "Y" 选择)
* [ Y] 没有类似的issue

## 2. 详细叙述
go lang 在MacOS 的一个错误
### (1) 具体问题
[
](https://github.com/golang/go/wiki/MacOS12BSDThreadRegisterIssue)按照个，brew reinstall go 还是一个样的，下载go1.18.darwin-amd64.pkg，安装后还是一样报这个错误。
A：fatal error: runtime: bsdthread_register error

### (2) 路由器型号和固件版本
Newifi D2 MT7621
A：

### (3) 详细日志

A：
go/build
cmd/go
fatal error: runtime: bsdthread_register error

runtime stack:
runtime.throw(0x3ec895)
	/Volumes/Openwrt/lede/build_dir/hostpkg/go-1.18/.go_bootstrap/src/runtime/panic.go:491 +0xad fp=0x7ff7bfefecc0 sp=0x7ff7bfefec90
runtime.goenvs()
	/Volumes/Openwrt/lede/build_dir/hostpkg/go-1.18/.go_bootstrap/src/runtime/os_darwin.c:104 +0x70 fp=0x7ff7bfefecd8 sp=0x7ff7bfefecc0
runtime.schedinit()
	/Volumes/Openwrt/lede/build_dir/hostpkg/go-1.18/.go_bootstrap/src/runtime/proc.c:141 +0x5f fp=0x7ff7bfefed00 sp=0x7ff7bfefecd8
runtime.rt0_go(0x7ff7bfefed30, 0x4, 0x7ff7bfefed30, 0x1000, 0x4, 0x7ff7bfeff068, 0x7ff7bfeff0c9, 0x7ff7bfeff0cf, 0x7ff7bfeff0d2, 0x0, ...)
	/Volumes/Openwrt/lede/build_dir/hostpkg/go-1.18/.go_bootstrap/src/runtime/asm_amd64.s:95 +0x116 fp=0x7ff7bfefed08 sp=0x7ff7bfefed00
make[3]: *** [Makefile:370: /Volumes/Openwrt/lede/build_dir/hostpkg/go-1.18/.built] Error 2
make[3]: Leaving directory '/Volumes/Openwrt/lede/feeds/packages/lang/golang/golang'
time: package/feeds/packages/golang/host-compile#24.42#5.45#14.93
    ERROR: package/feeds/packages/golang [host] failed to build.
make[2]: *** [package/Makefile:114: package/feeds/packages/golang/host/compile] Error 1
make[2]: Leaving directory '/Volumes/Openwrt/lede'
make[1]: *** [package/Makefile:110: /Volumes/Openwrt/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/Volumes/Openwrt/lede'
make: *** [/Volumes/Openwrt/lede/include/toplevel.mk:230: world] Error 2

iKindle, 2022-04-12:
> https://openwrt.org/docs/guide-developer/toolchain/buildroot.exigence.macosx#golang_related
> 
> 12 之后 golang 需要指定配置

Then set
lede目录中的.config 
CONFIG_GOLANG_EXTERNAL_BOOTSTRAP_ROOT="/usr/local/Cellar/go/1.18/libexec"

没有报错了
Thank You~

qjiang888, 2022-05-23:
我添加路径了还是报错
mkdir -p /Users/qjiang/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp
SHELL= flock /Users/qjiang/lede/tmp/.root-copy.flock -c 'cp -fpR /Users/qjiang/lede/build_dir/target-mipsel_24kc_musl/luci-app-filetransfer/.pkgdir/luci-i18n-filetransfer-zh-cn/. /Users/qjiang/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/'
touch /Users/qjiang/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp/.luci-i18n-filetransfer-zh-cn_installed
echo "luci-i18n-filetransfer-zh-cn" >> /Users/qjiang/lede/staging_dir/target-mipsel_24kc_musl/pkginfo/luci-app-filetransfer.default.install
make[3]: Leaving directory '/Users/qjiang/lede/feeds/luci/applications/luci-app-filetransfer'
time: package/feeds/luci/luci-app-filetransfer/compile#0.73#0.42#1.15
make[3]: Entering directory '/Users/qjiang/lede/feeds/packages/lang/golang/golang'
mkdir -p /Users/qjiang/lede/dl
SHELL= flock /Users/qjiang/lede/tmp/.go1.18.2.src.tar.gz.flock -c '  	/Users/qjiang/lede/scripts/download.pl "/Users/qjiang/lede/dl" "go1.18.2.src.tar.gz" "2c44d03ea2c34092137ab919ba602f2c261a038d08eb468528a3f3a28e5667e2" "" "https://dl.google.com/go/" "https://mirrors.ustc.edu.cn/golang/" "https://mirrors.nju.edu.cn/golang/"    '
mkdir -p "/Users/qjiang/lede/tmp/go-build"
( cd "/Users/qjiang/lede/build_dir/hostpkg/go-1.18.2/src" ; GOROOT_FINAL="/Users/qjiang/lede/staging_dir/hostpkg/lib/go-cross"  GOROOT_BOOTSTRAP="/usr/local/Cellar/go/1.18.2/libexec" GO_LDFLAGS="-buildmode pie" GOCACHE="/Users/qjiang/lede/tmp/go-build" GOENV=off CC="gcc" CXX="g++"  bash make.bash --no-banner ; )
readelf: Error: Not an ELF file - it has the wrong magic bytes at the start
Building Go cmd/dist using /usr/local/Cellar/go/1.18.2/libexec. (go1.18.2 darwin/amd64)
# runtime/internal/sys
/usr/local/Cellar/go/1.18.2/libexec/src/runtime/internal/sys/consts.go:13:30: undefined: StackGuardMultiplierDefault
make[3]: *** [Makefile:371: /Users/qjiang/lede/build_dir/hostpkg/go-1.18.2/.built] Error 2
make[3]: Leaving directory '/Users/qjiang/lede/feeds/packages/lang/golang/golang'
time: package/feeds/packages/golang/host-compile#0.40#0.40#0.67
    ERROR: package/feeds/packages/golang [host] failed to build.
make[2]: *** [package/Makefile:114: package/feeds/packages/golang/host/compile] Error 1
make[2]: Leaving directory '/Users/qjiang/lede'
make[1]: *** [package/Makefile:110: /Users/qjiang/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/Users/qjiang/lede'
make: *** [/Users/qjiang/lede/include/toplevel.mk:230: world] Error 2