coolsnowwolf/lede #9244: MacOS 12  Ruby configure: error

iKindle, 2022-04-12:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [y ] 没有类似的issue

## 2. 详细叙述
MacOS 12，编译真是麻烦啊。configure: error: something wrong with LDFLAGS。

尝试brew reinstall openssl@1.1
export LDFLAGS="-L/usr/local/opt/openssl@1.1/lib"

头大了，无办法解决。
### (1) 具体问题
A：configure: error: something wrong with LDFLAGS="-L/Volumes/Openwrt/lede/staging_dir/host/lib -L/Volumes/Openwrt/lede/staging_dir/hostpkg/lib -L/Volumes/Openwrt/lede/staging_dir/target-mipsel_24kc_musl/host/lib"

### (2) 路由器型号和固件版本
Newifi D2   MT7621
A：

### (3) 详细日志

A：configure: loading site script /Volumes/Openwrt/lede/include/site/darwin
checking for ruby... /usr/bin/ruby
tool/config.guess already exists
tool/config.sub already exists
checking build system type... x86_64-apple-darwin21.4.0
checking host system type... x86_64-apple-darwin21.4.0
checking target system type... x86_64-apple-darwin21.4.0
checking for x86_64-apple-darwin21.4.0-cl.exe... gcc
checking for x86_64-apple-darwin21.4.0-ld... no
checking for ld... ld
checking for x86_64-apple-darwin21.4.0-gcc-ar... no
checking for gcc-ar... no
checking for x86_64-apple-darwin21.4.0-g++... g++
checking for x86_64-apple-darwin21.4.0-gcc-nm... no
checking for gcc-nm... no
checking for x86_64-apple-darwin21.4.0-gcc-ranlib... no
checking for gcc-ranlib... no
checking for x86_64-apple-darwin21.4.0-gcc... (cached) gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for gcc option to accept ISO C99... none needed
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking how to run the C preprocessor... gcc -E
checking for x86_64-apple-darwin21.4.0-ranlib... no
checking for ranlib... ranlib
checking for x86_64-apple-darwin21.4.0-gar... no
checking for x86_64-apple-darwin21.4.0-ar... no
checking for gar... no
checking for ar... ar
checking for x86_64-apple-darwin21.4.0-gas... no
checking for x86_64-apple-darwin21.4.0-as... no
checking for gas... no
checking for as... as
checking for x86_64-apple-darwin21.4.0-gld... ld
checking for x86_64-apple-darwin21.4.0-gnm... no
checking for x86_64-apple-darwin21.4.0-nm... no
checking for gnm... no
checking for nm... nm
checking for x86_64-apple-darwin21.4.0-gobjcopy... no
checking for x86_64-apple-darwin21.4.0-objcopy... no
checking for gobjcopy... no
checking for objcopy... no
checking for x86_64-apple-darwin21.4.0-gobjdump... no
checking for x86_64-apple-darwin21.4.0-objdump... no
checking for gobjdump... no
checking for objdump... objdump
checking for x86_64-apple-darwin21.4.0-gstrip... no
checking for x86_64-apple-darwin21.4.0-strip... no
checking for gstrip... no
checking for strip... strip
checking for grep that handles long lines and -e... /Volumes/Openwrt/lede/staging_dir/host/bin/grep
checking for egrep... /Volumes/Openwrt/lede/staging_dir/host/bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking minix/config.h usability... no
checking minix/config.h presence... no
checking for minix/config.h... no
checking whether it is safe to define __EXTENSIONS__... yes
checking if minimum required OS X version is supported... yes
checking whether the linker is GNU ld... no
checking whether gcc -E accepts -o... yes
checking for gcc... /Volumes/Openwrt/lede/staging_dir/host/bin/gcc
checking for ar flags... rcu
checking whether ln -s works... yes
checking whether make sets $(MAKE)... yes
checking for a BSD-compatible install... /Volumes/Openwrt/lede/staging_dir/host/bin/install -c
checking for a thread-safe mkdir -p... /usr/local/opt/coreutils/libexec/gnubin/mkdir -p
checking for x86_64-apple-darwin21.4.0-dtrace... no
checking for dtrace... dtrace
checking for dot... no
checking for doxygen... no
checking for x86_64-apple-darwin21.4.0-pkg-config... /Volumes/Openwrt/lede/staging_dir/host/bin/pkg-config
checking whether it is Android... no
checking for cd using physical directory... cd -P
checking whether CFLAGS is valid... yes
checking whether LDFLAGS is valid... no
configure: error: something wrong with LDFLAGS="-L/Volumes/Openwrt/lede/staging_dir/host/lib -L/Volumes/Openwrt/lede/staging_dir/hostpkg/lib -L/Volumes/Openwrt/lede/staging_dir/target-mipsel_24kc_musl/host/lib"
make[3]: *** [Makefile:1140: /Volumes/Openwrt/lede/build_dir/hostpkg/ruby-3.0.3/.configured] Error 1
make[3]: Leaving directory '/Volumes/Openwrt/lede/feeds/packages/lang/ruby'
time: package/feeds/packages/ruby/host-compile#4.57#1.94#10.30
    ERROR: package/feeds/packages/ruby [host] failed to build.
make[2]: *** [package/Makefile:114: package/feeds/packages/ruby/host/compile] Error 1
make[2]: Leaving directory '/Volumes/Openwrt/lede'
make[1]: *** [package/Makefile:110: /Volumes/Openwrt/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/Volumes/Openwrt/lede'
make: *** [/Volumes/Openwrt/lede/include/toplevel.mk:230: world] Error 2

WYC-2020, 2022-04-12:
只是想编译建议你用ubuntu,macos问题很多，不是奔解决问题去的，建议别用这玩意

iKindle, 2022-04-12:
> 只是想编译建议你用ubuntu,macos问题很多，不是奔解决问题去的，建议别用这玩意

哈哈，大雕都写出MacOS的编译步骤来了，用这个系统，也想试试啊～这样以后不用转到Windows 的WSL ubuntu下编译了。让它完美～😆

closed 2023-11-14 by aiamadeus