coolsnowwolf/lede #9245: 5.4.188固件编译问题，请求帮助

zbluekuma, 2022-04-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X] 没有类似的issue

## 2. 详细叙述
固件编译，选择kernel5.4版本，编译错误，5.10和5.15版本编译正常
### (1) 具体问题

A：删除了一次lede文件夹，重新git clone也还是不行。

### (2) 路由器型号和固件版本

A：ARRIS SBR-AC1750路由器（QCA9880+QCA9558+QCA8337）

### (3) 详细日志

A：make[1]: Entering directory '/home/cnz/lede'
make[2]: Entering directory '/home/cnz/lede'
make[3]: Entering directory '/home/cnz/lede/target/linux'
make[4]: Entering directory '/home/cnz/lede/target/linux/ath79'
rm -f /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/localversion
/home/cnz/lede/scripts/kconfig.pl  + + /home/cnz/lede/target/linux/generic/config-5.4 /home/cnz/lede/target/linux/ath79/config-5.4 /home/cnz/lede/target/linux/ath79/nand/config-default > /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/.config.target
awk '/^(#[[:space:]]+)?CONFIG_KERNEL/{sub("CONFIG_KERNEL_","CONFIG_");print}' /home/cnz/lede/.config >> /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/.config.target
echo "# CONFIG_KALLSYMS_EXTRA_PASS is not set" >> /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/.config.target
echo "# CONFIG_KALLSYMS_ALL is not set" >> /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/.config.target
echo "CONFIG_KALLSYMS_UNCOMPRESSED=y" >> /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/.config.target
/home/cnz/lede/scripts/package-metadata.pl kconfig /home/cnz/lede/tmp/.packageinfo /home/cnz/lede/.config 5.4 > /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/.config.override
/home/cnz/lede/scripts/kconfig.pl 'm+' '+' /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/.config.target /dev/null /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/.config.override > /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/.config.set
mv /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/.config.set /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/.config.old
grep -v INITRAMFS /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/.config.old > /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/.config.set
echo 'CONFIG_INITRAMFS_SOURCE=""' >> /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/.config.set
echo '# CONFIG_INITRAMFS_FORCE is not set' >> /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/.config.set
rm -rf /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/modules
cmp -s /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/.config.set /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/.config.prev || { cp /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/.config.set /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/.config; cp /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/.config.set /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/.config.prev; }
export MAKEFLAGS= ; [ -d /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/user_headers ] || make -C /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188 KCFLAGS="-fmacro-prefix-map=/home/cnz/lede/build_dir/target-mips_24kc_musl=target-mips_24kc_musl" HOSTCFLAGS="-O2 -I/home/cnz/lede/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="mips-openwrt-linux-musl-" ARCH="mips" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Tue Apr 12 04:59:09 2022" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/cnz/lede/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/home/cnz/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.4.188 CC="mips-openwrt-linux-musl-gcc" INSTALL_HDR_PATH=/home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/user_headers headers_install
grep '=[ym]' /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/.config.set | LC_ALL=C sort | /home/cnz/lede/staging_dir/host/bin/mkhash md5 > /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/.vermagic
touch /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/.configured
rm -f /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/vmlinux /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188/System.map
make -C /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188 KCFLAGS="-fmacro-prefix-map=/home/cnz/lede/build_dir/target-mips_24kc_musl=target-mips_24kc_musl" HOSTCFLAGS="-O2 -I/home/cnz/lede/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="mips-openwrt-linux-musl-" ARCH="mips" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Tue Apr 12 04:59:09 2022" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/cnz/lede/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/home/cnz/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.4.188 CC="mips-openwrt-linux-musl-gcc" olddefconfig
make[5]: Entering directory '/home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188'
scripts/kconfig/conf  --olddefconfig Kconfig
net/sched/Kconfig:45: warning: menuconfig statement without prompt
net/Kconfig:478:warning: multi-line strings not supported
drivers/mfd/Kconfig:2016:warning: multi-line strings not supported
#
# No change to .config
#
make[5]: Leaving directory '/home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188'
make -C /home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188 KCFLAGS="-fmacro-prefix-map=/home/cnz/lede/build_dir/target-mips_24kc_musl=target-mips_24kc_musl" HOSTCFLAGS="-O2 -I/home/cnz/lede/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="mips-openwrt-linux-musl-" ARCH="mips" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Tue Apr 12 04:59:09 2022" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/cnz/lede/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/home/cnz/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.4.188 CC="mips-openwrt-linux-musl-gcc" all modules
make[5]: Entering directory '/home/cnz/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.4.188'
  CALL    scripts/checksyscalls.sh
  CALL    scripts/atomic/check-atomics.sh
  CHK     include/generated/compile.h
  CC      drivers/mtd/nand/raw/ar934x_nand.o
drivers/mtd/nand/raw/ar934x_nand.c: In function 'ar934x_nfc_cmdfunc':
drivers/mtd/nand/raw/ar934x_nand.c:723:16: error: 'struct nand_ecc_ctrl' has no member named 'engine_type'
   if (nand->ecc.engine_type == NAND_ECC_ENGINE_TYPE_ON_HOST) {
                ^
drivers/mtd/nand/raw/ar934x_nand.c:723:32: error: 'NAND_ECC_ENGINE_TYPE_ON_HOST' undeclared (first use in this function); did you mean 'NAND_CI_CELLTYPE_SHIFT'?
   if (nand->ecc.engine_type == NAND_ECC_ENGINE_TYPE_ON_HOST) {
                                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                NAND_CI_CELLTYPE_SHIFT
drivers/mtd/nand/raw/ar934x_nand.c:723:32: note: each undeclared identifier is reported only once for each function it appears in
drivers/mtd/nand/raw/ar934x_nand.c: In function 'ar934x_nfc_attach_chip':
drivers/mtd/nand/raw/ar934x_nand.c:1331:15: error: 'struct nand_ecc_ctrl' has no member named 'engine_type'
  if (nand->ecc.engine_type == NAND_ECC_ENGINE_TYPE_ON_HOST) {
               ^
[... 27 lines cut ...]
drivers/mtd/nand/raw/ar934x_nand.c: In function 'ar934x_nfc_probe':
drivers/mtd/nand/raw/ar934x_nand.c:1428:11: error: 'struct nand_ecc_ctrl' has no member named 'engine_type'
  nand->ecc.engine_type = NAND_ECC_ENGINE_TYPE_ON_HOST; /* default */
           ^
drivers/mtd/nand/raw/ar934x_nand.c:1428:26: error: 'NAND_ECC_ENGINE_TYPE_ON_HOST' undeclared (first use in this function); did you mean 'NAND_CI_CELLTYPE_SHIFT'?
  nand->ecc.engine_type = NAND_ECC_ENGINE_TYPE_ON_HOST; /* default */
                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
[... cut ...]
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/cnz/lede/target/linux'
time: target/linux/compile#6.77#4.53#10.83
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:25: target/linux/compile] Error 1
make[2]: Leaving directory '/home/cnz/lede'
make[1]: *** [target/Makefile:18: /home/cnz/lede/staging_dir/target-mips_24kc_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/cnz/lede'
make: *** [/home/cnz/lede/include/toplevel.mk:230: world] Error 2

WYC-2020, 2022-04-13:
到时贴一张 选择的图片撒，或者config 撒 不是每个人都完全知道 每个平台在那个位置，知道怎么选

WYC-2020, 2022-04-13:
https://github.com/WYC-2020/lede/commit/5da8b41f04ee78834767e2624b9be08ca0892690
@zbluekuma

zbluekuma, 2022-04-13:
> 到时贴一张 选择的图片撒，或者config 撒 不是每个人都完全知道 每个平台在那个位置，知道怎么选

 @谢谢好心人，大法师！

closed 2022-04-14 by zbluekuma