coolsnowwolf/lede #9247: 树莓派3B+ bcm27xx  5.10内核编译报错【dma-buf/heaps in archive is not an object】 含修正方法请确认

jry485, 2022-04-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue
y

## 2. 详细叙述

树莓派3B+ bcm27xx, 5.10内核编译报错 【dma-buf/heaps in archive is not an object】

### (1) 具体问题

编译报错

### (2) 路由器型号和固件版本

树莓派3B+ bcm27xx

### (3) 详细日志

aarch64-openwrt-linux-musl-ld: drivers/built-in.a: member drivers/dma-buf/heaps in archive is not an object
make[5]: *** [Makefile:1179: vmlinux] Error 1
make[5]: Leaving directory '/home/jry/lede-20211107/build_dir/target-aarch64_cortex-a53_musl/linux-bcm27xx_bcm2710/linux-5.10.78'
make[4]: *** [Makefile:35: /home/jry/lede-20211107/build_dir/target-aarch64_cortex-a53_musl/linux-bcm27xx_bcm2710/linux-5.10.78/.modules] Error 2
make[4]: Leaving directory '/home/jry/lede-20211107/target/linux/bcm27xx'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/jry/lede-20211107/target/linux'
time: target/linux/compile#674.91#61.47#741.03

【原因和修正方式】请确认修改

以下文件
[target/linux/generic/hack-5.10/904-debloat_dma_buf.patch]  

+dma-buf-objs-$(CONFIG_DMABUF_HEAPS) += heaps/  （出错）

+obj-$(CONFIG_DMABUF_HEAPS)		+= heaps/   （改成这个）

WYC-2020, 2022-04-12:
关于这玩意我很早就修复了，只是没人在意这个问题
https://github.com/WYC-2020/lede/commit/be6061a0974d5550af52af8838a9b26ce5324dc9

jry485, 2022-04-12:
> 关于这玩意我很早就修复了，只是没人在意这个问题 [WYC-2020/lede@be6061a](https://github.com/WYC-2020/lede/commit/be6061a0974d5550af52af8838a9b26ce5324dc9)

对的！ 正解！

closed 2022-04-14 by jry485