coolsnowwolf/lede #9250: 能否update一下zerotier

GreatMichaelLee, 2022-04-13:
原来更新到1.8.7了，不知什么原因又revert了，现在出到1.8.8了，哪位大大顺手更新一下，我自己更make file,好像提示patch 脚本不兼容。

WYC-2020, 2022-04-13:
懒得更新，稳定能用懒得折腾哈哈

wuzhuang123, 2022-04-13:
去openwrt的package仓库，找到net/zerotier文件夹，替换lean仓库的feeds/packages/net/zerotier文件夹。op现在是1.8.7，但最新的1.8.8也是可以的，修改该文件夹的makefile中版本号和哈希为：
[image]

GreatMichaelLee, 2022-04-13:
> 去openwrt的package仓库，找到net/zerotier文件夹，替换lean仓库的feeds/packages/net/zerotier文件夹。op现在是1.8.7，但最新的1.8.8也是可以的，修改该文件夹的makefile中版本号和哈希为： [image]

不行，X86有个相应的patch会报错。改这个我肯定会，但我的水平也就会改这个了，patch不知道哪里出错，哈哈

wuzhuang123, 2022-04-13:
我MT7621可以使用，不确定X86如何。全部覆盖feeds/packages/net/zerotier文件夹，大不了还原呗。提供我的http://wuzhuang.picp.net:8080/zerotier1.8.8.zip

WYC-2020, 2022-04-14:
https://github.com/coolsnowwolf/lede/issues/9113 
这可能就是回退的原因

Hiyajomaho-num9, 2022-04-15:
为什么要升级？能用就行了啊？
你们无脑追新的也差不多得了

oldskooler, 2022-04-15:
路由器上系统日志一直出现如下日志，但是还是能用的
Fri Apr 15 09:27:39 2022 daemon.err zerotier-one[7086]: connect: Operation timed out
Fri Apr 15 09:27:45 2022 daemon.err zerotier-one[7086]: connect: Operation timed out
Fri Apr 15 09:27:51 2022 daemon.err zerotier-one[7086]: connect: Operation timed out
Fri Apr 15 09:27:57 2022 daemon.err zerotier-one[7086]: connect: Operation timed out
Fri Apr 15 09:28:03 2022 daemon.err zerotier-one[7086]: connect: Operation timed out
Fri Apr 15 09:28:10 2022 daemon.err zerotier-one[7086]: connect: Operation timed out
Fri Apr 15 09:28:16 2022 daemon.err zerotier-one[7086]: connect: Operation timed out
Fri Apr 15 09:28:22 2022 daemon.err zerotier-one[7086]: connect: Operation timed out
Fri Apr 15 09:28:28 2022 daemon.err zerotier-one[7086]: connect: Operation timed out
Fri Apr 15 11:32:31 2022 daemon.err zerotier-one[7086]: connect: Operation timed out
Fri Apr 15 11:32:37 2022 daemon.err zerotier-one[7086]: connect: Operation timed out
Fri Apr 15 11:32:43 2022 daemon.err zerotier-one[7086]: connect: Operation timed out
Fri Apr 15 11:32:49 2022 daemon.err zerotier-one[7086]: connect: Operation timed out
Fri Apr 15 11:32:55 2022 daemon.err zerotier-one[7086]: connect: Operation timed out
Fri Apr 15 11:33:01 2022 daemon.err zerotier-one[7086]: connect: Operation timed out
Fri Apr 15 11:33:08 2022 daemon.err zerotier-one[7086]: connect: Operation timed out
Fri Apr 15 11:33:14 2022 daemon.err zerotier-one[7086]: connect: Operation timed out
Fri Apr 15 11:33:20 2022 daemon.err zerotier-one[7086]: connect: Operation timed out
Fri Apr 15 11:33:26 2022 daemon.err zerotier-one[7086]: connect: Operation timed out
Fri Apr 15 11:38:41 2022 daemon.err zerotier-one[7086]: connect: Operation timed out
Fri Apr 15 11:38:47 2022 daemon.err zerotier-one[7086]: connect: Operation timed out
Fri Apr 15 11:38:53 2022 daemon.err zerotier-one[7086]: connect: Operation timed out
Fri Apr 15 11:39:00 2022 daemon.err zerotier-one[7086]: connect: Operation timed out
Fri Apr 15 11:39:06 2022 daemon.err zerotier-one[7086]: connect: Operation timed out
Fri Apr 15 11:39:12 2022 daemon.err zerotier-one[7086]: connect: Operation timed out
Fri Apr 15 11:39:18 2022 daemon.err zerotier-one[7086]: connect: Operation timed out
Fri Apr 15 11:39:24 2022 daemon.err zerotier-one[7086]: connect: Operation timed out
Fri Apr 15 11:39:30 2022 daemon.err zerotier-one[7086]: connect: Operation timed out
Fri Apr 15 11:39:36 2022 daemon.err zerotier-one[7086]: connect: Operation timed out

Beginner-Go, 2022-04-15:
有空我去操作下

breakings, 2022-04-15:
官方有PR 但需要arm32的人提交反馈以确认是否修复了段错误 https://github.com/openwrt/packages/pull/18300

GreatMichaelLee, 2022-04-15:
> 为什么要升级？能用就行了啊？ 你们无脑追新的也差不多得了

你这逻辑很好笑

Hiyajomaho-num9, 2022-04-16:
> > 为什么要升级？能用就行了啊？ 你们无脑追新的也差不多得了
> 
> 你这逻辑很好笑，追新就是无脑？出口就要伤人？那你永远不要升级就是了。

你们追新党只管能不能过编译，至于armv7和其他平台的死活并不在乎

GreatMichaelLee, 2022-04-16:
> > > 为什么要升级？能用就行了啊？ 你们无脑追新的也差不多得了
> > 
> > 
> > 你这逻辑很好笑，追新就是无脑？出口就要伤人？那你永远不要升级就是了。
> 
> 你们追新党只管能不能过编译，至于armv7的死活并不在乎 这个月插件revert有多少次，你们心里没点数？差不多得了

你不会自己fork一个仓库自己 revert到你work的版本不就完了，就为了你一个有问题的型号就反对别人更新？差不多得了！

WYC-2020, 2022-04-16:
好了好了 一个个少说一点，开源界已经够乌烟瘴气了，那个pr有人验证了没嘛 能正常工作不嘛，如果没什么问题 会升级的

GreatMichaelLee, 2022-04-16:
> 好了好了 一个个少说一点，开源界已经够乌烟瘴气了，那个pr有人验证了没嘛 能正常工作不嘛，如果没什么问题 会升级的

还是你好说话:)

AmadeusGhost, 2022-04-16:
因为部分平台上有问题，所以还不能升级，就这样。

closed 2022-04-18 by AmadeusGhost