coolsnowwolf/lede #9254: 编译出错，中间有一个go语言包无法连接

xb442572328, 2022-04-14:
这边选择配置树莓派系统后，编译就一直报这个错误，也没有找到解决办法
Finding targets
go: github.com/mitchellh/mapstructure@v1.4.2: Get "https://proxy.golang.org/github.com/mitchellh/mapstructure/@v/v1.4.2.mod": dial tcp 142.251.42.241:443: connect: connection refused
go: downloading golang.org/x/text v0.3.3
go: downloading github.com/mitchellh/mapstructure v1.4.2
go: github.com/mitchellh/mapstructure@v1.4.2: Get "https://proxy.golang.org/github.com/mitchellh/mapstructure/@v/v1.4.2.mod": dial tcp 142.251.42.241:443: connect: connection refused

Building targets
go: github.com/mitchellh/mapstructure@v1.4.2: Get "https://proxy.golang.org/github.com/mitchellh/mapstructure/@v/v1.4.2.mod": dial tcp 142.251.42.241:443: connect: connection refused
go: github.com/mitchellh/mapstructure@v1.4.2: Get "https://proxy.golang.org/github.com/mitchellh/mapstructure/@v/v1.4.2.mod": dial tcp 142.251.42.241:443: connect: connection refused

make[3]: *** [Makefile:86: /home/safe-en/lede/build_dir/target-aarch64_cortex-a53_musl/UnblockNeteaseMusic-0.2.13/.built] Error 1
make[3]: Leaving directory '/home/safe-en/lede/feeds/packages/multimedia/UnblockNeteaseMusic-Go'
time: package/feeds/packages/UnblockNeteaseMusic-Go/compile#0.19#0.12#84.42
    ERROR: package/feeds/packages/UnblockNeteaseMusic-Go failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/UnblockNeteaseMusic-Go/compile] Error 1
make[2]: Leaving directory '/home/safe-en/lede'
make[1]: *** [package/Makefile:110: /home/safe-en/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/safe-en/lede'
make: *** [/home/safe-en/lede/include/toplevel.mk:230：world] 错误 2

xb442572328, 2022-04-14:
啊？.....这就难顶了
这也不知道怎么修改了....今天就一直卡在这里报错了
我就是想用它加个EAP认证

> 感觉是你够子很长哈哈

wuzhuang123, 2022-04-15:
我是 dial tcp timeout.你试试这个的解决方案。

closed 2022-04-18 by AmadeusGhost