coolsnowwolf/lede #9256: python-3.9.12编译错误

ricksuzade-maker, 2022-04-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：在编译python-3.9.12时出错

### (2) 路由器型号和固件版本

A：x86-64

### (3) 详细日志

A：

SHELL= flock /home/zhiliang/scripts/lede/tmp/.libffi-3.3.tar.gz.flock -c '  	/home/zhiliang/scripts/lede/scripts/download.pl "/home/zhiliang/scripts/lede/dl" "libffi-3.3.tar.gz" "72fba7922703ddfa7a028d513ac15a85c8d54c8d67f55fa5a4802885dc652056" "" "https://sourceware.org/pub/libffi/"    '
make[3]: Leaving directory '/home/zhiliang/scripts/lede/feeds/packages/libs/libffi'
time: package/feeds/packages/libffi/host-compile#0.26#0.03#0.25
make[3]: Entering directory '/home/zhiliang/scripts/lede/package/utils/bzip2'
mkdir -p /home/zhiliang/scripts/lede/dl
SHELL= flock /home/zhiliang/scripts/lede/tmp/.bzip2-1.0.8.tar.gz.flock -c '  	/home/zhiliang/scripts/lede/scripts/download.pl "/home/zhiliang/scripts/lede/dl" "bzip2-1.0.8.tar.gz" "ab5a03176ee106d3f0fa90e381da478ddae405918153cca248e682cd0c4a2269" "" "https://sourceware.org/pub/bzip2"    '
make[3]: Leaving directory '/home/zhiliang/scripts/lede/package/utils/bzip2'
time: package/utils/bzip2/host-compile#0.26#0.04#0.26
make[3]: Entering directory '/home/zhiliang/scripts/lede/feeds/packages/lang/python/python3'
mkdir -p /home/zhiliang/scripts/lede/dl
SHELL= flock /home/zhiliang/scripts/lede/tmp/.Python-3.9.12.tar.xz.flock -c '  	/home/zhiliang/scripts/lede/scripts/download.pl "/home/zhiliang/scripts/lede/dl" "Python-3.9.12.tar.xz" "2cd94b20670e4159c6d9ab57f91dbf255b97d8c1a1451d1c35f4ec1968adf971" "" "https://www.python.org/ftp/python/3.9.12"    '
CFLAGS="-O2 -I/home/zhiliang/scripts/lede/staging_dir/host/include -I/home/zhiliang/scripts/lede/staging_dir/hostpkg/include -I/home/zhiliang/scripts/lede/staging_dir/target-x86_64_musl/host/include" CPPFLAGS="-I/home/zhiliang/scripts/lede/staging_dir/host/include -I/home/zhiliang/scripts/lede/staging_dir/hostpkg/include -I/home/zhiliang/scripts/lede/staging_dir/target-x86_64_musl/host/include" CXXFLAGS="" LDFLAGS="-L/home/zhiliang/scripts/lede/staging_dir/host/lib -L/home/zhiliang/scripts/lede/staging_dir/hostpkg/lib -L/home/zhiliang/scripts/lede/staging_dir/target-x86_64_musl/host/lib -Wl,-rpath,/home/zhiliang/scripts/lede/staging_dir/hostpkg/lib -Wl,--no-as-needed -lrt" make  -C /home/zhiliang/scripts/lede/build_dir/hostpkg/Python-3.9.12  python
make[4]: Entering directory '/home/zhiliang/scripts/lede/build_dir/hostpkg/Python-3.9.12'
gcc -pthread -L/home/zhiliang/scripts/lede/staging_dir/host/lib -L/home/zhiliang/scripts/lede/staging_dir/hostpkg/lib -L/home/zhiliang/scripts/lede/staging_dir/target-x86_64_musl/host/lib -Wl,-rpath,/home/zhiliang/scripts/lede/staging_dir/hostpkg/lib -Wl,--no-as-needed -lrt -L/home/zhiliang/scripts/lede/staging_dir/host/lib -L/home/zhiliang/scripts/lede/staging_dir/hostpkg/lib -L/home/zhiliang/scripts/lede/staging_dir/target-x86_64_musl/host/lib -Wl,-rpath,/home/zhiliang/scripts/lede/staging_dir/hostpkg/lib -Wl,--no-as-needed -lrt   -Xlinker -export-dynamic -o python Programs/python.o libpython3.9.a -lcrypt -lintl -lpthread -ldl  -lutil -lm -lz   -lm 
/usr/bin/ld: libpython3.9.a(_localemodule.o): in function `PyIntl_bindtextdomain':
/home/zhiliang/scripts/lede/build_dir/hostpkg/Python-3.9.12/./Modules/_localemodule.c:677: undefined reference to `libintl_bindtextdomain'
/usr/bin/ld: libpython3.9.a(_localemodule.o): in function `PyIntl_dcgettext':
/home/zhiliang/scripts/lede/build_dir/hostpkg/Python-3.9.12/./Modules/_localemodule.c:631: undefined reference to `libintl_dcgettext'
/usr/bin/ld: libpython3.9.a(_localemodule.o): in function `PyIntl_dgettext':
/home/zhiliang/scripts/lede/build_dir/hostpkg/Python-3.9.12/./Modules/_localemodule.c:617: undefined reference to `libintl_dgettext'
/usr/bin/ld: libpython3.9.a(_localemodule.o): in function `PyIntl_gettext':
/home/zhiliang/scripts/lede/build_dir/hostpkg/Python-3.9.12/./Modules/_localemodule.c:604: undefined reference to `libintl_gettext'
/usr/bin/ld: libpython3.9.a(_localemodule.o): in function `PyIntl_bind_textdomain_codeset':
/home/zhiliang/scripts/lede/build_dir/hostpkg/Python-3.9.12/./Modules/_localemodule.c:699: undefined reference to `libintl_bind_textdomain_codeset'
/usr/bin/ld: libpython3.9.a(_localemodule.o): in function `PyIntl_textdomain':
/home/zhiliang/scripts/lede/build_dir/hostpkg/Python-3.9.12/./Modules/_localemodule.c:644: undefined reference to `libintl_textdomain'
collect2: error: ld returned 1 exit status
make[4]: *** [Makefile:604: python] Error 1
make[4]: Leaving directory '/home/zhiliang/scripts/lede/build_dir/hostpkg/Python-3.9.12'
make[3]: *** [Makefile:373: /home/zhiliang/scripts/lede/build_dir/hostpkg/Python-3.9.12/.built] Error 2
make[3]: Leaving directory '/home/zhiliang/scripts/lede/feeds/packages/lang/python/python3'
time: package/feeds/packages/python3/host-compile#0.50#0.20#0.57
    ERROR: package/feeds/packages/python3 [host] failed to build.
make[2]: *** [package/Makefile:114: package/feeds/packages/python3/host/compile] Error 1
make[2]: Leaving directory '/home/zhiliang/scripts/lede'
make[1]: *** [package/Makefile:110: /home/zhiliang/scripts/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/zhiliang/scripts/lede'
make: *** [/home/zhiliang/scripts/lede/include/toplevel.mk:230: world] Error 2

WYC-2020, 2022-04-15:
万能大法

breakings, 2022-04-16:
我在本地测试x86和arm64都没任何问题

closed 2022-04-18 by AmadeusGhost