coolsnowwolf/lede #9262: 任子行SURF G-NAT200有时候会重启

WowRouter, 2022-04-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue
无
## 2. 详细叙述

### (1) 具体问题

A：任子行SURF G-NAT200使用最新编译的系统R22.4.1（内核5.4.188），运行openclash刚启动时有一定概率重启。不开程序，日常使用也会重启，重启频率大约1天一次。

### (2) 路由器型号和固件版本

A：任子行SURF G-NAT200 ipq8064   R22.4.1（内核5.4.188）

### (3) 详细日志

A：[129479.902635] [<c034b8d4>] (cpu_startup_entry) from [<4230254c>] (0x4230254c)
[129480.910189] Rebooting in 3 seconds..

U-Boot 2012.07 [Standard IPQ806X.LN,r187] (Mar 21 2016 - 15:22:08)

smem ram ptable found: ver: 0 len: 5
DRAM:  1003 MiB
NAND:  SF: Detected S25FL256S with page size 64 KiB, total 32 MiB
ipq_spi: page_size: 0x100, sector_size: 0x10000, size: 0x2000000
544 MiB
MMC:   
In:    serial
Out:   serial
Err:   serial
Net:   MAC1 addr:94:88:5e:8:e:7
S17 chip id: 0x1302
athrs17_reg_init: complete
athrs17_vlan_config ...done
S17c init  done
MAC2 addr:94:88:5e:8:e:8
eth0, eth1
Hit any key to stop autoboot:  0 
MMC Device 0 not found
MMC Device 0 not found
Device 1: nand1... is now current device

Loading from nand1, offset 0x320000
   Image Name:   ARM OpenWrt Linux-5.4.188
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    2930656 Bytes = 2.8 MiB
   Load Address: 42208000
   Entry Point:  42208000
Automatic boot of image at addr 0x44000000 ...
   Image Name:   ARM OpenWrt Linux-5.4.188
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    2930656 Bytes = 2.8 MiB
   Load Address: 42208000
   Entry Point:  42208000
   Verifying Checksum ... OK
   Loading Kernel Image ... OK
OK
info: "mtdparts" not set
Using machid 0x1260 from environment

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.4.188 (jixiang@ubuntu-esxi) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r4458-ad34521be)) #0 SMP Thu Apr 14 12:38:40 2022
[    0.000000] CPU: ARMv7 Processor [512f04d0] revision 0 (ARMv7), cr=10c5787d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[... 230 lines cut ...]
[   13.774474] Code: 0a000018 e59c2008 e1500002 058ce008 (158ce004) 
[   13.782700] ---[ end trace eb48c7d6fed88712 ]---
[   13.788767] Kernel panic - not syncing: Fatal exception in interrupt
[   13.793473] CPU1: stopping
[   13.799786] CPU: 1 PID: 0 Comm: swapper/1 Tainted: G      D W         5.4.188 #0
[   13.802305] Hardware name: Generic DT based system
[   13.809872] [<c030fad4>] (unwind_backtrace) from [<c030b9f0>] (show_stack+0x14/0x20)
[   13.814469] [<c030b9f0>] (show_stack) from [<c09034d8>] (dump_stack+0x94/0xa8)
[   13.822365] [<c09034d8>] (dump_stack) from [<c030ed00>] (handle_IPI+0x184/0x1b8)
[   13.829398] [<c030ed00>] (handle_IPI) from [<c05dd328>] (gic_handle_irq+0xb4/0xb8)
[   13.836949] [<c05dd328>] (gic_handle_irq) from [<c0301b0c>] (__irq_svc+0x6c/0x90)
[   13.844321] Exception stack(0xef08ff18 to 0xef08ff60)
[   13.851876] ff00:                                                       00000000 00000003
[   13.856934] ff20: 2ebdf000 ef720a80 ee888800 00000000 ef71fe30 00000003 00000003 00000000
[   13.865094] ff40: 36278240 35e27600 00000015 ef08ff68 c0731688 c073168c 00000113 ffffffff
[   13.873242] [<c0301b0c>] (__irq_svc) from [<c073168c>] (cpuidle_enter_state+0x94/0x498)
[   13.881394] [<c073168c>] (cpuidle_enter_state) from [<c0731ad4>] (cpuidle_enter+0x30/0x4c)
[   13.889206] [<c0731ad4>] (cpuidle_enter) from [<c034b5c4>] (do_idle+0x1d8/0x240)
[   13.897539] [<c034b5c4>] (do_idle) from [<c034b8d4>] (cpu_startup_entry+0x1c/0x20)
[   13.905090] [<c034b8d4>] (cpu_startup_entry) from [<4230254c>] (0x4230254c)
[   13.912482] Rebooting in 1 seconds..

U-Boot 2012.07 [Standard IPQ806X.LN,r187] (Mar 21 2016 - 15:22:08)

smem ram ptable found: ver: 0 len: 5

coolsnowwolf, 2022-04-18:
[image]

内存越界了？ 可以测试看看把内存限制到 512MB 看看

yaoyuanzy, 2022-08-23:
球球大佬，TTL 要特殊改动吗 接线115200没反应

flexyangjie, 2022-10-31:
这位老哥不知道有没有修复这个问题，目前依然存在。感觉这应该不是固件问题，而是与UBOOT相关。

WowRouter, 2022-11-02:
> 

uboot啥问题？我的依然会重启，源码更新也编译过几次固件了，都不行

WowRouter, 2022-11-02:
> 这位老哥不知道有没有修复这个问题，目前依然存在。感觉这应该不是固件问题，而是与UBOOT相关。

uboot啥问题？我的依然会重启，源码更新也编译过几次固件了，都不行

flexyangjie, 2022-11-03:
我也是与你一样的情况。既然源码没问题，那只好怀疑到uboot上了。当然只是猜测。这机器怕是要QSDK固件才能玩得转了。我网上爬文的时候无意有看到有人讲这种商用机的uboot怎么的，要换别的固件就得换uboot。我现在找不到这个文章了。

closed 2026-09-05 by coolsnowwolf