coolsnowwolf/lede #9264: 一顿git pull,feeds update后,准备install发现很多depends没有，怎么回事？

Travel2Here, 2022-04-17:
截取片段：

Installing package 'transmission-web-control' from packages
WARNING: No feed for package 'libc' found
WARNING: No feed for package 'librt' found
WARNING: No feed for package 'libpthread' found
Installing package 'travelmate' from packages
WARNING: No feed for package 'libc' found
WARNING: No feed for package 'librt' found
WARNING: No feed for package 'libpthread' found
WARNING: No feed for package 'iwinfo' found
WARNING: No feed for package 'jshn' found
WARNING: No feed for package 'jsonfilter' found
WARNING: No feed for package 'ca-bundle' found
Installing package 'tree' from packages
WARNING: No feed for package 'libc' found
WARNING: No feed for package 'librt' found
WARNING: No feed for package 'libpthread' found
WARNING: No feed for package 'libgcc' found
Installing package 'triggerhappy' from packages
WARNING: No feed for package 'libc' found
WARNING: No feed for package 'librt' found
WARNING: No feed for package 'libpthread' found
Installing package 'ttyd' from packages
WARNING: No feed for package 'libc' found
WARNING: No feed for package 'librt' found
WARNING: No feed for package 'libpthread' found
WARNING: No feed for package 'libopenssl' found
WARNING: No feed for package 'libjson-c' found
WARNING: No feed for package 'zlib' found
Installing package 'tvheadend' from packages
WARNING: No feed for source package 'libiconv' found
WARNING: No feed for source package 'gettext' found

Travel2Here, 2022-04-17:
补充我的所有feeds:

src-git packages https://github.com/coolsnowwolf/packages
src-git luci https://github.com/coolsnowwolf/luci
src-git routing https://git.openwrt.org/feed/routing.git
src-git telephony https://git.openwrt.org/feed/telephony.git

src-git kenzo https://github.com/kenzok8/openwrt-packages
src-git small https://github.com/kenzok8/small
src-git passwall https://github.com/xiaorouji/openwrt-passwall
src-git helloworld https://github.com/fw876/helloworld

jin3014test2, 2022-04-17:
用了国内加速镜像了吧

Travel2Here, 2022-04-17:
> 用了国内加速镜像了吧

没啊 你看我上面的feeds列表

closed 2022-04-18 by AmadeusGhost