coolsnowwolf/lede #9266: 更新5.10.111后编译出错

mitsukileung, 2022-04-17:
make[3]: Entering directory '/workdir/openwrt/feeds/routing/batman-adv'
mkdir -p /workdir/openwrt/dl
SHELL= flock /workdir/openwrt/tmp/.batman-adv-2022.0.tar.gz.flock -c '  	/workdir/openwrt/scripts/download.pl "/workdir/openwrt/dl" "batman-adv-2022.0.tar.gz" "49338705bc207709ac84d766688e702571009c827c0a320788ea51fb887714aa" "" "https://downloads.open-mesh.org/batman/releases/batman-adv-2022.0"    '
rm -f /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/batman-adv-2022.0/.built
touch /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/batman-adv-2022.0/.built_check
make  -C "/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.10.111" KCFLAGS="-fmacro-prefix-map=/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl=target-aarch64_cortex-a53_musl" HOSTCFLAGS="-O2 -I/workdir/openwrt/staging_dir/host/include -I/workdir/openwrt/staging_dir/hostpkg/include -I/workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="aarch64-openwrt-linux-musl-" ARCH="arm64" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Sun Apr 17 05:21:27 2022" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/workdir/openwrt/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/workdir/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KBUILD_EXTRA_SYMBOLS="/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/symvers/cryptodev-linux.symvers /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/symvers/shortcut-fe-simulated-driver.symvers /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/symvers/gpio-button-hotplug.symvers /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/symvers/qca-nss-drv-64.symvers /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/symvers/nat46.symvers /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/symvers/mac80211.symvers /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/symvers/shortcut-fe.symvers /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/symvers/qca-nss-dp.symvers /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/symvers/qca-ssdk.symvers" KERNELRELEASE=5.10.111 M="/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/batman-adv-2022.0/net/batman-adv" CONFIG_BATMAN_ADV=m CONFIG_BATMAN_ADV_DEBUG=n CONFIG_BATMAN_ADV_BLA=y CONFIG_BATMAN_ADV_DAT=y CONFIG_BATMAN_ADV_MCAST=y CONFIG_BATMAN_ADV_NC=n CONFIG_BATMAN_ADV_BATMAN_V=y CONFIG_BATMAN_ADV_TRACING=n  EXTRA_CFLAGS=" -DCONFIG_BATMAN_ADV=1  -DCONFIG_BATMAN_ADV_BLA=1  -DCONFIG_BATMAN_ADV_DAT=1  -DCONFIG_BATMAN_ADV_MCAST=1  -DCONFIG_BATMAN_ADV_BATMAN_V=1 " NOSTDINC_FLAGS="-nostdinc  -isystem /workdir/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/include -I/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/batman-adv-2022.0/net/batman-adv -I/workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/usr/include/mac80211-backport -I/workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/usr/include/mac80211-backport/uapi -I/workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/usr/include/mac80211 -I/workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/usr/include/mac80211/uapi -I/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/batman-adv-2022.0/include/ -include backport/autoconf.h -include backport/backport.h -include /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/batman-adv-2022.0/compat-hacks.h -DBATADV_SOURCE_VERSION=\\\"2022.0-openwrt-1\\\"" modules
make[4]: Entering directory '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.10.111'
  CC [M]  /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/batman-adv-2022.0/net/batman-adv/multicast.o
/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/batman-adv-2022.0/net/batman-adv/multicast.c: In function 'batadv_mcast_mla_rtr_flags_softif_get_ipv6':
/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/batman-adv-2022.0/net/batman-adv/multicast.c:137:14: error: invalid operands to binary && (have 'int' and 'atomic_t' {aka 'struct <anonymous>'})
  if (in6_dev && in6_dev->cnf.mc_forwarding)
      ~~~~~~~ ^~ ~~~~~~~~~~~~~~~~~~~~~~~~~~
/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/batman-adv-2022.0/net/batman-adv/multicast.c:141:1: error: control reaches end of non-void function [-Werror=return-type]
 }
 ^
cc1: some warnings being treated as errors
make[5]: *** [scripts/Makefile.build:280: /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/batman-adv-2022.0/net/batman-adv/multicast.o] Error 1
make[4]: *** [Makefile:1822: /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/batman-adv-2022.0/net/batman-adv] Error 2
make[4]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.10.111'
make[3]: *** [Makefile:95: /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/batman-adv-2022.0/.built] Error 2
make[3]: Leaving directory '/workdir/openwrt/feeds/routing/batman-adv'
time: package/feeds/routing/batman-adv/compile#0.95#0.34#1.27
    ERROR: package/feeds/routing/batman-adv failed to build.
make[2]: *** [package/Makefile:116: package/feeds/routing/batman-adv/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2

大佬们有遇到吗？

JoseCoW, 2022-04-17:
kernel 部分改了https://github.com/gregkh/linux/commit/145c7a793838add5e004e7d49a67654dc7eba147 ，影响了 batman-dev

mitsukileung, 2022-04-17:
> kernel 部分改了[gregkh/linux@145c7a7](https://github.com/gregkh/linux/commit/145c7a793838add5e004e7d49a67654dc7eba147) ，影响了 batman-dev-2022.0

大佬，具体要怎么改？你发的链接目录架构和LEDE不同

mitsukileung, 2022-04-17:
看到大佬的PR了，谢谢

WYC-2020, 2022-04-17:
这个我个人不建议这样改，更喜欢修改应用软件atomic_read

JoseCoW, 2022-04-17:
> 这个我个人不建议这样改，更喜欢修改应用软件atomic_read

这个在上游，改了又会影响还在低版本内核的，我就给个参考吧，还望大佬们指教

WYC-2020, 2022-04-17:
> > 这个我个人不建议这样改，更喜欢修改应用软件atomic_read
> 
> 这个在上游，改了又会影响还在低版本内核的，我就给个参考吧，还望大佬们指教

这个给那个驱动打个patch，算了明天我看下

JoseCoW, 2022-04-17:
> > > 这个我个人不建议这样改，更喜欢修改应用软件atomic_read
> > 
> > 
> > 这个在上游，改了又会影响还在低版本内核的，我就给个参考吧，还望大佬们指教
> 
> 这个给那个驱动打个patch，算了明天我看下

https://github.com/JoseCoW/routing/commit/daf685cd26d1a668f9a1f6639405e1d386eefa8b 这个我也 fork 下来改过

WYC-2020, 2022-04-17:
> > > > 这个我个人不建议这样改，更喜欢修改应用软件atomic_read
> > > 
> > > 
> > > 这个在上游，改了又会影响还在低版本内核的，我就给个参考吧，还望大佬们指教
> > 
> > 
> > 这个给那个驱动打个patch，算了明天我看下
> 
> [JoseCoW/routing@daf685c](https://github.com/JoseCoW/routing/commit/daf685cd26d1a668f9a1f6639405e1d386eefa8b) 这个我也 fork 下来改过

这样是不行的，兼容不了所有内核版本，还得加宏，明天看了才知道具体怎么改，我不赞成改那个内核是因为既然他用的是原子类型肯定存在多线程情况，直接改成普通类型，不排除会出问题

JoseCoW, 2022-04-17:
> 这样是不行的，兼容不了所有内核版本，还得加宏，明天看了才知道具体怎么改，我不赞成改那个内核是因为既然他用的是原子类型肯定存在多线程情况，直接改成普通类型，不排除会出问题

好的

WYC-2020, 2022-04-18:
> 好的

今天看了一下，有两种改法，一种是 直接在代码里加一个宏，如果是小于5.10的就用原来的逻辑，如果大于等于5.10的就用atomic_read，还有一种是直接吧5.4的也修复成用atomic类型，然后驱动里面就不用加宏了直接全用atomic_read，我倾向第二种，看你们喜欢那种，我先按我第二种改了

WYC-2020, 2022-04-18:
其实我看这个驱动 linux 内核是自带有的 我是不知道为啥 要去单独使用另外的，是linux 配套的不够好还是咋的

WYC-2020, 2022-04-18:
这个问题我看可以关闭了，上游已经有人提了，只要上游更新了就没问题了，这个类型在5.4.189版本后也会变成和5.10的一样，所以问题不大，暂时就不用加patch了，哈哈

mitsukileung, 2022-04-18:
> 这个问题我看可以关闭了，上游已经有人提了，只要上游更新了就没问题了，这个类型在5.4.189版本后也会变成和5.10的一样，所以问题不大，暂时就不用加patch了，哈哈

大佬说的是这个吗？https://github.com/openwrt/routing/pull/852

WYC-2020, 2022-04-18:
> > 这个问题我看可以关闭了，上游已经有人提了，只要上游更新了就没问题了，这个类型在5.4.189版本后也会变成和5.10的一样，所以问题不大，暂时就不用加patch了，哈哈
> 
> 大佬说的是这个吗？[openwrt/routing#852](https://github.com/openwrt/routing/pull/852)

[image]
慢慢等上游合并吧，自己改的话还需要fork这个仓库，懒得搞，没时间去维护，偷下懒

mitsukileung, 2022-04-18:
刚刚试了云编译，成功编译

closed 2022-04-18 by mitsukileung