coolsnowwolf/lede #9272: softether V4.x编译不成功

badgv, 2022-04-18:
云编译环境，直接git clone全新编译，没有编译缓存，使用make -j1 V=sc得到的日志，在VPN菜单中选了softether server，自动选中了softether base，依赖环境是主页提供的依赖
报错的日志,大佬们帮看看是怎么回事：
[image]

badgv, 2022-04-18:
@march1993  大佬有空帮看看怎么回事吗

march1993, 2022-04-18:
ncurses 相关的库出了问题吧 可能是没编译

badgv, 2022-04-18:
> ncurses 相关的库出了问题吧 可能是没编译

这把云编译我是专门设置的make -j1 V=sc，ncurses是需要apt安装吗？我看了下主页提供的依赖里面好像有libncurses5-dev libncursesw5-dev这两个包的安装

march1993, 2022-04-18:
在 make menuconfig 的时候勾上 ncurses 相关的试试看？

badgv, 2022-04-18:
> 在 make menuconfig 的时候勾上 ncurses 相关的试试看？

我去试试，试完再来回复

march1993, 2022-04-18:
还有我怀疑你的出错信息不一定在你截图的地方，往上找找

badgv, 2022-04-18:
> 还有我怀疑你的出错信息不一定在你截图的地方，往上找找

[7_Build AMD64.zip](https://github.com/coolsnowwolf/lede/files/8504710/7_Build.AMD64.zip)

大佬这个是build的日志，有空帮看下吗

march1993, 2022-04-18:
> > 还有我怀疑你的出错信息不一定在你截图的地方，往上找找
> 
> [7_Build AMD64.zip](https://github.com/coolsnowwolf/lede/files/8504710/7_Build.AMD64.zip)
> 
> 大佬这个是build的日志，有空帮看下吗

你这个zip我怎么打不开

badgv, 2022-04-18:
> > > 还有我怀疑你的出错信息不一定在你截图的地方，往上找找
> > 
> > 
> > [7_Build AMD64.zip](https://github.com/coolsnowwolf/lede/files/8504710/7_Build.AMD64.zip)
> > 大佬这个是build的日志，有空帮看下吗
> 
> 你这个zip我怎么打不开

不好意思，刚才那个zip有问题，我重新上传了下

march1993, 2022-04-18:
你说的好像没错，hamcorebuilder是在host环境下的，用apt装上ncurses相关的包吧

badgv, 2022-04-18:
> 你说的好像没错，hamcorebuilder是在host环境下的，用apt装上ncurses相关的包吧

我在本地的ubuntu试了下
[image]
安装这个提示是用的**libncurses**代替了，主页提供的是libncurses5-dev libncursesw5-dev
我现在是需要加个libncurses-dev这个依赖吗？

march1993, 2022-04-18:
一个个试试。。

badgv, 2022-04-18:
> 一个个试试。。

我已经直接加了个ncurses* 的依赖，已经开始在跑了，1个多小时出结果

badgv, 2022-04-18:
> 一个个试试。。

大佬，加了 ncurses*  这个还是编译失败了，搞不定了

march1993, 2022-04-18:
这个是我机器上的包
```shell
$ dpkg --list |grep ncurse
ii  libncurses-dev:amd64                 6.2+20201114-2                     amd64        developer's libraries for ncurses
ii  libncurses6:amd64                    6.2+20201114-2                     amd64        shared libraries for terminal handling
ii  libncursesw6:amd64                   6.2+20201114-2                     amd64        shared libraries for terminal handling (wide character support)
ii  ncurses-base                         6.2+20201114-2                     all          basic terminal type definitions
ii  ncurses-bin                          6.2+20201114-2                     amd64        terminal-related programs and man pages
ii  ncurses-term                         6.2+20201114-2                     all          additional terminal type definitions
```

badgv, 2022-04-18:
> 这个是我机器上的包
>
> ```shell
> $ dpkg --list |grep ncurse
> ii  libncurses-dev:amd64                 6.2+20201114-2                     amd64        developer's libraries for ncurses
> ii [...]
我本地的包好像和你这个差不多，编译能成功，按理说我下面给云编译环境直接加的ncurses*   我在本地试了下，应该是包含了这几个包的
[image]

march1993, 2022-04-18:
libncurses6 和 libncursesw6 呢

WYC-2020, 2022-04-18:
什么鬼哦，这个依赖的库 不是本机安装 而是需要openwrt 的编译环境编译吧，

badgv, 2022-04-18:
> 什么鬼哦，这个依赖的库 不是本机安装 而是需要openwrt 的编译环境编译吧，

现在就是不知道为什么云编译会失败，我本地编译都成功了，同一个.config文件，云编译失败，本地编译成功的

march1993, 2022-04-18:
> 什么鬼哦，这个依赖的库 不是本机安装 而是需要openwrt 的编译环境编译吧，

那就更奇怪了。。勾选softether的时候不应该会自动带上依赖吗。。

badgv, 2022-04-18:
> libncurses6 和 libncursesw6 呢

我看了下，云服务器编译的apt日志，只有
[image]
[image]
确实没有你上面说的这个libncurses6 和 libncursesw6

WYC-2020, 2022-04-18:
> > 什么鬼哦，这个依赖的库 不是本机安装 而是需要openwrt 的编译环境编译吧，
> 
> 现在就是不知道为什么云编译会失败，我本地编译都成功了，同一个.config文件，云编译失败，本地编译成功的

云编译启用了缓存么 不要启用试下

WYC-2020, 2022-04-18:
> > 什么鬼哦，这个依赖的库 不是本机安装 而是需要openwrt 的编译环境编译吧，
> 
> 那就更奇怪了。。勾选softether的时候不应该会自动带上依赖吗。。

玄学自己体会哈哈

badgv, 2022-04-18:
> > > 什么鬼哦，这个依赖的库 不是本机安装 而是需要openwrt 的编译环境编译吧，
> > 
> > 
> > 现在就是不知道为什么云编译会失败，我本地编译都成功了，同一个.config文件，云编译失败，本地编译成功的
> 
> 云编译启用了缓存么 不要启用试下

云编译每次都是git clone开始的全新安装，没有缓存的

march1993, 2022-04-18:
@badgv 我看了 libncurse 的源码，v5.x 的时候的确没有 `_nc_tiparm`，v6.x 才有

WYC-2020, 2022-04-18:
> > > > 什么鬼哦，这个依赖的库 不是本机安装 而是需要openwrt 的编译环境编译吧，
> > > 
> > > 
> > > 现在就是不知道为什么云编译会失败，我本地编译都成功了，同一个.config文件，云编译失败，本地编译成功的
> > 
> > 
> > 云编译启用了缓存么 不要启用试下
> 
> 云编译每次都是git clone开始的全新安装，没有缓存的

云编译是可以开启缓存的自己去看 pr里面的那个

badgv, 2022-04-18:
> @badgv 我看了 libncurse 的源码，v5.x 的时候的确没有 `_nc_tiparm`，v6.x 才有

[image]
我添加了几段脚本，打算重新跑一把看下状态

WYC-2020, 2022-04-18:
> 

都说了 这个不是你机器安装的那个 是openwrt 需要去编译的 op自带的就是6.x系列的 。makefile 在package/libs里面 自己去看

WYC-2020, 2022-04-18:
openwrt 会去编译这个版本 wenyanchao@ubuntu:~/lede/package/libs/ncurses$ cat Makefile 
#
# Copyright (C) 2006-2015 OpenWrt.org
#
# This is free software, licensed under the GNU General Public License v2.
# See /LICENSE for more information.
#

include $(TOPDIR)/rules.mk

PKG_NAME:=ncurses
PKG_VERSION:=6.3
PKG_RELEASE:=$(AUTORELEASE)

PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.gz
PKG_SOURCE_URL:=@GNU/$(PKG_NAME)
PKG_HASH:=97fc51ac2b085d4cde31ef4d2c3122c21abc217e9090a43a30fc5ec21684e059

march1993, 2022-04-18:
> > 
> 
> 都说了 这个不是你机器安装的那个 是openwrt 需要去编译的 op自带的就是6.x系列的 。makefile 在package/libs里面 自己去看

hamcorebuilder 是一个在 host 编译的二进制

WYC-2020, 2022-04-18:
> > > 
> > 
> > 
> > 都说了 这个不是你机器安装的那个 是openwrt 需要去编译的 op自带的就是6.x系列的 。makefile 在package/libs里面 自己去看
> 
> hamcorebuilder 是一个在 host 编译的二进制

$(eval $(call HostBuild))
$(eval $(call BuildPackage,terminfo))
$(eval $(call BuildPackage,libncurses))
$(eval $(call BuildPackage,libncurses-dev))

badgv, 2022-04-18:
> > > 
> > 
> > 
> > 都说了 这个不是你机器安装的那个 是openwrt 需要去编译的 op自带的就是6.x系列的 。makefile 在package/libs里面 自己去看
> 
> hamcorebuilder 是一个在 host 编译的二进制

[image]
大佬，云编译的apt环境安装的包是带了这些的

march1993, 2022-04-18:
> > > > 
> > > 
> > > 
> > > 都说了 这个不是你机器安装的那个 是openwrt 需要去编译的 op自带的就是6.x系列的 。makefile 在package/libs里面 自己去看
> > 
> > 
> > hamcorebuilder 是一个在 host 编译的二进制
> 
> $(eval $(call HostBuild)) $(eval $(call BuildPackage,terminfo)) $(eval $(call BuildPackage,libncurses)) $(eval $(call BuildPackage,libncurses-dev))

我在 makefile 里就看到 guest 依赖有
https://github.com/coolsnowwolf/packages/blob/a1eaff103840bf61eb9742cfd98604b4934a29fd/net/softethervpn/Makefile#L107
但是 host 编译阶段没找到依赖啊

march1993, 2022-04-18:
> > > > 
> > > 
> > > 
> > > 都说了 这个不是你机器安装的那个 是openwrt 需要去编译的 op自带的就是6.x系列的 。makefile 在package/libs里面 自己去看
> > 
> > 
> > hamcorebuilder 是一个在 host 编译的二进制
> 
> [image] 大佬，云编译的apt环境安装的包是带了这些的

大佬 试过换系统吗 你本地和云端都是同版本的ubuntu吗？能换debian试试看吗？

badgv, 2022-04-18:
> 

云编译脚本用的ubuntu-latest，意思是换成debian试下？我去试试

badgv, 2022-04-18:
> > > > > 
> > > > 
> > > > 
> > > > 都说了 这个不是你机器安装的那个 是openwrt 需要去编译的 op自带的就是6.x系列的 。makefile 在package/libs里面 自己去看
> > > 
> > > 
> > > hamcorebuilder 是一个在 host 编译的二进制
> > 
> > 
> > [image] 大佬，云编译的apt环境安装的包是带了这些的
> 
> 大佬 试过换系统吗 你本地和云端都是同版本的ubuntu吗？能换debian试试看吗？

好像云编译环境只有个ubuntu-18.04和ubuntu-latest，我换个18.04试下

WYC-2020, 2022-04-18:
这玩意我刚试了一下,本地都编译不过的，有时间再看，现在没时间
十有八九有坑哈哈
[image]

march1993, 2022-04-18:
> 这玩意我刚试了一下,本地都编译不过的，有时间再看，现在没时间 十有八九有坑哈哈 [image]

看手册啊
https://github.com/SoftEtherVPN/SoftEtherVPN_Stable/blob/master/src/BUILD_UNIX.TXT
要装readline依赖的

badgv, 2022-04-18:
> 这玩意我刚试了一下,本地都编译不过的，有时间再看，现在没时间 十有八九有坑哈哈 [image]

还真报错一毛一样的，不过我本地前几天编译arm64和x86都成功了，最近本地直接git pull后一直能成功

march1993, 2022-04-18:
我能给的唯一成功经验是前两天我用 debian11 编译。。一遍过。。

march1993, 2022-04-18:
> > 这玩意我刚试了一下,本地都编译不过的，有时间再看，现在没时间 十有八九有坑哈哈 [image]
> 
> 还真报错一毛一样的，不过我本地前几天编译arm64和x86都成功了，最近本地直接git pull后一直能成功

不一样，他这个连readline都没装

badgv, 2022-04-18:
> 我能给的唯一成功经验是前两天我用 debian11 编译。。一遍过。。

我本地的ubuntu也能编译成功，不过我这ubuntu已经安装好久了，也好久没有apt upgrade啥的了，就这云编译环境不行

WYC-2020, 2022-04-18:
我就编译这一个，其他什么都没选，有时间在看下吧，这玩意居然还有人用哈哈

march1993, 2022-04-18:
> 我就编译这一个，其他什么都没选，有时间在看下吧，这玩意居然还有人用哈哈

不用不知道它的香，有个图形界面的管理器，方便的很，支持绝大部分设备接入。

WYC-2020, 2022-04-18:
[image]
九十九是这个提交导致我那个问题

march1993, 2022-04-18:
我也不知道，得进入具体的他那个虚拟机看一下才行。
不过盲猜链接过程没问题，毕竟 ld 没报错。

WYC-2020, 2022-04-19:
搞得问题就我说的那样，很多人跑去官方去拉去提交过来结果有的没有拉去全面 导致的，我这边测试没问题了溜了

badgv, 2022-04-19:
> 

大佬是本地编译测试的吗？方便试下云编译能否成功呢？

WYC-2020, 2022-04-19:
> > 
> 
> 大佬是本地编译测试的吗？方便试下云编译能否成功呢？

懒得测你自己拉去我的仓库去测试，云编译那么慢 没时间折腾

march1993, 2022-04-19:
@WYC-2020 那为啥我拉的 coolsnowwolf/lede 直接就能编译呢。。

WYC-2020, 2022-04-19:
> @WYC-2020 那为啥我拉的 coolsnowwolf/lede 直接就能编译呢。。

你可以试下 其他全部删除 然后其他什么都不选 然后x86  只编译softether server这个看下

march1993, 2022-04-19:
> > @WYC-2020 那为啥我拉的 coolsnowwolf/lede 直接就能编译呢。。
> 
> 你可以试下 其他全部删除 然后其他什么都不选 然后x86 只编译softether server这个看下

我就是 git clone 下来，feed update，然后 make menuconfig ，勾上 softether4，最后 make。。其他什么都没动过啊。。

WYC-2020, 2022-04-19:
> @WYC-2020 那为啥我拉的 coolsnowwolf/lede 直接就能编译呢。。

再说能直接编译过还不好吗?哈哈 玄学问题没得解释，理论上都应该会出现找不到这个库才对，因为那个提交已经去掉了链接的时候寻找目录

WYC-2020, 2022-04-19:
并且那个readline 的makefile 也是有问题的 有一个地方安装路径有问题

march1993, 2022-04-19:
玄学了

WYC-2020, 2022-04-19:
反正我只管修我仓库遇到的问题 哈哈，至于l大的 等你们遇到了 我在说

> **[march1993](/march1993) ** commented [1分钟前](#issuecomment-1101976866)

march1993, 2022-04-19:
> 反正我只管修我仓库遇到的问题 哈哈，至于l大的 等你们遇到了 我在说
> 
> > **[march1993](/march1993) ** commented [1分钟前](#issuecomment-1101976866)

有维护 package 的 ipk 库吗

WYC-2020, 2022-04-19:
> > > @WYC-2020 那为啥我拉的 coolsnowwolf/lede 直接就能编译呢。。
> > 
> > 
> > 你可以试下 其他全部删除 然后其他什么都不选 然后x86 只编译softether server这个看下
> 
> 我就是 git clone 下来，feed update，然后 make menuconfig ，勾上 softether4，最后 make。。其他什么都没动过啊。。

我知道是什么原因 ，你是在luci app 哪里选择的luci-app-softethervpn 这个插件吧，不是从这里选择的
[image]

march1993, 2022-04-19:
> > > > @WYC-2020 那为啥我拉的 coolsnowwolf/lede 直接就能编译呢。。
> > > 
> > > 
> > > 你可以试下 其他全部删除 然后其他什么都不选 然后x86 只编译softether server这个看下
> > 
> > 
> > 我就是 git clone 下来，feed update，然后 make menuconfig ，勾上 softether4，最后 make。。其他什么都没动过啊。。
> 
> 我知道是什么原因 ，你是在luci app 哪里选择的luci-app-softethervpn 这个插件吧，不是从这里选择的 [image]

怎么可能。https://github.com/coolsnowwolf/packages/pull/409 我测试我的patch肯定要编译server的

WYC-2020, 2022-04-19:
不要再luci 哪里选 那个luci 哪里选了 是依赖的5 不是依赖的上面那个

[8 more replies]

closed 2026-09-05 by coolsnowwolf