coolsnowwolf/lede #9273: 抵制openmesh重复模块，wifi自带wds桥接默认，也自带ap默认，也自带移动域，也自带KVR

zhoujinshi, 2022-04-19:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译openmesh容易报错

### (2) 路由器型号和固件版本

A：youkul2/newifi3d2

### (3) 详细日志

A：

JoseCoW, 2022-04-19:
没看懂。😅

WYC-2020, 2022-04-19:
他应该说的是 这个依赖的batman-adv 驱动吧，这个 其实每个linux 内核自带有的

> 没看懂。😅

JoseCoW, 2022-04-19:
> 他应该说的是 这个依赖的batman-adv 驱动吧，这个 其实每个linux 内核自带有的
> 
然后呢？

WYC-2020, 2022-04-19:
> > 他应该说的是 这个依赖的batman-adv 驱动吧，这个 其实每个linux 内核自带有的
> 
> 然后呢？

然后就是把linux自带的这个提取成模块，依赖这个模块，不要去依赖rou仓库的这个第三方驱动

JoseCoW, 2022-04-19:
除了编译 easymesh，还有啥会调用到这个呢？

WYC-2020, 2022-04-19:
然后也就不会出现 这个第三方驱动和内核驱动不匹配导致 编译问题了哈哈

WYC-2020, 2022-04-19:
其实这个插件我就没用过，感觉没啥卵用，可能你们喜欢玩这个mesh组合，喜欢折腾吧

JoseCoW, 2022-04-19:
上游修好问题了啊，我这边，编译正常，测试正常。

WYC-2020, 2022-04-19:
> 上游修好问题了啊，我这边，编译正常，测试正常。
我知道，这个可能只是这个问题发起者的一个疑问，算了反正无所谓的，我又不需要这个插件

WYC-2020, 2022-04-19:
第三方驱动和内核自带的驱动唯一的优势就是可能功能更多，但是最大的问题就是兼容性，有的驱动使用的函数在新版的linux内核中可能就变了，导致编译不通过，linux自带的肯定是和当前内核版本匹配的这也是一大优势之一

JoseCoW, 2022-04-19:
问题发起者跑错场了，请到 [上游](https://github.com/openwrt/routing) 抒发自己抵制

WYC-2020, 2022-04-19:
有时间我到是可以提取这个linux 自带的模块出来

WYC-2020, 2022-04-19:
define KernelPackage/native-batman-adv
  SUBMENU:=$(NETWORK_SUPPORT_MENU)
  TITLE:=B.A.T.M.A.N. Basic Module
  DEPENDS:=+kmod-lib-crc32c
  KCONFIG:= \
	CONFIG_BATMAN_ADV \
	CONFIG_BATMAN_ADV_BATMAN_V=y \
	CONFIG_BATMAN_ADV_BLA=y \
	CONFIG_BATMAN_ADV_DAT=y \
	CONFIG_BATMAN_ADV_NC =n \
	CONFIG_BATMAN_ADV_MCAST=y \
	CONFIG_BATMAN_ADV_DEBUG=n \
	CONFIG_BATMAN_ADV_TRACING=n \
	CONFIG_BATMAN_ADV_SYSFS =n
  FILES:= \
	$(LINUX_DIR)/net/batman-adv/batman-adv.ko
	
   AUTOLOAD:=$(call AutoProbe,batman-adv)
endef

define KernelPackage/native-batman-adv/description
	  B.A.T.M.A.N. (better approach to mobile ad-hoc networking) is
	  a routing protocol for multi-hop ad-hoc mesh networks. The
	  networks may be wired or wireless.
endef

$(eval $(call KernelPackage,native-batman-adv))
 稍微提取了一下，没有在进行细分，看都是bool 如果细分还要给内核打patch 懒得搞，不怕死的可以拿去测试功能哈哈

WYC-2020, 2022-04-19:
https://github.com/WYC-2020/lede/commit/f6cec788776b8436ae3ab787a81af6b40e7839a7

closed 2022-04-19 by coolsnowwolf