coolsnowwolf/lede #9279: rblibtorrent  缺少依赖

tqangxl, 2022-04-19:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：find /home/james/test/openwrt-coolsnowwolf/build_dir/target-powerpc_464fp_musl/rblibtorrent-1.2.15/ipkg-powerpc_464fp/rblibtorrent -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
Package rblibtorrent is missing dependencies for the following libraries:
libatomic.so.1
make[3]: *** [Makefile:71: /home/james/test/openwrt-coolsnowwolf/bin/packages/powerpc_464fp/packages/rblibtorrent_1.2.15-2_powerpc_464fp.ipk] Error 1
make[3]: Leaving directory '/home/james/test/openwrt-coolsnowwolf/feeds/packages/libs/rblibtorrent'
time: package/feeds/packages/rblibtorrent/compile#473.50#37.18#505.43
    ERROR: package/feeds/packages/rblibtorrent failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/rblibtorrent/compile] Error 1
make[2]: Leaving directory '/home/james/test/openwrt-coolsnowwolf'
make[1]: *** [package/Makefile:110: /home/james/test/openwrt-coolsnowwolf/staging_dir/target-powerpc_464fp_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/james/test/openwrt-coolsnowwolf'
make: *** [/home/james/test/openwrt-coolsnowwolf/include/toplevel.mk:230：world]

tqangxl, 2022-04-19:
需要修改feeds/packages/libs/rblibtorrent/Makefile
在DEPENDS:=+libgcc +libstdcpp +libopenssl +boost +boost-system +boost-chrono +boost-random **+libatomic**

再make package/feeds/packages/rblibtorrent/{clean,compile} V=99

breakings, 2022-04-21:
我
[image]
PR了新版，也加上这个依赖了，不过大雕源默认还是用1代库，我自己现在一直用2代库，目前使用上没什么问题。

lingandken, 2022-04-21:
需要修改feeds/packages/libs/rblibtorrent/Makefile
在DEPENDS:=+libgcc +libstdcpp +libopenssl +boost +boost-system +boost-chrono +boost-random +libatomic

我用这个方案通过了编译 从Luci启动后 通过端口访问不到页面 浏览器不停转圈圈 打不开webui

breakings, 2022-04-21:
> 需要修改feeds/packages/libs/rblibtorrent/Makefile 在DEPENDS:=+libgcc +libstdcpp +libopenssl +boost +boost-system +boost-chrono +boost-random +libatomic
> 
> 我用这个方案通过了编译 从Luci启动后 通过端口访问不到页面 浏览器不停转圈圈 打不开webui

转⭕️也可能没启动或者端口被占用 `lsof -i:8080` 看下
再终端运行一下 `/usr/bin/qbittorrent-nox --profile=/etc`
再检查一下动态库 `ldd /usr/bin/qbittorrent-nox`

lingandken, 2022-04-21:
端口换过 还是打不开 
今晚回去试试看

lingandken, 2022-04-22:
还有一个问题 我使用

https://www.right.com.cn/forum/forum.php?mod=viewthread&tid=1456090&highlight=qbittorrent

这位大哥编译的安装包 安装成功了 但luck界面没有qb 的选项(各个版本都试了)求教

breakings, 2022-04-22:
> 还有一个问题 我使用
> 
> https://www.right.com.cn/forum/forum.php?mod=viewthread&tid=1456090&highlight=qbittorrent
> 
> 这位大哥编译的安装包 安装成功了 但luck界面没有qb 的选项(各个版本都试了)求教

大概率是js版

lingandken, 2022-04-22:
[image]
。。貌似运行不起来

breakings, 2022-04-22:
> [image] 。。貌似运行不起来

qt库有问题 重新编译

lingandken, 2022-04-22:
我再试试。。大神。能帮忙编译个 7621用的QB  ipk包吗？

breakings, 2022-04-22:
> 我再试试。。大神。能帮忙编译个 7621用的QB ipk包吗？

我没有这个环境

breakings, 2022-04-22:
> > [image] 。。貌似运行不起来
> 
> qt库有问题 重新编译qt 或者针对mips平台还得调整编译参数 具体我也不懂这玩意 没mips平台

lingandken, 2022-04-22:
哈。。貌似超出我能力范围了。- _ - !!

lingandken, 2022-04-22:
目前重新编译一个看看吧。。不行只能等大神编译了

WYC-2020, 2022-04-22:
这玩意自己编译就是了啊

lingandken, 2022-04-22:
编译出来的用不了。。- _ - ||!

WYC-2020, 2022-04-22:
我仓库的编译出来保证能用

WYC-2020, 2022-04-22:
我记得 这玩意有动态和静态两种，静态那玩意应该可以用撒，不想编译那就用静态的嘛，都是别人编译好的

lingandken, 2022-04-22:
之前用了别人的。安装成功 luci没找到入口。换了几个LUCI都不行。。你能帮我编译个静态的吗。。7621的。。

WYC-2020, 2022-04-22:
如果只是luci的话 是不区分型号的，只要找到对应luci 编译的就行了 ，毕竟现在有那个渣渣js

lingandken, 2022-04-22:
是的。自己编译的luci。又装不上。强制安装都不行。。。

WYC-2020, 2022-04-22:
我这里有luci 18.06的，但是 这个ipk 有我新增的功能，设置web密码的功能，如果不搭配我那个qb 是没法使用这个功能的，我发给你你安装看下可以不，如果可以了，把那个设置密码屏蔽就行了
[luci-app-qbittorrent.zip](https://github.com/coolsnowwolf/lede/files/8538467/luci-app-qbittorrent.zip)

lingandken, 2022-04-22:
https://www.right.com.cn/forum/forum.php?mod=viewthread&tid=1456090&highlight=qbittorrent

用的是这位大哥编译的安装包 安装成功了 但luck界面没有 我自己i编译luci。又装不了

WYC-2020, 2022-04-22:
至于那个qbittorrent 你去下载对应的静态库就行了

lingandken, 2022-04-22:
我试试。

WYC-2020, 2022-04-22:
https://github.com/userdocs/qbittorrent-nox-static/releases/  去这里下载静态的看有你对应的平台没

lingandken, 2022-04-22:
新3 7621.貌似没看到？

breakings, 2022-04-23:
> https://github.com/userdocs/qbittorrent-nox-static/releases/ 去这里下载静态的看有你对应的平台没

他那是mips 估计也没几个人在上面折腾qb 感觉性能方面就不是太足

closed 2022-05-04 by tqangxl