coolsnowwolf/lede #9283: 想编译中兴e8820v2，各种参照dts都没成功

jin3014test2, 2022-04-20:
有时能编译成功但刷入后不启动，偶尔能启动的也是巨卡无比，要么就是编译失败。哪位大侠给发个基于coolsnowwolf/lede的dts文件？

WYC-2020, 2022-04-20:
dts这玩意应该和用的谁的op没关系吧，那玩意和硬件相关

jin3014test2, 2022-04-20:
我在网上找到好几个8820v2的dts，无线部分不一样，其他的部分差不多。
这是一种写法（好几个来源都是这样）
&pcie0 {
    mt76@0,0 {
        reg = <0x0000 0 0 0 0>;
        mediatek,mtd-eeprom = <&factory 0x0000>;
        led {
            led-active-low;
        };
    };
};
&pcie1 {
    mt76@0,0 {
        reg = <0x0000 0 0 0 0>;
        mediatek,mtd-eeprom = <&factory 0x8000>;
        ieee80211-freq-limit = <5000000 6000000>;
        led {
            led-sources = <2>;
            led-active-low;
        };
    };
};
这是另一种写法（来源[siwind](https://github.com/siwind)）
&pcie0 {
	wifi@0,0 {
		compatible = "mediatek,mt76";
		reg = <0x0000 0 0 0 0>;
		mediatek,mtd-eeprom = <&factory 0x0000>;
		nvmem-cells = <&macaddr_factory_e000>;
		nvmem-cell-names = "mac-address";

		led {
			led-active-low;
		};
	};
};

&pcie1 {
	wifi@0,0 {
		compatible = "mediatek,mt76";
		reg = <0x0000 0 0 0 0>;
		mediatek,mtd-eeprom = <&factory 0x8000>;
		nvmem-cells = <&macaddr_factory_e006>;
		nvmem-cell-names = "mac-address";

		ieee80211-freq-limit = <5000000 6000000>;
		led {
			led-sources = <2>;
			led-active-low;
		};
	};
};

LINGJP, 2022-04-21:
这机器是64mb的ram，跑5内核跑不动了，msg1500 z.00一模一样的情况，跑5内核得大幅度精简插件，跑4.x差不多

LINGJP, 2022-04-21:
64mb机器，去跑跑op19还行

jin3014test1, 2022-04-30:
> 

是说19.07么？8820v2在openwrt-19.07下没有2.4g，很奇怪

LINGJP, 2022-04-30:
确定7603 7612都编译进去了吧？19.07时代的开源驱动有时候开机会掉驱动，重启或者重刷试试

closed 2022-07-25 by aiamadeus