coolsnowwolf/lede #9285: fatal: refusing to merge unrelated histories

netlovehf, 2022-04-21:
./scripts/feeds update -a 更新时报错

Updating feed 'kenzo' from 'https://github.com/kenzok8/openwrt-packages' ...
remote: Enumerating objects: 688, done.
remote: Counting objects: 100% (688/688), done.
remote: Compressing objects: 100% (201/201), done.
remote: Total 535 (delta 265), reused 504 (delta 247), pack-reused 0
Receiving objects: 100% (535/535), 1.64 MiB | 2.11 MiB/s, done.
Resolving deltas: 100% (265/265), completed with 101 local objects.
From https://github.com/kenzok8/openwrt-packages
   5b742eb..7a31b62  master     -> origin/master
fatal: refusing to merge unrelated histories
failed.

WYC-2020, 2022-04-21:
你真是，这个仓库作者全部重新搞了一边 ，自己删除本地的 重新拉去

netlovehf, 2022-04-21:
error: Your local changes to the following files would be overwritten by merge: 	target/linux/x86/Makefile 
Please commit your changes or stash them before you merge.

closed 2022-04-22 by coolsnowwolf